nextcloud/nextcloudpi #818: Internal Server Error after Update

misterlindsey, 2019-02-27:
Looked through previous concerns from OCt, but not seeing a resolution that fits my issue. 

After running sudo apt update and sudp apt upgrade I am not able to load the page at drive.mattlindsey.com. During the update, I got a config question regarding PHP 7.2, some file that was different. I opted to keep the existing file. I believe it was a php.ini that was too long to use the compare option. 

Any pointers on what might be causing this would be greatly appreciated. I'm not sure what error log would give me more verbose information about this.

misterlindsey, 2019-02-27:
Updated NextCloudPi in admin console and everything is back up. Not sure what happened.

shadowstreik, 2019-02-28:
Possible hint - NCP is written in such a way that running update and upgrade may not necessarily be required. turning on auto updates may already cover everything needed to operate the software, properly.

A link for support is available on the NextCloudPi Mastodon page...

https://mastodon.social/@NextCloudPi

Thank you.

nachoparker, 2019-03-01:
thanks @shadowstreik

closed 2019-03-01 by nachoparker