nextcloud/nextcloudpi #823: Update failes because letsencrypt update failes

Letterus, 2019-03-06:
I'm running ncp 1.8.0 and trying to update to 1.9.1. During the update process letsencrypt is called. But letsencrypt failes because of

`FailedChallenges: Failed authorization procedure. domain.com (http-01): urn:ietf:params:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching https://domain.com/.well-known/acme-challenge/abc: Timeout during connect (likely firewall problem)`

The web server is reachable from outside, but it seems Nextcloud is called before any access to .well-known is possible. Do I need to have a specific setting in .htaccess to make this work?

In my .htaccess I once have
`  RewriteCond %{REQUEST_URI} !^/\.well-known/(acme-challenge|pki-validation)/.*
  RewriteRule ^(?:\.|autotest|occ|issue|indie|db_|console).* - [R=404,L]`

and a little down I have
`  RewriteCond %{REQUEST_URI} !^/\.well-known/(acme-challenge|pki-validation)/.*
  RewriteRule . index.php [PT,E=PATH_INFO:$1]
  RewriteBase /`

So there are two parts for `<IfModule mod_rewrite.c>`. Is this correct?

acesabe, 2019-03-06:
But are port 80 & 443 accessible from the web? Had same issue myself yesterday but a quick (and temporary if not needed/wanted permanently) port forward configuration on the router, re-run ncp-update and process no longer fails

Whether this is a bug or a feature,  could be up for discussion, I would rather an option to ignore the problem and continue anyway was available and I imagine plenty of scenarios where public https web access is not required or desired even if it is best practice regardless.

nachoparker, 2019-03-07:
LE requires both 80 and 443 to be open in order to work, not much we can do about that :)

You can close 80, but then LE autorenewal won't work

closed 2019-03-07 by nachoparker

Letterus, 2019-03-07:
I double checked that port 80 and 443 are available from the outside. Are you sure this isn't a webserver configuration issue?

[image]

nachoparker, 2019-03-08:
that might not be accurate, try to access from your phone

Letterus, 2019-03-08:
I checked using mobile connection. Checked A und AAAA record. Retried update today. Result remains the same. Any idea on how to debug?

nachoparker, 2019-03-09:
also port 80?

Letterus, 2019-03-09:
Yep, just checked the IP again using telnet.

nachoparker, 2019-03-09:
make sure you have `allowoverride all` in apache -> nextcloud.conf

Letterus, 2019-03-09:
Yes, that's set. Do you have a correct/working .htaccess somewhere to compare it to my version?

Letterus, 2019-03-09:
I do not know about letsencrypt, but I was just able to update successfully to 1.9.3. ;-)