nextcloud/nextcloudpi #826: letsencrypt broken after first time update of nextcloudpi

Chippel, 2019-03-06:
I tried to get a letsencrypt certificate but unfortunately after updating to the newest Nextcloud version I get the following error message:

[ letsencrypt ]
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh: line 62: /etc/letsencrypt/letsencrypt-auto: No such file or directory

I got the same error when I use the SSH login.

nachoparker, 2019-03-07:
You are using an outdated image. Please download the latest one

closed 2019-03-07 by nachoparker

jeanremy, 2019-03-17:
Hi, same error for me, with the latest image (03-09-2019). Is there a way to download the content of the folder, or reinstall just this part ? Thank you

jeanremy, 2019-03-17:
I've found your script here: https://help.nextcloud.com/t/unable-to-start-let-s-encrypt/49110/7, it worked like a charm ! Thank you for it, and for NCPi, great job !

bigr4232, 2019-05-16:
I am having a problem where my certificate didnt renew and so I try to run letsencrypt in the ncp control panel and it runs with no errors. The certificate is still not renewed though. I still have an unsecure connection and can't connect to nextcloud since my connection isn't secured.

cohenjm98, 2019-05-30:
I have the same problem.  Is there a way to run letsencrypt from the command line?

cohenjm98, 2019-05-30:
letsencrypt runs successfully, but creates new keys and certs in a different directory: /etc/letsencrypt/live/<my-domain>-0001

I think apache still goes to the /etc/letsencrypt/live/<my-domain> directory for the certs - which still exists and has my expired certs.  Any idea what to do?

Thanks,

ovpc, 2019-05-30:
Just remove all the folders in /live and /archive, rm files from /renewal  and run it again.

cohenjm98, 2019-05-30:
Thanks - that seems to have worked.  Will renewals work going forward?