nextcloud/nextcloudpi #827: v1.9.1 update failed: ncc: command not found

nero120, 2019-03-06:
Tried to update via `sudo ncp-config`, this is the output I got:

```
Downloading updates
Performing updates
Running letsencrypt
Creating virtual environment...
Installing Python packages...
Installation succeeded.
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Cert not yet due for renewal
Keeping the existing certificate

-------------------------------------------------------------------------------
Certificate not yet due for renewal; no action taken.
-------------------------------------------------------------------------------
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh: line 93: ncc: command not found
```

Line 93 of that script refers to these commands:

```
# Configure Nextcloud
ncc config:system:set trusted_domains 4 --value=$DOMAIN
ncc config:system:set overwrite.cli.url --value=https://"$DOMAIN"/
```

I've never seen this `ncc` command, if I want to access the nextcloud cli I always use `sudo -u www-data php /var/www/nextcloud/occ` as per the [manual](https://docs.nextcloud.com/server/stable/admin_manual/configuration_server/occ_command.html?highlight=occ). What is `ncc`?

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
NextCloudPi version  v1.8.0
NextCloudPi image    NextCloudPi_09-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda sdb
datadir              /media/data/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      50G/932G
rootfs usage         2.0G/28G
swapfile             /dev/mmcblk0p3
dbdir                 (doesn't exist)
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               51min

```
</details>

[... 122 lines cut ...]
```

{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Invalidating remembered session tokens older than 2019-02-19T23:30:02+00:00","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 15 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OC\\Preview\\BackgroundCleanupJob job with ID 2680","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OC\\Preview\\BackgroundCleanupJob job with ID 2680 in 1 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\DAV\\BackgroundJob\\UpdateCalendarResourcesRoomsBackgroundJob job with ID 2683","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\DAV\\BackgroundJob\\UpdateCalendarResourcesRoomsBackgroundJob job with ID 2683 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\Support\\BackgroundJobs\\CheckSubscription job with ID 2686","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\Support\\BackgroundJobs\\CheckSubscription job with ID 2686 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 2517","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 2517 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 2522","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Invalidating session tokens older than 2019-03-05T23:30:03+00:00","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Invalidating remembered session tokens older than 2019-02-19T23:30:03+00:00","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Invalidating session tokens older than 2019-03-05T23:30:03+00:00","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Invalidating remembered session tokens older than 2019-02-19T23:30:03+00:00","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OC\\Authentication\\Token\\DefaultTokenCleanupJob job with ID 2522 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OC\\Log\\Rotate job with ID 2523","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OC\\Log\\Rotate job with ID 2523 in 0 seconds","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Run OCA\\Files\\BackgroundJob\\ScanFiles job with ID 5","userAgent":"--","version":"15.0.5.3"}
{"reqId":"uIg6ATsVFzbOJLv6pcJb","level":0,"time":"2019-03-06T23:30:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Finished OCA\\Files\\BackgroundJob\\ScanFiles job with ID 5 in 0 seconds","userAgent":"--","version":"15.0.5.3"}

```
</details>

nachoparker, 2019-03-07:
seems like you missed some important update in the past. `ncc` is an alias to `sudo -u www-data php /var/www/nextcloud/occ`

closed 2019-03-07 by nachoparker

nero120, 2019-03-07:
Was ncc created by Nextcloud or NCP? I might just do a refresh of ncp and then restore the db from backup, might solve some of the other issues I'm having.

nachoparker, 2019-03-07:
by NCP. If you have missed updates it is always a good idea

TimTaler007, 2019-03-16:
> by NCP. If you have missed updates it is always a good idea

I have a similar problem to update an old version of NCP (1.5.1) to current version 1.10.0.
It will stop the update with message:
`./update.sh: line 202: ncc: command not found`
So I tried to add the alias ncc with command `sudo -u www-data php /var/www/nextcloud/occ` but this alias does not work for sudo. How can I solve this problem or do a refresh of ncp?

nachoparker, 2019-03-16:
better to backup, and restore in the latest image

nisses, 2019-04-18:
The other way is to put the following content into /usr/local/bin/ncc file:
```
#!/bin/bash
sudo -u www-data php /var/www/nextcloud/occ "$@"
```
remember to `chmod +x /usr/local/bin/ncc` 

But... since ncc was introduced in v0.56.19 which was long time ago 2018-06-06 and so many people have this problem (and other problems caused by skipped/failed/unfinished updates) there should be a way to check the state of NextCloudPi and NextCloud for errors and inconsistency. Relational DB and the files and directory structure of NextCloud and NextCloudPi can be inspected by a script. There is ncc integrity:check-app and ncc integrity:check-core and ncc status but I think it does not cover all area of possible problems.

TimTaler007, 2019-04-18:
> The other way is to put the following content into `/usr/local/bin/ncc` file:
> 
> ```
> #!/bin/bash
> sudo -u www-data php /var/www/nextcloud/occ "$@"
> ```
> remember to `chmod +x /usr/local/bin/ncc`
> 
>
This solved my update-problem with version 1.11.2.
Thank you so much for your help and support! I had no hope to solve it without a restore and found no time to do it. Your tip was easier to realize and works for the moment and the current version.

nachoparker, 2019-04-18:
@nisses  there's an issue / improvement request to improve the update process because people skip updates even though they shouldn't.

But there is nobody working on it right now