nextcloud/nextcloudpi #832: Fix/find a workaround for btrfs snapshots on docker

theCalcaholic, 2019-03-10:
When using btrfs-snapshots in docker, a variety of ncp-apps fail (due to not being able to delete or create the btrfs volume, for example).
It should be evaluated

1) which apps have these kind of problems
2) if it's feasible to implement workarounds
3) if functioning btrfs-snapshots can be achieved by elevating container privileges (which then should be added to the wiki)

theCalcaholic, 2019-03-10:
@nachoparker I don't seem to be able to assign this to myself, so feel free to do that. :)

nachoparker, 2019-03-10:
thanks for the help, that would be awesome. Also `nc-update-nextcloud` is one of them

Zakspeed, 2019-07-15:
In docker - NexcloudPi   "x.x.x.x:4443"
Update current instance to a new Nextcloud version
```
[nc-update-nexcloud]
BTRFS not supported

```

ovpc, 2019-07-15:
[image]
Need to do that from commandline, comment out the btrfs check with
> sudo nano /usr/local/bin/ncp-update-nc

Zakspeed, 2019-07-15:
Thanks
I made that change in ncp-update-nc (comment that line)
And the update of docker-NexcloudPi, and update !!!
:)

Zakspeed, 2019-07-15:
But, I do not know what I do wrong, that if I do tests, and stop the service or restart the machine (the raspidocker :)))
do not lift the container with the NCP

nachoparker, 2021-10-15:
BTRFS is now supported

closed 2021-10-15 by nachoparker