nextcloud/nextcloudpi #836: ./update.sh: line 181: /etc/php/7.2/fpm/conf.d/90-ncp.ini: No such file or directory

bernd-wechner, 2019-03-12:
Every update for the past month or so always reports this error:
```
    $ sudo ncp-update
    Downloading updates
    Performing updates
    ./update.sh: line 181: /etc/php/7.2/fpm/conf.d/90-ncp.ini: No such file or directory
```
Something is broken.

Trying to find that file:
```
$ sudo updatedb
$ locate 90-ncp.ini
$ locate ncp.ini
$ 
```
turns nothing up. Oddly I can't find an update.sh on the system anywhere either. So I will continue to attempt diagnosis but wanted to report this in case it is a known issue or easily identified by a dev with more familiarity with ncp-update than I.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.8.0
NextCloudPi image    NextCloudPi_05-28-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      51G/917G
rootfs usage         2.4G/30G
swapfile             /media/Passport/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               7days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.0.14",
            "2": "nephele",
[... 70 lines cut ...]
```

{"reqId":"XHuhO38AAQEAAC@GPfEAAABG","level":0,"time":"2019-03-03T09:41:15+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhQn8AAQEAAC@GPfIAAEcA","level":0,"time":"2019-03-03T09:41:22+00:00","remoteAddr":"192.168.0.11","user":"--","app":"core","method":"GET","url":"\/csrftoken","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery-ui-fixes.scss, ignoring","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"14.0.4.2"}
{"reqId":"XHuhQn8AAQEAAC@GPfIAAEcA","level":0,"time":"2019-03-03T09:41:22+00:00","remoteAddr":"192.168.0.11","user":"--","app":"core","method":"GET","url":"\/csrftoken","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/server.scss, ignoring","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"14.0.4.2"}
{"reqId":"XHuhQn8AAQEAAC@GPfIAAEcA","level":0,"time":"2019-03-03T09:41:22+00:00","remoteAddr":"192.168.0.11","user":"--","app":"core","method":"GET","url":"\/csrftoken","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/css-variables.scss, ignoring","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"14.0.4.2"}
{"reqId":"XHuhQn8AAQEAAC@GPfIAAEcA","level":0,"time":"2019-03-03T09:41:22+00:00","remoteAddr":"192.168.0.11","user":"--","app":"core","method":"GET","url":"\/csrftoken","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"14.0.4.2"}
{"reqId":"XHuhQn8AAQEAAC@GPfIAAEcA","level":0,"time":"2019-03-03T09:41:22+00:00","remoteAddr":"192.168.0.11","user":"--","app":"core","method":"GET","url":"\/csrftoken","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"14.0.4.2"}
{"reqId":"XHuhRX8AAQEAAC@GPfMAAABE","level":0,"time":"2019-03-03T09:41:25+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery-ui-fixes.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhRX8AAQEAAC@GPfMAAABE","level":0,"time":"2019-03-03T09:41:25+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/server.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhRX8AAQEAAC@GPfMAAABE","level":0,"time":"2019-03-03T09:41:25+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/css-variables.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhRX8AAQEAAC@GPfMAAABE","level":0,"time":"2019-03-03T09:41:25+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhRX8AAQEAAC@GPfMAAABE","level":0,"time":"2019-03-03T09:41:25+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhUH8AAQEAAC@GPfQAAABU","level":0,"time":"2019-03-03T09:41:36+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery-ui-fixes.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhUH8AAQEAAC@GPfQAAABU","level":0,"time":"2019-03-03T09:41:36+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/server.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhUH8AAQEAAC@GPfQAAABU","level":0,"time":"2019-03-03T09:41:36+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/css-variables.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhUH8AAQEAAC@GPfQAAABU","level":0,"time":"2019-03-03T09:41:36+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHuhUH8AAQEAAC@GPfQAAABU","level":0,"time":"2019-03-03T09:41:36+00:00","remoteAddr":"192.168.0.1","user":"--","app":"core","method":"MKCOL","url":"\/remote.php\/webdav\/Joplin\/.sync\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/search\/css\/results.scss, ignoring","userAgent":"node-fetch\/1.0 (+https:\/\/github.com\/bitinn\/node-fetch)","version":"14.0.4.2"}
{"reqId":"XHug0n8AAQEAAC@GPeAAAABB","level":1,"time":"2019-03-03T09:41:48+00:00","remoteAddr":"192.168.0.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=CLG1sEOeZ7DG1d0XZmqLdhmTRXFZy6yR6Eur9vWUXkM%3D%3ATvz3nwfkM4ilrY5mHzi%2FHUjnDCYtquPYiy3pw5mnHxU%3D","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"Mozilla\/ [... cut ...] }
{"reqId":"XHug0n8AAQEAAC@GPeAAAABB","level":1,"time":"2019-03-03T09:41:48+00:00","remoteAddr":"192.168.0.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=CLG1sEOeZ7DG1d0XZmqLdhmTRXFZy6yR6Eur9vWUXkM%3D%3ATvz3nwfkM4ilrY5mHzi%2FHUjnDCYtquPYiy3pw5mnHxU%3D","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko\/20100101 Firefox\/65.0","version":"15.0.5.3"}

```
</details>

bernd-wechner, 2019-03-12:
OK I found the code that trips. It's here:

https://github.com/nextcloud/nextcloudpi/blob/master/update.sh

and at line 181 (which the cat > line):
```
  # disable .user.ini
  PHPVER=7.2
  [[ -f /etc/php/${PHPVER}/fpm/conf.d/90-ncp.ini ]] || {
    MAXFILESIZE="$(grep upload_max_filesize /var/www/nextcloud/.user.ini | cut -d= -f2)"
    MEMORYLIMIT="$(grep memory_limit        /var/www/nextcloud/.user.ini | cut -d= -f2)"
    cat > /etc/php/${PHPVER}/fpm/conf.d/90-ncp.ini <<EOF
; disable .user.ini files for performance and workaround NC update bugs
user_ini.filename =
; from Nextcloud .user.ini
upload_max_filesize=$MAXFILESIZE
post_max_size=$MAXFILESIZE
memory_limit=$MEMORYLIMIT
mbstring.func_overload=0
always_populate_raw_post_data=-1
default_charset='UTF-8'
output_buffering=0
; slow transfers will be killed after this time
max_execution_time=3600
max_input_time=3600
EOF
    bash -c "sleep 3 && service php$PHPVER-fpm restart" &
}
```

And a tree of my php dir:

```
 $ tree /etc/php
/etc/php
├── 5.6
│   └── mods-available
│       ├── igbinary.ini
│       ├── imagick.ini
│       ├── redis.ini
│       └── smbclient.ini
├── 7.0
│   ├── cli
│   │   ├── conf.d
│   │   │   ├── 10-mysqlnd.ini -> /etc/php/7.0/mods-available/mysqlnd.ini
│   │   │   ├── 10-opcache.ini -> /etc/php/7.0/mods-available/opcache.ini
│   │   │   ├── 10-pdo.ini -> /etc/php/7.0/mods-available/pdo.ini
│   │   │   ├── 15-xml.ini -> /etc/php/7.0/mods-available/xml.ini
│   │   │   ├── 20-bz2.ini -> /etc/php/7.0/mods-available/bz2.ini
│   │   │   ├── 20-calendar.ini -> /etc/php/7.0/mods-available/calendar.ini
│   │   │   ├── 20-ctype.ini -> /etc/php/7.0/mods-available/ctype.ini
│   │   │   ├── 20-curl.ini -> /etc/php/7.0/mods-available/curl.ini
│   │   │   ├── 20-dom.ini -> /etc/php/7.0/mods-available/dom.ini
│   │   │   ├── 20-exif.ini -> /etc/php/7.0/mods-available/exif.ini
│   │   │   ├── 20-fileinfo.ini -> /etc/php/7.0/mods-available/fileinfo.ini
│   │   │   ├── 20-ftp.ini -> /etc/php/7.0/mods-available/ftp.ini
│   │   │   ├── 20-gd.ini -> /etc/php/7.0/mods-available/gd.ini
│   │   │   ├── 20-gettext.ini -> /etc/php/7.0/mods-available/gettext.ini
│   │   │   ├── 20-iconv.ini -> /etc/php/7.0/mods-available/iconv.ini
│   │   │   ├── 20-igbinary.ini -> /etc/php/7.0/mods-available/igbinary.ini
│   │   │   ├── 20-imagick.ini -> /etc/php/7.0/mods-available/imagick.ini
│   │   │   ├── 20-intl.ini -> /etc/php/7.0/mods-available/intl.ini
│   │   │   ├── 20-json.ini -> /etc/php/7.0/mods-available/json.ini
│   │   │   ├── 20-ldap.ini -> /etc/php/7.0/mods-available/ldap.ini
│   │   │   ├── 20-mbstring.ini -> /etc/php/7.0/mods-available/mbstring.ini
│   │   │   ├── 20-mcrypt.ini -> /etc/php/7.0/mods-available/mcrypt.ini
│   │   │   ├── 20-mysqli.ini -> /etc/php/7.0/mods-available/mysqli.ini
│   │   │   ├── 20-pdo_mysql.ini -> /etc/php/7.0/mods-available/pdo_mysql.ini
│   │   │   ├── 20-pdo_pgsql.ini -> /etc/php/7.0/mods-available/pdo_pgsql.ini
│   │   │   ├── 20-pdo_sqlite.ini -> /etc/php/7.0/mods-available/pdo_sqlite.ini
[... 141 lines cut ...]
│       ├── pdo_mysql.ini
│       ├── pdo_pgsql.ini
│       ├── pdo_sqlite.ini
│       ├── pgsql.ini
│       ├── phar.ini
│       ├── posix.ini
│       ├── redis.ini
│       ├── shmop.ini
│       ├── smbclient.ini
│       ├── sockets.ini
│       ├── sqlite3.ini
│       ├── sysvmsg.ini
│       ├── sysvsem.ini
│       ├── sysvshm.ini
│       └── tokenizer.ini
└── 7.3
    └── mods-available
        ├── igbinary.ini
        ├── imagick.ini
        ├── redis.ini
        └── smbclient.ini

15 directories, 182 files
```
Looks like version confusion and somehow my system is not what ncp-update expects anymore. Which is odd as it's stock standard NCP box. Thoughts welcome.

nachoparker, 2019-03-13:
do you have autoupdates enabled?

bernd-wechner, 2019-03-13:
I just checked. If by "autoupdates enabled" you mean, either or both of the packages nc-autoupdate-nc and/or nc-autoupdate-ncp were enabled, no neither was, and I just enabled them now (wondering if that's a good idea).

Not sure how or why PHP might have deviated form the iso image of NCP I used. Only thing I can imagine (and it only imagine, not at all based on evidence) is that I did recently install awstats on that Pi.

Mind you I suspect update.sh should be robust against such things or inform of them in a more elegant manner. Or?

bernd-wechner, 2019-03-13:
Oh, I notice now too, that this seems not merely advisory, but to be a fatal error as:
```
NextCloudPi v1.8.0 is outdated
update to v1.9.3 through 'ncp-config' or type 'sudo ncp-update'
$ sudo ncp-update
Downloading updates
Performing updates
Running nc-autoupdate-nc
automatic Nextcloud updates enabled
./update.sh: line 181: /etc/php/7.2/fpm/conf.d/90-ncp.ini: No such file or directory
```

in short, it's not updating. So I guess I have to resolve this in order to keep up to date.

nachoparker, 2019-03-13:
If you don't have autoupdates enabled you have to be careful to update at least once a month.

You either skipped the month where we upgraded to 7.2, or most likely messed the system up by installing other things.

We assume that no updates are skipped because we don't have the manpower to handle this better. 

Who know what else you skipped, so I suggest backup + restore in a new instance, and activate autoupdates

bernd-wechner, 2019-03-13:
I'd much rather apply the interim updates one by one or update to PHP 7.2. I mean there is always an appeal to starting any system fresh, but to be honest, there's work either way (as getting SSL working is a bit of job when behind a NAS firewall and it is working now - mostly, one of my postrenewal hooks has not been firing so every 3 months I find myself doing another round of diagnosis and pushing the certs manually to the gateway (which runs lightttpd which sadly cannot proxy SSL cert requests to LAN servers so one trick is to publish them to the gateway router and lightttpd serves the certs from there). Either way the whole schmozzle takes some setting up which as much work probably as applying updates past  v1.8.0 one by one. They surely exist on gitthub and I'll have an explore (on basis that learning is much more fun than chores).

Assuming no updates were skipped is fine and sensible of course. No problem with that. Appreciate the hands on deck problem very much, we all have it. No pressure from me to do better than the all too awesome job that NCP already does! It is just awesome!

Of course any tips that can direct my manpower are welcome always ;-).

And as a fallback, yes, I can always backup and restore to a fresh instance. But I'd rather try updates first and see if I can find them int he github history as cleanly defined commits that I could fetch in place of the head version one by one and apply. I shall see.

nachoparker, 2019-03-13:
cool, just track the history of `update.sh`

closed 2019-03-13 by nachoparker

bernd-wechner, 2019-03-14:
Nacho, I'm happy to do that, but just took a look at the history:

https://github.com/nextcloud/nextcloudpi/commits/master/update.sh

and I see loads of commits, and no system of tagging is evident to see which ones were versioned updates. So I need a quick tip to proceed if you can provide it. 

1. Is every commit of update.sh a formal update that should be applied?
2. Really? Like there were 5 commits on March 6th alone. That's a lot of updates!
3. If not, which snapshots of update.sh represent formal updates and how do I identify them?
4. Is update.sh self contained? I mean if it fetches stuff from git is it likely to work. Technically an update.sh from March 6 for example if it needs stuff from github fetches it at state March 6 not latest state.

nachoparker, 2019-03-16:
if you really want to take this path, check out every commit and apply `update.sh`. Don't promise anything though.

bernd-wechner, 2019-03-17:
Nacho, but I still need to know:

Which commit relates to v1.8.0 as  as it would nuts to run updtae.sh from the version I have and before.

And it would help enormously to receive one small tip on where and how version numbers tie to commits. That alone would answer the question above and let me try some diagnosis.

bernd-wechner, 2019-03-17:
Well, I forked, cloned and grepped. SO have the most likely aid now here:

https://github.com/bernd-wechner/nextcloudpi/blob/master/changelog.md

In short, this is the version trail I need to install:

v1.10.2 (2019-03-16) freeDNS: fix hash
v1.10.1 (2019-03-13) nc-update-nc-apps-auto: notify user
v1.10.0 (2019-03-13) add nc-previews-auto
v1.9.8 (2019-03-13) nc-scan: improvements
v1.9.7 (2019-03-13) nc-previews: improvements
v1.9.6 (2019-03-13) nc-scan-auto: improvements
v1.9.5 (2019-03-09) nc-init: previews settings
v1.9.4 (2019-03-09) unattended upgrades: update labels
v1.9.3 (2019-03-09) wizard: fix headers
v1.9.2 (2019-03-08) cleanup update.sh
v1.9.1 (2019-03-03) fix LE cron
v1.9.0 (2019-03-03) upgrade to NC15.0.5
v1.8.4 (2019-02-23) nc-nextcloud: disable .user.ini
v1.8.3 (2019-02-23) nc-limits: autocalculate database memory
v1.8.2 (2019-02-22) lamp: adjust mariadb parameters
v1.8.1 (2019-03-05) nc-datadir: Add SATA to description (#822)

And by finding the commits on changelog.md that add each version I can grab the update.sh current at that time. I think what this reveals is:

1. That you do (surprisingly) at times push mutliple versions in one day! 2019-03-13 was busy with 5 versions pushed! 
2. That the commit trail on update.sh which has for example 5 commits on 2019-03-06 does not relate one to one with version pushes it seems. More research needed on lag and reliability of changelog.md though.

nachoparker, 2019-03-17:
version numbers are git tags, easy :)

bernd-wechner, 2019-03-17:
Thanks Nacho! Found that in the cloned copy. Alas what is the story with 1.8.5 which is dangling on a branch of its own not merged back into master?  I have a feeling I should not apply its update.sh!

bernd-wechner, 2019-03-17:
Seems to me also the easiest way by a mile to do this is simply to patch ncp-update to take an optional second parameter, a tag name and if provided clone that tagged release! I shall do this locally to perform the updates one by one, but I remain curious:

1. What about 1.8.5? Oddball release that seems dangling
2. Would you be interested in a PR for a thus patched ncp-update? It seems a great way to conveniently apply retrospective updates by release number! git rocks!

bernd-wechner, 2019-03-17:
Another question to explore is where is the code that detects and update posts the little notice on top of the page and if auto-updates are on kick starts an ncp-update? I will look for it, but you can save me time if you know where it is.

bernd-wechner, 2019-03-17:
Here's a neat trick I worked out to list all the versions to walk:
```
git tag --sort=version:refname | tail -n +`git tag --sort=version:refname | grep -n 'v1.8.0' | cut -f1 -d:` 
````
though instead of 'v1.8.0' on a live system I'd use `cat /usr/local/etc/ncp-version`.

It however includes 1.8.5 which has weird standing on the branches (seems not to be ever made live). Would appreciate commentary on that. Here's current output:

```
v1.8.0
v1.8.1
v1.8.2
v1.8.3
v1.8.4
v1.8.5
v1.9.0
v1.9.1
v1.9.2
v1.9.3
v1.9.4
v1.9.5
v1.9.6
v1.9.7
v1.9.8
v1.10.0
v1.10.1
v1.10.
```

bernd-wechner, 2019-03-17:
OK, I looked closely at 1.8.1 by checking out that version. But, changelog.md  has as its latest entry:
```
[v1.7.0](https://github.com/nextcloud/nextcloudpi/commit/b0a83b0) (2019-02-17) add nc-restore-snapshot
```
which is disconcerting. So I now have two open questions for you Nacho:

1. When a version is tagged, is it fair to expect that version is entered into changelog.md? If it's not, why not?
2. Version 1.8.5 sits on a dangling branch, not merged to master. Did everyone upgrade from .18.4 to 1.8.6 directly?

nachoparker, 2019-03-17:
`v1.8.5 ` was never applied, only apply in `master`. We jumped to `v1.9.0` instead.

We develop things first in the `devel` branch and they don't always make it to `master`. I deleted the dangling tag

nachoparker, 2019-03-17:
There's certainly room for improvement in the update process. The main reason I only support one month worth of patches is lack of manpower.

See https://github.com/nextcloud/nextcloudpi/issues/759 and feel free to join the conversation with suggestions. If we come to a good solution it would be great to receieve a PR with it ;)

bernd-wechner, 2019-03-17:
Thanks for confirming!

Out of interest I noted:

```
$ cat /usr/local/etc/ncp-changelog
| [v1.7.0] add nc-restore-snapshot
| [v1.6.7] nc-backup-auto: notify failures
| [v1.6.6] nc-automount: fix NFS delay
| [v1.6.5] nc-trusted-domains: empty values by default
| [v1.6.4] nc-nextcloud: update description
| [v1.6.3] ncp-update-nc: BTRFS basedir not supported
| [v1.6.2] nc-datadir: lift mountpoint restriction
| [v1.6.1] nc-nextcloud: add a warning
| [v1.6.0] upgrade to NC15.0.4
| [v1.5.2] nc-trusted-domains: add description
| [v1.5.1] nc-update-nc-apps-auto: log upgrades
| * [grafted, tag: v1.5.0] added nc-update-nc-apps and nc-update-nc-apps-auto
[v1.10.2] freeDNS: fix hash
[v1.10.1] nc-update-nc-apps-auto: notify user
[v1.10.0] add nc-previews-auto
[v1.9.8] nc-scan: improvements
[v1.9.7] nc-previews: improvements
[v1.9.6] nc-scan-auto: improvements
[v1.9.5] nc-init: previews settings
[v1.9.4] unattended upgrades: update labels
[v1.9.3] wizard: fix headers
[v1.9.2] cleanup update.sh
[v1.9.1] fix LE cron
```

which looks odd, and unreliable as my UI says I'm on 1.8.0 as does:

```
$ cat /usr/local/etc/ncp-version
v1.8.0
```

a-jacobo, 2019-03-26:
Hi,

 I'm having this same issue but I have auto updates turned one. Despite this my installation is stuck in 1.8.0 with the same error reported in this thread.
 I tried running the install.sh script to overwrite my installation with a new one, but nothing, I'm still in 1.8.0. So I have some questions on how to proceed:
What is the proper way to reinstall and restore an old configuration? 
Can I use my configuration which is for 1.8.0 with newer versions?
If I point this restored installation to my old data directory will it pick up all the data? Or will I loose it?

Thanks!
Adrian.

Thanks!
Adrian.

nachoparker, 2019-03-26:
> If I point this restored installation to my old data directory will it pick up all the data? Or will I loose it?

It will use it

a-jacobo, 2019-03-27:
How can I completely uninstall nextcloudpi to install it again? If I just run the install.sh script it still installs v1.8.0.

nachoparker, 2019-03-27:
The easiest is to start over with a fresh OS

foocp, 2019-04-07:
I had the same problem, but just tried running updating again, and now it works! updated from v1.6.3 to v1.11.2.

a-jacobo, 2019-04-07:
I tried updating now and it worked! Thanks for the heads up, I was getting ready to reinstall everything!

bernd-wechner, 2019-05-21:
My home system is also seems to be updating fine again.  I would love to see some confirmation that this was a real fix, and our systems have integrity. One day I might backup and reinstall form an image, but hey, I'm time poor and that's a fair bit of work.