nextcloud/nextcloudpi #837: letsencrypt script with multiple domains

feutl, 2019-03-12:
I would like to have a feature to add an additional domain to the letsencrypt script handled by ncp admin panel.
I have setup collabora using a seperate domain on the NCP instance using and running into letsencrypt issues since that. I need to run letsencrypt manually via CLI because it cron seems to not like the second domain.

I would just like to see a comma seperated list in the ncp admin panel which allows to create and place just one certificate for 2 or more domains :)

thanks

nachoparker, 2019-03-13:
no, this has not been implemented

feutl, 2019-03-13:
great to have the "enhancement" tag being added :)

thanks, NCP is really just the best way to run nextcloud ;)

flosky, 2019-04-07:
I would like to do the same thing. 

What would I do to change the certificate myself with certbot? For example, I can not find the auto-renew script, there is no cron job configured and no systemd service registered, at least not one I could find

feutl, 2019-04-09:
I am using certbot manually right now, little annoying because NCP cron complains about renewing certs not possible, but not really an issue ;)

/etc/letsencrypt is the location where my files are located

tjohs, 2019-05-27:
I have the very same request... 
My ISP provides a very crypting domain name that works well, but you would never be able to remember this. Therefore I have used an additional free dynamic dns service so far. I'd like to have both domains being handled by letsencrypt to create/renew certificates.

nachoparker, 2019-05-28:
For me this is low priority, but PRs are welcome to implement this. It shouldn't hard at all.

The cron entry is here `/etc/cron.weekly/letsencrypt-ncp`

Siam1205, 2019-10-30:
Hi,
what's the status of this issue? Will this be implemented in near future? The only alternative is the one @feutl mentioned. However, I wonder whether manual configuring letsencrypt/certbot/apache might interfere with other apache- or letsencrypt-related options that you set via the NCP web interface or the CLI.

feutl, 2019-10-31:
> Hi,
> what's the status of this issue? Will this be implemented in near future? The only alternative is the one @feutl mentioned. However, I wonder whether manual configuring [...]

I have the second domain setup with the letsencrypt CLI tool and this is working smoothly for month. Even I cannot add my second domain into the NCP webui, NCP is renewing it with its scheduled letsencrypt job.

It works, but it is not integrated transparently in the webui.

Siam1205, 2019-10-31:
> I have the second domain setup with the letsencrypt CLI tool and this is working smoothly for month. Even I cannot add my second domain into the NCP webui, NCP is renewing it with its scheduled letsencrypt job.

Thank you for your reply.
So, you did not use certbot with a cronjob like you said before but instead you used the NCP CLI? How did you specify your domains? Space-separated like "example.com www.example.com"?

feutl, 2019-11-04:
nonono, sorry for the misunderstanding
I just added a second domain using the certbot cli tool as described on the certbot homepage.
because NCP uses the `letsencrypt-auto` tool for the crojob, the second domain gets renewed automatically.

You can not use NCP-Web nor NCP-CLI but the certbot cli as normal and because of the generic auto renew cronjob from NCP you are gonna be fine.

Siam1205, 2019-11-05:
Thank you for your answer. I just want to add the 'www' domain to my certificate as mentioned in my example before. Because otherwise, connecting to "example.com" will work properly but connecting to "www.example.com" will result in an insecure connection. It seems like I am the only one experiencing that issue?

I just wonder whether yours is the best workaround for that use case...

Siam1205, 2019-11-05:
Nevermind. I found another solution: What I'm doing right now is completeley disabling the NCP letsencrypt functions while using the certbot with arguments "certonly [...] -d example.com -d www.example.com". Therefore, you need to manually edit the nextcloud.conf file, backup it and finally hope that potential NCP changes don't mess it up

NCP is great. It simply works and offers so many features out of the box. But for this particular case, I chose a workaround without using NCP.

joshtrichards, 2025-06-17:
Added long ago in #1025

PR #1025 (2019-11-18): Letsencrypt: support second domain
The ncp letsencrypt app now allows a second domain to be specified.
Fix: #837
closed 2025-06-17 by joshtrichards