nextcloud/nextcloudpi #838: nc-restore.sh: Check free space before untar the backupfile.tar

Tanarri, 2019-03-12:
**Description**  
When a restore is done with nc-restore.sh the user needs at least as much free disc space as the backup-tar-file is large.

**Example**  
I had a backup-file "/media/USB-STICK/ncp-backup/nextcloud-bkp_xxxxxxxx.tar " on a usb-stick, which is 25 GB large.
In line 44 in https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-restore.sh this .tar file is extracted in a subfolder, without checking, if the is enough free space on the drive.

It should be checked (or the user warned) that there has to be enough free disc space for the whole nc-data + nc-db

TanarRi

nachoparker, 2019-03-13:
Good idea, we do check during the backup 

https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup.sh#L50

Feeling brave to port that code chunk to `nc-restore`?

Tanarri, 2019-03-14:
I have tried it yesterday, but have a small issue...
If the patch is ready I will make a pull-request

TanarRi

nachoparker, 2019-03-16:
cool, please let me know once it's ready

closed 2019-03-16 by nachoparker