nextcloud/nextcloudpi #841: Hi,

pehebo, 2019-03-14:
**System information**

pehebo, 2019-03-16:
Hi nachoparker,
i have a question: my freedns-ip isnt updated, as mentioned before, though hash and domain are 100% okay. The script, that cron executes frequently, is freedns.sh, located at /usr/local/bin. It has following content:
#!/bin/bash
echo "FreeDNS client started"
echo "https://freedns.afraid.org/dynamic/update.php?"
registeredIP=XXX.XXX.XXX.XXX (removed by me)
currentIP=$(wget -q -O - http://checkip.dyndns.org|sed s/[^0-9.]//g)
    [ "$currentIP" != "$registeredIP" ] && {
        wget -q -O /dev/null https://freedns.afraid.org/dynamic/update.php?
  }
echo "Registered IP: $registeredIP | Current IP: $currentIP"

My question: when registeredip and current ip dont match, wget -q -O /dev/null https://freedns.afraid.org/dynamic/update.php? is called, but WITHOUT hash. So, it cant update the registered domain, or am i wrong?

I would very much like to get a short answer whether Im wrong or right...
Thank you very much for this great peace of work!
Yours sincerely, Peter

nachoparker, 2019-03-16:
Yes, that looks to be the issue. It should have the hash after the `?`

nachoparker, 2019-03-16:
It should be fixed now in `v1.10.3`. Thanks for the help!

closed 2019-03-16 by nachoparker