nextcloud/nextcloudpi #843: nc-update-nextcloud not restoring data

MichaelHettmer, 2019-03-15:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.9.3
NextCloudPi image    NextCloudPi_docker_01-18-19
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      54G/148G
rootfs usage         54G/148G
swapfile             /dev/sda2
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    15.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "***REMOVED SENSITIVE VALUE***",
            "1": "172.20.0.6",
            "2": "nextcloudpi.local",
            "3": "nextcloudpi",
            "4": "nextcloudpi.lan",
            "11": "***REMOVED SENSITIVE VALUE***",
            "6": "***REMOVED SENSITIVE VALUE***",
            "20": "***REMOVED SENSITIVE VALUE***",
            "21": "proxy",
            "22": "ncp"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.4.0",
        "overwrite.cli.url": "***REMOVED SENSITIVE VALUE***",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 97 lines cut ...]
{"reqId":"oHMSthtMPgEr4tZLAWof","level":2,"time":"2019-03-15T10:30:01+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Hx4ix1BhXtImNY8D5NG4","level":3,"time":"2019-03-15T10:45:01+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Hx4ix1BhXtImNY8D5NG4","level":2,"time":"2019-03-15T10:45:01+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"QnGgDcGFvr8JCcV6WMZf","level":3,"time":"2019-03-15T11:00:01+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"QnGgDcGFvr8JCcV6WMZf","level":2,"time":"2019-03-15T11:00:01+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Cgv7cGSwF036GQFNql7E","level":3,"time":"2019-03-15T11:15:01+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Cgv7cGSwF036GQFNql7E","level":2,"time":"2019-03-15T11:15:01+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"MYRbqtcd3ilehiHRkqvX","level":3,"time":"2019-03-15T11:30:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"MYRbqtcd3ilehiHRkqvX","level":2,"time":"2019-03-15T11:30:02+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Xq2orw90UtmnvDHnl322","level":3,"time":"2019-03-15T11:45:02+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"Xq2orw90UtmnvDHnl322","level":2,"time":"2019-03-15T11:45:02+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"L7I3pq37y7DhrahLWvgm","level":3,"time":"2019-03-15T11:53:26+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"L7I3pq37y7DhrahLWvgm","level":2,"time":"2019-03-15T11:53:26+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"m9HSgW3eqVQUCAAyK4LI","level":3,"time":"2019-03-15T11:53:43+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"m9HSgW3eqVQUCAAyK4LI","level":2,"time":"2019-03-15T11:53:43+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}
{"reqId":"j4oGYehTMUCCIEretIQg","level":3,"time":"2019-03-15T11:53:46+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"file_get_contents(\/var\/www\/nextcloud\/data\/news\/config\/config.ini): failed to open stream: No such file or directory at \/data\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#216","userAgent":"--","version":"15.0.4.0"}
{"reqId":"j4oGYehTMUCCIEretIQg","level":2,"time":"2019-03-15T11:53:46+00:00","remoteAddr":"","user":"--","app":"news","method":"","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"15.0.4.0"}

```
</details>

**Description**
Updating from 15.0.4.0 to 15.0.5 with nc-update-nextcloud does not restore my files in the data directory, so they are basically just deleted. Had a similar issue before updating from 14.0.4.2 to 15.0.0.2 on a debian system with curl installer, but this was fixed by @nachoparker in 338da330f4ce312a4d1fc44beaf9ababe2863888.

[ncp.log](https://github.com/nextcloud/nextcloudpi/files/2970871/ncp.log)

nachoparker, 2019-03-16:
I think I know what's going on. Did you import this instance from a non docker NCP installation?

MichaelHettmer, 2019-03-18:
Yes I did. The Debian curl installer one I mentioned above.

nachoparker, 2019-03-19:
cool, fixed now

closed 2019-03-19 by nachoparker