nextcloud/nextcloudpi #844: nc-update-nextcloud path for temporary backup

paschaef, 2019-03-15:
In the update process of the nc-update-nextcloud functionality a temporary backup is made. However, this backup is always made at `/var/www`. 
Feature request: Make the location of the backup directory configurable in the ncp-web panel like in e.g. nc-backup. 

If nextcloudpi is hosting a lot of of data and is hosted on a raspberry pi there is not that much disk space (SD card) left for all the data to backup on the SD card. Data is located on an external drive. So it would be very helpful to make the temporary backup of nc-update-nextcloud also on this external drive.

nachoparker, 2019-03-16:
Hi,

The backup performed by `nc-update-nextcloud` is dataless, so it shouldn't be too big.

closed 2019-03-16 by nachoparker

paschaef, 2019-03-18:
I am running a nextcloudpi docker instance and during the mentioned update process backup is failing due to disk size check... So updating with ncp-webpanel is not possible. 
Therefor this feature would be very appreciated.