nextcloud/nextcloudpi #846: Parts of the data directory can be included in dataless backups

theCalcaholic, 2019-03-16:
**Expected**  
When the *INCLUDEDATA* option is unticked in nc-backup, the data directory is excluded from backups

**Actual**
If - and only if - the data directory is still in the default location (/var/www/nextcloud/data), some subdirectories can be included in backups (e.g. those added by nc-apps).
For example on one of my ncp-setups (using docker), the `/var/www/nextcloud/data/__groupfolders` directory was included.

nachoparker, 2019-03-16:
nice new avatar :D 

because if you restore without data it will still work. If you exclude the whole data dir and restore NC will complain

closed 2019-03-19 by nachoparker

nero120, 2019-05-17:
@theCalcaholic I'd like to try and reduce the size of my "dataless" backups which are very large due to previewgenerator thumbnails - are you saying that if you move the datadir away from the default then backups won't include app data?

nachoparker, 2019-05-26:
App data is always backed up, but you could look into removing the thumbnails data, then you have to refresh with `ncc files:scan-app-data` IIRC