nextcloud/nextcloudpi #848: Can't run any update in web panel or via SSH (v0.67.11)

labeste, 2019-03-18:
Hello,

My Nextcloupi instance is stuck since several days - I had an error in my Nextcloud Android app saying "couldn't connect to Internet". I couldn't access the Pi in any way. 
I tried to access from SSH, and succeeded, but my ncp user wasn't in the "list of sudoers", so I couldn't update anything, or access some folders. I therefore cannot run "sudo ncp-report", for example.
After examination, I had to replace some files from Berryboot on the SD card with newer version, and "use existing files" to run only the Nextcloudpi web panel on the local ip address of the Raspberry. My system/Nextcloud files are on the WD external drive.
Every update ("ncp-update", "ncp-update-nextcloud", etc...) I'm trying to "Run" doesn't work : the red dot at the right of the "Run" button stays in red status, and no log is visible.

Here are the system info from the web panel for now :
------
System Info

‣ You should run Lets Encrypt for trusted encrypted access

‣ You should use nc-datadir to move your files to your plugged in USB drive

‣ You should enable automount to use your plugged in USB drive
NextCloudPi version	v0.67.11
NextCloudPi image	NextCloudPi_02-06-18
distribution	Raspbian GNU/Linux 9 \n \l
automount	no
USB devices	sda
datadir	/var/www/nextcloud/data
data in SD	yes
data filesystem	aufs
data disk usage	11G/344G
rootfs usage	11G/344G
swapfile	none
dbdir	(doesn't exist)
Nextcloud check	error
HTTPD service	up
PHP service	up
MariaDB service	up
Redis service	down
Postfix service	up
internet check	ok
port check 80	open
port check 443	open
IP	192.168.0.22
gateway	192.168.0.254
interface	eth0
certificates	none
NAT loopback	no
uptime	52min
------

Many thanks in advance for any help with these issues, and please forgive my bad english.

hurrikan77, 2019-03-19:
Have the same issue.

Due to that there is no possibility to connect to the Pi via SSH or web.

hurrikan77, 2019-03-19:
it seams that there is an error in one of the packages which were loaded due to unattended upgrade funtionality. (issue: 849)

Due to that there is an dhcpcd error after rebooting. So the pi have no ip after reboot.

I have done the apt-get update manually on an new copied image of ncp and got the following error message.

[image]

After apt-get upgrade and an reboot no access via ncp-web or ssh is possible because of missing ip.

Is there any possibility to recover the system?

labeste, 2019-03-19:
Not really the same issue as mine : I still have a local IP (visible above in system info) for my pi. But the "nextcloud check" indicates some error. So I cannot start it, I cannot try many of the solutions given here and there, because many of them are using the terminal and beginning usually with "sudo...". My ncp user isn't in the sudoers list, and I don't have any other user available...

nachoparker, 2019-03-19:
`ncp` user is not for SSH, on  the pi it is the `pi` user

@hurrikan77 I see no DHCP error, only some untrusted source, which is unrelated

labeste, 2019-03-19:
> `ncp` user is not for SSH, on the pi it is the `pi` user

thanks for your answer, but I didn't remember to initiate the pi user with some password, so there must be a default password, I suppose ?

I can access the pi with ncp user. But I can't execute any command line with sudo with it.

hurrikan77, 2019-03-20:
I tried to boot the system ones more and now it workes. Seems that something with the network was not ok.

labeste, 2019-03-21:
Against all odds, I finally managed to have a root access via SSH - really hard for me. So I can paste the `sudo ncp-report` result here, as my Nextcloud installation doesn't seem to update when I run `sudo ncp-update` or `sudo ncp-config`.
(I didn't include the Nextcloud logs, since they contain personal data)

<details>
<summary>NextCloudPi diagnostics</summary>

```

grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
Cannot load Zend OPcache - it was already loaded
PHP Fatal error:  Uncaught Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045]                                                                                             Access denied for user 'ncadmin'@'localhost' (using password: YES) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(D in /var/www/                                                                                            nextcloud/lib/private/DB/Connection.php on line 64
NextCloudPi version  v0.67.11
NextCloudPi image    NextCloudPi_02-06-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      aufs
data disk usage      11G/344G
rootfs usage         11G/344G
swapfile             none
dbdir                 (doesn't exist)
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               15min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'nc                                                                                            admin'@'localhost' (using password: YES) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\                                                                                            Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configura                                                                                            tion), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(630): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(118): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(117): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(165): OC\ServerContainer->query('OCP\\IDBConnecti...')
[... 88 lines cut ...]
```

2019-03-21 12:19:34 1988562944 [Note] InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of small buffer pool size. In o                                                                                            rder to use backoff, increase buffer pool at least up to 20MB.

2019-03-21 12:19:34 1988562944 [Note] InnoDB: Using mutexes to ref count buffer pool pages
[... cut ...]
2019-03-21 12:19:34 1988562944 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 4031174047
2019-03-21 12:19:34 1405088576 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-03-21 12:19:34 1988562944 [Note] Plugin 'FEEDBACK' is disabled.
2019-03-21 12:19:34 1988562944 [Note] Server socket created on IP: '127.0.0.1'.
2019-03-21 12:19:34 1988562944 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

labeste, 2019-03-23:
Any help ? What's more, trying different things with `ncp-config`, I disabled the webui by mistake, and now I can't enable it ! It says each time :

`grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
/usr/local/etc/library.sh: line 174: jq: command not found
/usr/local/etc/library.sh: line 175: jq: command not found
Running nc-webui
Site ncp already disabled
ncp-web disabled
Done. Press any key...
`

I tried to find a solution to the failed database connection, but didn't succeed. I added : 
```
[mysqld]
default_authentication_plugin=mysql_native_password
```

to: ` /etc/mysql/my.cnf` and then `systemctl start mariadb` and `systemctl start mysqld`, but it does nothing.

If I run `systemctl status mariadb` or `systemctl status mysqld`, I get the same output : 
```
mariadb.service - MariaDB 10.1.37 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-03-21 12:49:29 UTC; 1 day 22h ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
 Main PID: 1286 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─1286 /usr/sbin/mysqld

Mar 21 12:49:07 nextcloudpi systemd[1]: Starting MariaDB 10.1.37 database server...
Mar 21 12:49:29 nextcloudpi mysqld[1286]: 2019-03-21 12:49:29 1989271552 [Note] /usr/sbin/mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 1286 ...
Mar 21 12:49:29 nextcloudpi systemd[1]: Started MariaDB 10.1.37 database server.
```
It seems to me the more I try solutions by myself, the more I'm corrupting my instance... :-(

I really need to get my Nextcloudpi back again... Thanks for any help.

labeste, 2019-03-23:
Seeing this error many times : 
```
ncp-update
Downloading updates
Performing updates
./update.sh: line 74: jq: command not found
```
I installed `jq` : `sudo apt-get install jq`

Then I could perform ` sudo ncp-update` : 
```
Downloading updates
Performing updates
Installing letsencrypt
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Err:4 https://packages.sury.org/php stretch InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
Fetched 11.7 MB in 16s (726 kB/s)
Reading package lists... Done
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-minimal is already the newest version (3.5.3-1).
0 upgraded, 0 newly installed, 0 to remove and 36 not upgraded.
--2019-03-23 12:14:44--  https://github.com/letsencrypt/letsencrypt/archive/master/latest.tar.gz
Resolving github.com (github.com)... 140.82.118.3, 140.82.118.4
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://github.com/certbot/certbot/archive/master/latest.tar.gz [following]
--2019-03-23 12:14:44--  https://github.com/certbot/certbot/archive/master/latest.tar.gz
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/certbot/certbot/tar.gz/master [following]
--2019-03-23 12:14:45--  https://codeload.github.com/certbot/certbot/tar.gz/master
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120, 192.30.253.121
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘STDOUT’

-                                                               [      <=>                                                                                                                                ]   1.22M   976KB/s    in 1.3s

2019-03-23 12:14:46 (976 KB/s) - written to stdout [1279336]

mv: cannot move 'certbot-master' to 'letsencrypt/certbot-master': Directory not empty

```
I tried with `sudo ncp-config` and had the same output as above.

If anything has changed since my first `ncp-report`, here is the new one : 

<details>
<summary>NextCloudPi diagnostics</summary>

```

grep: /etc/mysql/mariadb.conf.d/90-ncp.cnf: No such file or directory
Cannot load Zend OPcache - it was already loaded
PHP Fatal error:  Uncaught Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
[... 135 lines cut ...]
```

2019-03-23 11:49:24 1995606832 [Note] InnoDB: innodb_empty_free_list_algorithm has been changed to legacy because of small buffer pool size. In order to use backoff, increase buffer pool at least up to 20MB.

2019-03-23 11:49:24 1995606832 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2019-03-23 11:49:24 1995606832 [Note] InnoDB: The InnoDB memory heap is disabled
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2019-03-23 11:49:24 1995606832 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Using Linux native AIO
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Using generic crc32 instructions
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Completed initialization of buffer pool
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Highest supported file format is Barracuda.
2019-03-23 11:49:24 1995606832 [Note] InnoDB: 128 rollback segment(s) are active.
2019-03-23 11:49:24 1995606832 [Note] InnoDB: Waiting for purge to start
2019-03-23 11:49:24 1995606832 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 4031174097
2019-03-23 11:49:24 1379922752 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-03-23 11:49:24 1995606832 [Note] Plugin 'FEEDBACK' is disabled.
2019-03-23 11:49:24 1995606832 [Note] Server socket created on IP: '127.0.0.1'.
2019-03-23 11:49:24 1995606832 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

nachoparker, 2019-03-23:
You have skipped updates, and now you are in a bad state. I suggest  you restore to a new instance, and enable autoupdates

closed 2019-03-23 by nachoparker

labeste, 2019-03-25:
...

labeste, 2019-03-26:
> You have skipped updates, and now you are in a bad state. I suggest you restore to a new instance, and enable autoupdates

@nachoparker : I did that. Now Nextcloudpi is stuck with this error : 
```
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
```
When doing `sudo -u www-data php /var/www/nextcloud/occ upgrade`, I have this exception : 
```
Set log level to debug
Exception: Updates between multiple major versions and downgrades are unsupported.
Update failed
Maintenance mode is kept active
Reset log level
```

Infos below, I have removed Nextcloud logs since they are too long for comment : 

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.10.6
NextCloudPi image    NextCloudPi_03-09-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sdb 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      aufs
data disk usage      19G/350G
rootfs usage         19G/350G
swapfile             none
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               18:09

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.0.22",
[... 81 lines cut ...]
2019-03-25 19:57:27 1969898304 [Note] /usr/sbin/mysqld: Shutdown complete

2019-03-25 19:58:02 1995816752 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2019-03-25 19:58:02 1995816752 [Note] InnoDB: The InnoDB memory heap is disabled
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2019-03-25 19:58:02 1995816752 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Using Linux native AIO
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Using generic crc32 instructions
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Initializing buffer pool, size = 336.0M
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Completed initialization of buffer pool
2019-03-25 19:58:02 1995816752 [Note] InnoDB: Highest supported file format is Barracuda.
2019-03-25 19:58:03 1995816752 [Note] InnoDB: 128 rollback segment(s) are active.
2019-03-25 19:58:03 1995816752 [Note] InnoDB: Waiting for purge to start
2019-03-25 19:58:03 1995816752 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 114781937
2019-03-25 19:58:03 1149236032 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-03-25 19:58:03 1995816752 [Note] Plugin 'FEEDBACK' is disabled.
2019-03-25 19:58:03 1995816752 [Note] Server socket created on IP: '127.0.0.1'.
2019-03-25 19:58:03 1995816752 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

Any idea of what's going on and how to resolve it ? Thanks.

labeste, 2019-03-26:
Resolved it partially using this fix : [https://mergy.org/2018/10/nextcloud-14-upgrade-error-updates-between-major-versions-and-downgrades-are-unsupported-fix/](url)

But after a reboot, I have "Nextcloud check : error" in system infos... All this is just a nightmare for the past several weeks. Nothing I can't rely on, just very disappointing.

labeste, 2019-03-26:
```
systemctl status mariadb.service
● mariadb.service - MariaDB 10.1.37 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/mariadb.service.d
           └─ncp-delay-automount.conf
   Active: activating (start-pre) since Tue 2019-03-26 14:39:51 GMT; 9s ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
 Main PID: 1579 (code=exited, status=1/FAILURE); Control PID: 1784 (sleep)
   CGroup: /system.slice/mariadb.service
           └─control
             └─1784 /bin/sleep 20

Mar 26 14:39:51 nextcloudpi systemd[1]: Starting MariaDB 10.1.37 database server...

```

And : 

`Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:64`

I'm tired of trying to get back my NextcloudPi.
Starting with version 0.67, I did some update of everything (NC/NCP/APPS) in november. 6 months later, everything is broke because of needed updates. I tried to repair : fail. Advice was given to me that I install a new instance, enable autoupdates. I restored my backup from november in it. Updates to do (NCP/NC) : did it. Nextcloud available at last. But : apps upgrade broken. Tried a fix. Apps to update. Rebooted. Database unavailable, Nextcloud unavailable. 

I'm sick of it. And please excuse my emotions, but since Nextcloud & NextcloudPi were such great solutions, and I'm not an advanced user with maintaining all this, just trying what I can with my linux notions...

labeste, 2019-03-26:
Restoring again my full (database + files) backup from november failed, when it succeeded yesterday : 
```
[ nc-restore ]
extracting backup file /media/USBdrive/ncdata/ncp/files/nextcloud-bkp_20181128.tar...
restore files...
restore database...
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "No such file or directory")
Cleanup...
```