nextcloud/nextcloudpi #849: Error: packages.sury.org/php stretch InRelease

Haraade, 2019-03-18:
Err:7 https://packages.sury.org/php stretch InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
Reading package lists... Done
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B188E2B695BD4743
W: Some index files failed to download. They have been ignored, or old ones used instead.

bytehead, 2019-03-21:
https://www.patreon.com/posts/dpa-new-signing-25451165

nachoparker, 2019-03-21:
thanks @bytehead will apply that fix

closed 2019-03-21 by nachoparker

bytehead, 2019-03-21:
You're welcome! 👍

sergey-shambir, 2021-03-18:
Hint: use `sudo apt-key adv --keyserver keys.gnupg.net --recv-keys B188E2B695BD4743` to update GPG key if local copy expired.

sergey-shambir, 2021-04-15:
Hint: the same solution helps when `apt-get update` warns about expired key:
```
Reading package lists... Done
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch http://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```