nextcloud/nextcloudpi #850: nc-scan-auto has stopped working after one of recent updates

smartfonreddit, 2019-03-19:
Hello and thanks for NCP!

I have 2 user accounts. Both are configured as local Network Shares on Windows 10 PC, and synced to Android mobile. I want to make changes to files on Network Share and use the nc-scan-auto at 5 minute interval to make the changes done on Windows 10 PC visible on Android within 5 minutes. This used to work few weeks ago, before the updates that introduced the extra path fields in nc-scan-auto.

Today I tried to set it up again but it doesn't do anything. If you search "pdf" in the error log below you'll see the errors it recorded when I was dealing with a pdf file. I uploaded the pdf from Android and it synced to Windows 10 network share instantly. Deleting on android would delete it from Windows 10 instantly. When I tried to manually delete it from network share, however, it went away on network share, but it wasn't deleted from Android, meaning the nc-scan-auto isn't functional.

nc-scan-auto is configured as YES , 5 minutes, paths are all default (blank?). I've also tried /ncp/ as the path (account username is ncp).

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.10.4
NextCloudPi image    NextCloudPi_01-14-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.1T/1.9T
rootfs usage         2.1G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.1.234",
            "11": "***REMOVED SENSITIVE VALUE***",
            "4": "***REMOVED SENSITIVE VALUE***",
            "20": "192.168.1.107",
            "21": "192.168.1.234",
            "22": "***REMOVED SENSITIVE VALUE***"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.5.3",
        "overwrite.cli.url": "https:\/\/192.168.1.234\/",
[... 89 lines cut ...]

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"XJB4fX8AAQEAAAQUVicAABgd","level":3,"time":"2019-03-19T05:05:02+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"PHP","method":"COPY","url":"\/remote.php\/dav\***REMOVED SENSITIVE VALUE***","message":"fopen(\/media\/USBdrive\/ncdata\/ncp\***REMOVED SENSITIVE VALUE***): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#284","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/73.0.3683.75 Safari\/537.36","version":"15.0.5.3"}
{"reqId":"XJB4fX8AAQEAAAQUVicAABgd","level":4,"time":"2019-03-19T05:05:02+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"webdav","method":"COPY","url":"\/remote.php\/dav\***REMOVED SENSITIVE VALUE***","message":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Could not open file","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Tree.php","line":307,"function":"get","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Tree.php","line":326,"function":"copyNode","class":"Sabre\\DAV\\Tree","type":"->","args":[{"__class__":"OCA\\DAV\\Connector\\Sabre\\File"},{"__class__":"OCA\\DAV\\Connector\\Sabre\\Directory"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Tree.php","line":130,"function":"copyNode","class":"Sabre\\DAV\\Tree","type":"->","args":[{"__class__":"OCA\\DAV\\Connector\\Sabre\\Directory"},{"__class__":"OCA\\DAV\\Connector\\Sabre\\Directory"},"_img"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":684,"function":"copy","class":"Sabre\\DAV\\Tree","type":"->","args":["files\/ncp\***REMOVED SENSITIVE VALUE***","files\/ncp\/New folder\/_img"]},{"function":"httpCopy","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/192.168.1.107\/remote.php\/dav\/files\/ncp\***REMOVED SENSITIVE VALUE***","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpCopy"],[{"absoluteUrl":"https:\/\/192.168.1.107\/remote.php\/dav\/files\/ncp\***REMOVED SENSITIVE VALUE***","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:COPY",[{"absoluteUrl":"https:\/\/192.168.1.107\/remote.php\/dav\/files\/ncp\***REMOVED SENSITIVE VALUE***","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/192.168.1.107\/remote.php\/dav\/files\/ncp\***REMOVED SENSITIVE VALUE***","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":378,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/73.0.3683.75 Safari\/537.36","version":"15.0.5.3"}
{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":378,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/73.0.3683.75 Safari\/537.36","version":"15.0.5.3"}
{"reqId":"XJB4AX8AAQEAAAQUVbYAABIK","level":3,"time":"2019-03-19T05:07:09+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"PHP","method":"DELETE","url":"\/remote.php\/dav\/trashbin\/ncp\/trash","message":"rmdir(\/media\/USBdrive\/ncdata\/ncp\/files_trashbin\/files): Directory not empty at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#112","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/73.0.3683.75 Safari\/537.36","version":"15.0.5.3"}
{"reqId":"XJB4AX8AAQEAAAQUVbYAABIK","level":3,"time":"2019-03-19T05:07:09+00:00","remoteAddr":"192.168.1.112","user":"ncp","app":"PHP","method":"DELETE","url":"\/remote.php\/dav\/trashbin\/ncp\/trash","message":"rmdir(\/media\/USBdrive\/ncdata\/ncp\/files_trashbin): Directory not empty at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#118","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/73.0.3683.75 Safari\/537.36","version":"15.0.5.3"}
[... cut ...] ent":"Mozilla\/5.0 (Android) ownCloud-android\/3.5.1","version":"15.0.5.3"}
{"reqId":"XJCHOH8AAQEAAAPOKRkAAAAI","level":3,"time":"2019-03-19T06:07:52+00:00","remoteAddr":"***REMOVED SENSITIVE VALUE***","user":"--","app":"PHP","method":"GET","url":"\/ocs\/v1.php\/cloud\/user?format=json","message":"RedisException: Redis server went away at \/var\/www\/nextcloud\/lib\/private\/Memcache\/Redis.php#54","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.5.1","version":"15.0.5.3"}

```
</details>

nachoparker, 2019-03-19:
please try to set it again from ncp-web, even if you already had it set it up before

smartfonreddit, 2019-03-19:
The ncp-webui section was showing enabled with an asterisk sign on its side in the TerminalUI, but the config inside said "no". I had to manually activate it from terminal for the webui at 4443 port to be available on desktop. I never disabled it, it must have been disabled by a prior update. This could be related to other settings not functioning?

I enabled nc-scan-audo on WebUI but it still doesn't work. On the web, below the fields, I can see how it periodically executes the command  [ nc-scan-auto ] every 5 minutes. Only the changes made on Android are visible on Windows 10 Network Share, but not vice versa.

**Here is the strange part.** Let's say I create a folder named "TEST" on Network Share. It's not visible on Android after the sync runs. But if I manually create a similarly named folder "TEST" on Android (while I already have one on Network Share), it tells me the folder already exists on Android. After this error, the folder TEST becomes visible on Android, along with its files, and they all work. In other words, the sync did actually work, but the folder wasn't visible until I triggered it by creating a new folder with a similar name.

This also applies to files. I can create lol.txt on Network Share and sync it. Nothing is visible on Android. I then manually create a file named lol.txt on Android. It says error, file name is taken by another file. Immediately after that, the lol.txt that I created on Network Share, with its content, becomes visible on Android. It's as if I have to "trigger" it to become visible, by creating a dummy file/folder on Android. 

Once lol.txt is present on both devices, however, deleting it on Network Share still doesn't delete it from Android, meaning the sync feature itself is still broken.

When I do this dummy folder trick, the folder that appears on Android shows the size as "Pending". It only populates it with xx GB when I open the folder (and all folders inside it).

Anything else I can do?

smartfonreddit, 2019-03-20:
Update: The nc-scan-auto still doesn't work, but the manual nc-scan function works from the WebUI.

nachoparker, 2019-03-21:
What are the contents of `/etc/cron.d/ncp-scan-auto` and `/usr/local/bin/ncp-scan-auto`?

It's better to test this with the Nextcloud web instead of Android.

smartfonreddit, 2019-03-21:
/etc/cron.d/ncp-scan-auto
`*/5  *  *  *  *  root /usr/local/bin/ncp-scan-auto`

/usr/local/bin/ncp-scan-auto
```
#!/bin/bash
(

  echo -e "\n[ nc-scan-auto ]"

  [[ "" != "" ]] && ncc files:scan -n -v -p ""
  [[ "" != "" ]] && ncc files:scan -n -v -p ""
  [[ "" != "" ]] && ncc files:scan -n -v -p ""

  [[ "" == "" ]] && ncc files:scan -n -v --all

) 2>&1 >>/var/log/ncp.log

```

Tested it with the web version to bypass Android, too. Scan interval is set at 5 minutes at the moment.

smartfonreddit, 2019-03-21:
I'm no programmer so accept this as a newb's detective work.

Part of /usr/local/bin/ncp-scan-auto 
`[[ "" == "" ]] && ncc files:scan -n -v --all`

vs part of /usr/local/bin/ncp-scan 
`sudo -u www-data php occ files:scan --all`

Is "ncc" essentially telling it to execute parts of "occ"? If so, could this be a problem with ncc? 
The nc-scan uses occ and it works. The nc-scan-auto uses ncc and it fails.

I found this from 5 days ago:
https://github.com/nextcloud/nextcloudpi/issues/845
However, when I type "ncc -h", the terminal displays the help document for ncc successfully.

nachoparker, 2019-03-22:
Everything looks good, please try now to run the command directly

```
sudo /usr/local/bin/ncp-scan-auto
```

smartfonreddit, 2019-03-22:
`sudo /usr/local/bin/ncp-scan-auto`

I ran it twice and both times it synced properly. The web and Android can see the files and the folder sizes. 
The second time I ran the command, it replied **You have new mail in /var/mail/root**.

nachoparker, 2019-03-22:
it seems like everything works. You should see the scan results everyday on `/var/log/ncp.log`

Thanks for the help

closed 2019-03-22 by nachoparker

smartfonreddit, 2019-03-22:
The nc-scan-auto only worked as a one-time sync (just like nc-scan) when I ran the command you gave me **sudo /usr/local/bin/ncp-scan-auto**. It still doesn't sync automatically at the pre-configured intervals.

The /var/log/ncp.log says 

```
[ nc-update-nc-apps-auto ]
checking for updates...
[ncp-update-nc]
Current   Nextcloud version 15.0.5.3
Available Nextcloud version 15.0.5
Nothing to update

[ nc-scan-auto ]
automatic scans enabled

[ nc-scan-auto ]
```

nachoparker, 2019-03-22:
you should be seeing `[ nc-scan-auto ]` every five minutes, or whatever interval you selected. Is that what you see?

smartfonreddit, 2019-03-22:
Yes, the line appear every x minutes, but the file changes don't show up. It only worked as a single-instance sync (as in nc-scan) when I manually ran the command `sudo /usr/local/bin/ncp-scan-auto`

`nc-scan` manual execution works every time.
`sudo /usr/local/bin/ncp-scan-auto` execution on terminal acts as nc-scan, only syncs once, then stops. I can manually execute that command 100 times and it will sync 100 times, but it never syncs on its own every x minutes afterwards.
`nc-scan-auto` from the WebUI and TerminalUI doesn't do anything.

Could be cron job scheduling problem?

nachoparker, 2019-03-23:
Well, then check cron, maybe something is wrong there

`journalctl -u cron` and `var/mail/root`

smartfonreddit, 2019-03-23:
Alright, I think we're onto something. I have the same log every 5 minutes (the scan interval) in /var/mail/root

```
From root@raspberrypi  Fri Mar 22 17:20:01 2019
Return-Path: <root@raspberrypi>
X-Original-To: root
Delivered-To: root@raspberrypi
Received: by raspberrypi (Postfix, from userid 0)
	id 786591FD68; Fri, 22 Mar 2019 17:20:01 -0700 (PDT)
From: root@raspberrypi (Cron Daemon)
To: root@raspberrypi
Subject: Cron <root@nextcloudpi> /usr/local/bin/ncp-scan-auto
MIME-Version: 1.0
Content-Type: text/plain; charset=UTF-8
Content-Transfer-Encoding: 8bit
X-Cron-Env: <SHELL=/bin/sh>
X-Cron-Env: <HOME=/root>
X-Cron-Env: <PATH=/usr/bin:/bin>
X-Cron-Env: <LOGNAME=root>
Message-Id: <20190323002001.786591FD68@raspberrypi>
Date: Fri, 22 Mar 2019 17:20:01 -0700 (PDT)

/usr/local/bin/ncp-scan-auto: line 10: ncc: command not found
```

**ncc: command not found**. I found this other post https://github.com/nextcloud/nextcloudpi/issues/506

`ls -l /usr/local/bin/ncp-scan-auto` in my case also shows
`-rwxr-xr-x 1 root staff 263 Mar 22 10:18 /usr/local/bin/ncp-scan-auto`

The line 10 of `/usr/local/bin/ncp-scan-auto` says
`  [[ "" == "" ]] && ncc files:scan -n -v --all`

In my earlier post I also suspected a broken ncc command to be the culprit. Any way to test?

nachoparker, 2019-03-23:
Ok, we found it! working  on a fix

nachoparker, 2019-03-23:
Can you try it out and confirm if the fix work please? Just do `sudo ncp-update devel` and try again.

smartfonreddit, 2019-03-23:
Works beautifully now. Thanks for the quick fix!

nachoparker, 2019-03-23:
awesome! thanks for the help