nextcloud/nextcloudpi #853: Issue with webdav and data dir in browser

devilsharp, 2019-03-20:
### Installation architecture

System A --> Nginx reverse proxy, with strong TLS_AES_256_GCM_SHA384, TLS 1.3 encryption
System B --> Nextcloudpi reachable through the System A
System C --> Jeedom reachable through the System A

### System A configuration

**/etc/nginx/conf.d/proxy.conf**

```
proxy_redirect off;
proxy_set_header Host $host;
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_hide_header Strict-Transport-Security;
proxy_hide_header X-Powered-By;
proxy_hide_header *;
proxy_intercept_errors on;
proxy_buffering on;
proxy_cache_key "$scheme://$host$request_uri";
proxy_cache_path /var/cache/nginx levels=1:2 keys_zone=cache:10m inactive=7d max_size=700m;
```

**/etc/nginx/sites-enabled/systemB.vhost**

```
server {
    listen 80;
    server_name example.com;
    return 301 https://$host$request_uri;

location '/.well-known/acme-challenge' {
    root /var/www/example.com/;
    try_files $uri /$1;
    }
}

server {
    listen      443 http2;
    server_name  example.com;
    ssl on;
    ssl_certificate /etc/certbot/live-ecdsa/example.com/chain.pem;
    ssl_certificate_key /etc/certbot/live-ecdsa/example.com/privkey-p384.pem;
    if ($request_method !~ ^(GET|HEAD|POST)$ ) {
    return 444;
    }

location '/.well-known/acme-challenge' {
    root /var/www/example.com/;
    try_files $uri /$1;
    }

location / {
    proxy_pass http://192.168.0.44/;
    proxy_cache cache;
    proxy_cache_valid 12h;
    expires 12h;
    proxy_cache_use_stale error timeout invalid_header updating;
    }
}
```

### System B configuration

**/etc/apache2/sites-enabled/nextcloud.conf**

  ```
<VirtualHost _default_:80>
    DocumentRoot /var/www/nextcloud
[... 183 lines cut ...]
```

{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): 0px) 100vw, 300px\" \/><p>Seulement 8&euro;&nbsp;pour cette micro SD de 32Go&nbsp; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 1: parser error : Entity 'euro' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): \/><p>Seulement 8&euro;&nbsp;pour cette micro SD de 32Go&nbsp;au lieu de 12&euro; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 1: parser error : Entity 'agrave' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): ur cette micro SD de 32Go&nbsp;au lieu de 12&euro; habituellement ! Voil&agrave; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 1: parser error : Entity 'eacute' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): 32Go&nbsp;au lieu de 12&euro; habituellement ! Voil&agrave; une bonne id&eacute; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 1: parser error : Entity 'nbsp' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): gmenter le stockage de votre raspberry pi &#128293;&#128293; voir la promo&nbsp; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 2: parser error : Entity 'euro' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): -sandisk-32-domotique-deal-promo\/\">La carte micro SD 32Go pour seulement 8&euro; at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string(): Entity: line 2: parser error : Entity 'eacute' not defined at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
[... cut ...] /var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}
{"reqId":"APFekbrxZlDetyzhEZpy","level":3,"time":"2019-03-20T22:45:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"\/cron.php","message":"simplexml_load_string():                                                                                ^ at \/var\/www\/nextcloud\/apps\/news\/lib\/Fetcher\/FeedFetcher.php#213","userAgent":"--","version":"15.0.5.3"}

```
</details>

nachoparker, 2019-03-21:
Hi,

Please, use the forums for support. This doesn't seem like a NCP issue

Thanks!

closed 2019-03-21 by nachoparker

devilsharp, 2019-03-21:
Hi @nachoparker,

Yes indeed, it was not a NCP issue. the issue was located in my vhost file in reverse proxy side:

**/etc/nginx/sites-enabled/systemB.vhost**
```
    if ($request_method !~ ^(GET|HEAD|POST)$ ) {
    return 444;
    }
```

By remove/comment this block, I am able to access to my files :)