nextcloud/nextcloudpi #855: nc-update-nextcloud cannot skip major versions

nachoparker, 2019-03-24:
It's better to check this before trying to update

```
Exception: Updates between multiple major versions and downgrades are unsupported.
Update failed
```

labeste, 2019-03-26:
Trying to fix this leads to major problems, I can confirm...

closed 2021-10-15 by nachoparker

labeste, 2021-10-15:
Well done. Unfortunately, 2 and a half year later, I abandoned NextcloudPi as a reliable and user-friendly solution...

nachoparker, 2021-10-15:
You know, there is no need to be passive aggressive with people that just try to help others in their free time. This is a volunteer activity, and it works by everyone helping a bit.

Have a good day.

labeste, 2021-10-16:
Well, I'm not passive aggressive, actually - or tell me where ? I'm just giving an user feedback.
I'm not a pro, I'm not either a newbie on Pi hardware or Nextcloud-related matter. I tried my best to use NextcloudPi following the documentation, which was sparse and not really updated at the time I had this annoying bug, if I remember well.
I tried hard to resolve it, really hard. And, as the user I described above, I did what could help me and help others : open an issue. And I waited, waited... while my files and data were still unaccessible. Then, you answered me, laconically, but that didn't help. That was a bitter experience. Don't be surprised if I abandoned NextcloudPi, two and a half year later. Maybe you should promote NCPi in order not to be alone developing it. Maybe you should (have) focus(ed) on end-user development and stability, I don't know. Maybe there's no need to be passive developing some not-so-obscure-not-so-technical-software, supposed to be used by not-so-next-door-people, if you cannot organize the project to make sure you'll have others developers caring about end-users critical issues, or people to update documentation.
In the end, it's open-source, it's free, you're volunteer, so I guess I should not give feedback. These days, some people make a living doing some UX research because, hey, developers cannot do it all, although they seem to think differently sometimes...
Don't worry too much about this, you probably have by now plenty of NextcloudPi users ready to help ;-)

Have a nice day too.