nextcloud/nextcloudpi #861: Can't set custom data dir when running fresh docker image

nero120, 2019-03-26:
If you try to run the latest docker ownyourbits/nextcloudpi-armhf image with a custom folder set as the data volume, NCP fails to initialise.

In my case I'm running on a fresh docker installation (no existing images, containers or volumes), and replacing `ncdata` with `/media/data`:

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/data:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf
```

When I watch the init logs I see the following:

```
Starting PHP-fpm
Starting Apache
Starting mariaDB
2019-03-26 13:00:23 1995549488 [Note] mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 88 ...
2019-03-26 13:00:23 1995549488 [Warning] Can't create test file /data/database/2d33b9d29db1.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2019-03-26 13:00:23 1995549488 [ERROR] Aborting
```

If I change back to `-v ncdata:/data` it initialises fine.

nachoparker, 2019-03-26:
It should work. Please, clean the container and the volume and paste the full output

nero120, 2019-03-27:
```
$ docker stop nextcloudpi
nextcloudpi
$ docker container prune
WARNING! This will remove all stopped containers.
Are you sure you want to continue? [y/N] y
Deleted Containers:
cda4c4cda663d737b644ddc95695bafcf4203d8afb69b833c409ca1353912be6

Total reclaimed space: 2.22kB
$ docker volume prune
WARNING! This will remove all local volumes not used by at least one container.
Are you sure you want to continue? [y/N] y
Total reclaimed space: 0B
$ ls -l /media
total 36
drwxr-xr-x 1 root root  268 Mar  7 06:22 backups
drwxr-xr-x 1 root root   48 Mar 26 12:37 data
drwxrwxrwx 6 rich rich 4096 Mar 26 13:00 data_test
$ docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /media/data_test:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf
fb72734a41937713684767b2f3ba6b21ae88c3c577ce4ecef0daf139f85fff67
$ docker logs nextcloudpi
Starting PHP-fpm
Starting Apache
Starting mariaDB
2019-03-26 20:51:02 1995442992 [Note] mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 81 ...
2019-03-26 20:51:02 1995442992 [Warning] Can't create test file /data/database/fb72734a4193.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2019-03-26 20:51:02 1995442992 [ERROR] Aborting
```

ovpc, 2019-03-27:
@nero120 
Seems like a permission problem. Make sure /media/data is writable by the docker user or group.

Presumably;

> sudo chown rich:docker /media/data

should do the trick, if "rich" is the user, launching the container.

nero120, 2019-03-27:
@ovpc thank you that seems to have done the trick. I didn't think it would be permissions as I had set the volume folder to `777` so figured docker would be able to read/write without issue.

nero120, 2019-03-27:
@nachoparker after following @ovpc's advise above, I'm getting a related issue. After "Init done" is reported in the log, I attempt to load nextcloud but get a 403 Forbidden error in the browser. Checking permissions on the `/media/data_test` dir shows:

```
drwxr-xr-x 7 root root 4096 Mar 27 00:13 data_test
```

I don't know why the `docker` group has been removed and it's been set to `755`, however changing it back to `root:docker 775` and restarting the container does not resolve the issue.

For info, the init log contains:
```
Initializing empty volume..
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/services-enabled.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Starting mariaDB
2019-03-27  0:13:45 1995791152 [Note] mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 89 ...
Starting Redis
Starting Cron
Starting Postfix
Uninitialized instance, running nc-init...
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 11 set to string 90.243.100.226
System config value trusted_domains => 1 set to string 172.17.0.2
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
nextcloudpi enabled
calendar installed
calendar enabled
contacts installed
contacts enabled
notes installed
notes enabled
tasks installed
tasks enabled
news installed
news enabled
previewgenerator installed
previewgenerator enabled
calendar enabled
contacts enabled
notes enabled
tasks enabled
news enabled
previewgenerator enabled
Config value squareSizes for app previewgenerator set to 32
Config value widthSizes for app previewgenerator set to 128 256 512
Config value heightSizes for app previewgenerator set to 128 256
System config value jpeg_quality set to string 60
System config value overwriteprotocol set to string https
Following columns will be updated:

* filecache.mtime
* filecache.storage_mtime

This can take up to hours, depending on the number of files in your instance!
NC init done
System config value trusted_domains => 1 set to string 172.17.0.2
System config value trusted_domains => 6 set to string nero120.cloud
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 768138240B for PHP
Using 4 PHP threads
Using 409673728 memory for the database
Stopping MariaDB database server: mysqld.
Starting MariaDB database server: mysqld ..
Stopping redis-server: redis-server.
Starting redis-server: redis-server.
Init done
stopping Cron...
stopping Redis...
stopping Postfix...
stopping logs...
Stopping apache
Stopping PHP-fpm
Stopping mariaDB
LAMP cleanup complete
Starting PHP-fpm
Starting Apache
Starting mariaDB
2019-03-27  0:45:02 1996045104 [Note] mysqld (mysqld 10.1.37-MariaDB-0+deb9u1) starting as process 83 ...
```

I've checked the logs in the container but can't see anything obvious.

nextcloud.log:
```
{"reqId":"K7ZkuW1oISx8HLAeh7e1","level":3,"time":"2019-03-27T00:15:01+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"\/data\/nextcloud\/lib\/base.php","line":660,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"\/data\/nextcloud\/lib\/base.php","line":1068,"function":"init","class":"OC","type":"::","args":[]},{"file":"\/data\/nextcloud\/cron.php","line":41,"args":["\/data\/nextcloud\/lib\/base.php"],"function":"require_once"}],"File":"\/data\/nextcloud\/lib\/base.php","Line":277,"CustomMessage":"--"},"userAgent":"--","version":"15.0.5.3"}
```

Any ideas?

nero120, 2019-03-27:
> After "Init done" is reported in the log, I attempt to load nextcloud but get a 403 Forbidden error in the browser.

So my issue was caused by the fact that access to the `/var/www/ncp-web/` folder is restricted in the apache activation conf to local IPs only. I'm doing this over the web and therefore the 403.

I didn't see any mention of this in the [docs](http://docs.nextcloudpi.com/en/latest/) so might be good to add to avoid any future confusion.

closed 2019-03-27 by nero120

varhub, 2019-11-14:
Same issue there with following docker-compose:
```
version: '3'
services:
  nextcloudpi:
    restart: unless-stopped
    image: ownyourbits/nextcloudpi-armhf:v1.15.2
    command: localhost
    ports:
     - "80:80"
     - "443:443"
    volumes:
     - nc_data:/data
     - nc_userdata:/data/app/data
     - nc_backups:/backups

volumes:
  nc_data:
    driver: local
  nc_userdata:
    driver: local
  nc_backups:
    driver: local

```

It seems that database directory does not exist at this point of bootstrap:
```
root@rpi3-nc:/var/lib/docker/volumes# ls -l nextcloudpi2_nc_data/_data/
total 16
drwxr-xr-x 3 root root     4096 nov 13 23:29 app
drwxr-xr-x 3 root root     4096 jul 20 22:11 bin
drwxr-xr-x 9 root root     4096 nov 13 23:29 etc
drwxr-xr-x 3 root www-data 4096 jul 20 22:12 ncp
```

As hint, error raises due **missing** initialization of volume:
* Without volumes:
```
docker-compose up
Recreating nextcloudpi2_nextcloudpi_1 ... 
Recreating nextcloudpi2_nextcloudpi_1 ... done
Attaching to nextcloudpi2_nextcloudpi_1
nextcloudpi_1  | Initializing empty volume..  <--- HERE IS THE PROBLEM
nextcloudpi_1  | Making /usr/local/etc/ncp-config.d persistent ...
nextcloudpi_1  | Making /etc/services-enabled.d persistent ...
...
nextcloudpi_1  | Init done
```

* WIth volumes
```
pi@rpi3-nc:~/git/nextcloudpi2 $ docker-compose up
Creating network "nextcloudpi2_default" with the default driver
Creating volume "nextcloudpi2_nc_backups" with local driver
Creating volume "nextcloudpi2_nc_userdata" with local driver
Creating volume "nextcloudpi2_nc_data" with local driver
Creating nextcloudpi2_nextcloudpi_1 ... 
Creating nextcloudpi2_nextcloudpi_1 ... done
Attaching to nextcloudpi2_nextcloudpi_1
nextcloudpi_1  | Making /usr/local/etc/ncp-config.d persistent ...
nextcloudpi_1  | 2019-11-13 23:29:39 0 [Note] mysqld (mysqld 10.3.15-MariaDB-1) starting as process 103 ...
...
nextcloudpi_1  | 2019-11-13 23:29:39 0 [Warning] Can't create test file /data/database/1d27c3bb2054.lower-test
nextcloudpi_1  | mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
nextcloudpi_1  | 2019-11-13 23:29:39 0 [ERROR] Aborting
```

For my setup, root cause is that **nested volume** is fucking your test of empty volume:
```
# if an empty volume is mounted to /data, pre-populate it
[[ $( ls -1A /data | wc -l ) -eq 0 ]] && { echo "Initializing empty volume.."; cp -raT /data-ro /data; }
```

So we are not allowed to map `/data/app/data` or even user files at first-run, are we?
@nachoparker Did I fall down the cliff of the edge's case?

daschmidt1994, 2020-08-16:
docker logs -f nextcloudpi
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/services-enabled.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-08-16 17:54:00 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 230 ...
2020-08-16 17:54:00 0 [Warning] Can't create test file /data/database/e0642dabf350.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2020-08-16 17:54:00 0 [ERROR] Aborting

same issu if i start the Docker with own directory 
docker run -d -p 4443:4443 -p 32768:443 -p 32769:80 -v /volume1/nextcloud:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 192.168.178.200

daschmidt1994, 2020-08-18:
Same issue when i start on a Synology I get this error 
docker logs -f nextcloudpi
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-08-18  6:19:02 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 222 ...
2020-08-18  6:19:02 0 [Warning] Can't create test file /data/database/bacd9b9aaae3.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2020-08-18  6:19:02 0 [ERROR] Aborting

the folder permissions are 
ls -all /volume1/nextcloud/ncdata
total 0
drwxr-xr-x+ 1 root root 672 Aug 18 08:21 .
drwxr-xr-x+ 1 root root 336 Aug 18 08:06 ..

varhub, 2021-08-02:
@daschmidt1994 @JonasDaWi , you are using host's directories instead of volumes. So try with volumes. Otherwise proper permission setup is your responsibility (look for docker doc).

Fooughhy, 2022-01-08:
> @daschmidt1994 @JonasDaWi , you are using host's directories instead of volumes. So try with volumes. Otherwise proper permission setup is your responsibility (look for docker doc).

Sorry, but you say that as if that should be clear. Everywhere in this thread they mention volumes ("-v /volume1/nextcloud/ncdata:/data", "# if an empty volume is mounted to /data, pre-populate it") so it is not clear (to me, at least) that volumes are actually not used. But could you, or anyone, help with how to set up a folder on an external drive for NCP data then? 

I am also assuming that the settings in `/etc/fstab` matter alot here as well. 

`UUID=3A0AF83A0XXXXXXX	/mnt/mediadrive	ntfs	defaults,auto,users,rw,nofail,noatime	0	0` are the settings for my external drive. I have added "exec" as well, but haven't restarted yet to try if it would help. I have a few difficulties understanding how the parameters work here, more specifically the "users" parameter. Is that a usergroup (I have that with two users, current user and "pi"). I suppose that would make it more difficult for e.g `www-data` to do things on the drive, as that user is not in "users" group.