nextcloud/nextcloudpi #862: md5sum for image download doesn't match

05nelsonm, 2019-03-27:
Downloading from https://ownyourbits.com/downloads/NextCloudPi_RPi_03-09-19/

Tried both direct download from server and the torrent.

hashing the md5sum results in:
707634c1ccbe7ca4a56202b836e4a6e6  md5sum
f0d1d181353491a10f555d662747d0bd  NextCloudPi_RPi_03-09-19.tar.bz2

ovpc, 2019-03-27:
Better try downloading again, my torrent download shows:
>  $ cat md5sum 
f0d1d181353491a10f555d662747d0bd  torrent/NextCloudPi_RPi_03-09-19/NextCloudPi_RPi_03-09-19.tar.bz2

and
> $ md5sum NextCloudPi_RPi_03-09-19.tar.bz2 
f0d1d181353491a10f555d662747d0bd  NextCloudPi_RPi_03-09-19.tar.bz2

This image has been tested and reported to work.

05nelsonm, 2019-03-27:
Downloaded and tested again. Got what you got. No clue what was up. Closing issue, ty for verifying.

closed 2019-03-27 by 05nelsonm