nextcloud/nextcloudpi #870: Installation on Turris Omnia

johnnypea, 2019-04-05:
When I try to install on Turris Omnia but it fails when installing "fail2ban".

> Linux version 4.4.176-7bc33afbb1b35f5830b2b1b42c9cd8a0-0 (beast@build) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r47055) )

```
Setting up fail2ban (0.9.6-2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/fail2ban.service → /lib/systemd/system/fail2ban.service.
Job for fail2ban.service failed because the control process exited with error code.
See "systemctl status fail2ban.service" and "journalctl -xe" for details.
invoke-rc.d: initscript fail2ban, action "start" failed.
● fail2ban.service - Fail2Ban Service
   Loaded: loaded (/lib/systemd/system/fail2ban.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2019-04-05 11:58:45 UTC; 22ms ago
     Docs: man:fail2ban(1)
  Process: 22533 ExecStart=/usr/bin/fail2ban-client -x start (code=exited, status=255)

Apr 05 11:58:45 LXC_NAME systemd[1]: fail2ban.service: Control process exit…=255
Apr 05 11:58:45 LXC_NAME systemd[1]: Failed to start Fail2Ban Service.
Apr 05 11:58:45 LXC_NAME systemd[1]: fail2ban.service: Unit entered failed …ate.
Apr 05 11:58:45 LXC_NAME systemd[1]: fail2ban.service: Failed with result '…de'.
Hint: Some lines were ellipsized, use -l to show in full.
dpkg: error processing package fail2ban (--configure):
 subprocess installed post-installation script returned error exit status 1
Setting up whois (5.2.17~deb9u1) ...
Processing triggers for systemd (232-25+deb9u9) ...
Errors were encountered while processing:
 fail2ban
E: Sub-process /usr/bin/dpkg returned an error code (1)
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
bin/ncp/CONFIG/nc-init.sh: line 73: ncc: command not found
```

<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncp-report: line 42: column: command not found
cat: /usr/local/etc/ncp-version: No such file or directory

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Fri Apr 05 11:52:10.630034 2019] [mpm_event:notice] [pid 4949:tid 3069635056] AH00489: Apache/2.4.25 (Debian) configured -- resuming normal operations
[Fri Apr 05 11:52:10.630386 2019] [core:notice] [pid 4949:tid 3069635056] AH00094: Command line: '/usr/sbin/apache2'
[Fri Apr 05 11:57:07.099432 2019] [mpm_event:notice] [pid 4949:tid 3069635056] AH00491: caught SIGTERM, shutting down
[Fri Apr 05 11:57:07.307065 2019] [mpm_event:notice] [pid 19843:tid 3069303280] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured -- resuming normal operations
[Fri Apr 05 11:57:07.307170 2019] [core:notice] [pid 19843:tid 3069303280] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2019-04-05 11:56:44 3070122064 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Using Linux native AIO
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Using generic crc32 instructions
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Initializing buffer pool, size = 256.0M
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Completed initialization of buffer pool
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Highest supported file format is Barracuda.
2019-04-05 11:56:44 3070122064 [Warning] InnoDB: Resizing redo log from 2*3072 to 2*2048 pages, LSN=1616737
2019-04-05 11:56:44 3070122064 [Warning] InnoDB: Starting to delete and rewrite log files.
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Setting log file ./ib_logfile101 size to 32 MB
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Setting log file ./ib_logfile1 size to 32 MB
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Renaming log file ./ib_logfile101 to ./ib_logfile0
2019-04-05 11:56:45 3070122064 [Warning] InnoDB: New log files created, LSN=1616908
2019-04-05 11:56:45 3070122064 [Note] InnoDB: 128 rollback segment(s) are active.
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Waiting for purge to start
2019-04-05 11:56:45 3070122064 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616737
2019-04-05 11:56:45 2311058240 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-04-05 11:56:45 3070122064 [Note] Plugin 'FEEDBACK' is disabled.
2019-04-05 11:56:45 3070122064 [Note] Server socket created on IP: '127.0.0.1'.
2019-04-05 11:56:45 3070122064 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.6

```
<details>
<summary>NextCloudPi diagnostics</summary>

```

/usr/local/bin/ncp-report: line 42: column: command not found
cat: /usr/local/etc/ncp-version: No such file or directory

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "config" namespace.

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Fri Apr 05 11:52:10.630034 2019] [mpm_event:notice] [pid 4949:tid 3069635056] AH00489: Apache/2.4.25 (Debian) configured -- resuming normal operations
[Fri Apr 05 11:52:10.630386 2019] [core:notice] [pid 4949:tid 3069635056] AH00094: Command line: '/usr/sbin/apache2'
[Fri Apr 05 11:57:07.099432 2019] [mpm_event:notice] [pid 4949:tid 3069635056] AH00491: caught SIGTERM, shutting down
[Fri Apr 05 11:57:07.307065 2019] [mpm_event:notice] [pid 19843:tid 3069303280] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured -- resuming normal operations
[Fri Apr 05 11:57:07.307170 2019] [core:notice] [pid 19843:tid 3069303280] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2019-04-05 11:56:44 3070122064 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Using Linux native AIO
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Using generic crc32 instructions
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Initializing buffer pool, size = 256.0M
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Completed initialization of buffer pool
2019-04-05 11:56:44 3070122064 [Note] InnoDB: Highest supported file format is Barracuda.
2019-04-05 11:56:44 3070122064 [Warning] InnoDB: Resizing redo log from 2*3072 to 2*2048 pages, LSN=1616737
2019-04-05 11:56:44 3070122064 [Warning] InnoDB: Starting to delete and rewrite log files.
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Setting log file ./ib_logfile101 size to 32 MB
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Setting log file ./ib_logfile1 size to 32 MB
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Renaming log file ./ib_logfile101 to ./ib_logfile0
2019-04-05 11:56:45 3070122064 [Warning] InnoDB: New log files created, LSN=1616908
2019-04-05 11:56:45 3070122064 [Note] InnoDB: 128 rollback segment(s) are active.
2019-04-05 11:56:45 3070122064 [Note] InnoDB: Waiting for purge to start
2019-04-05 11:56:45 3070122064 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616737
2019-04-05 11:56:45 2311058240 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-04-05 11:56:45 3070122064 [Note] Plugin 'FEEDBACK' is disabled.
2019-04-05 11:56:45 3070122064 [Note] Server socket created on IP: '127.0.0.1'.
2019-04-05 11:56:45 3070122064 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.6

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

ovpc, 2019-04-05:
Thank you for using [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for installation help.

closed 2019-04-05 by ovpc

nachoparker, 2021-09-29:
added experimental support for LXC/LXD

https://github.com/nextcloud/nextcloudpi/commit/212f4e01325978b9bd937171814e564164c5f4af

there will be an image next release