nextcloud/nextcloudpi #871: Some problem updating to latest version of nextcloud

Ricardosgeral, 2019-04-05:
Hi

I've tried to update to latest nextcloud version in the webbrowser.

Got an error in the phase of moving files and could not get out of there.

Them i made the latest update of nextcloudpi in putty.

Now when i try to access the cloud in the webpage i get only "Update in process."

tried to turn off maintnance mode 
sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off

but this did not work

Any ideas?
 

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.10.11
NextCloudPi image    NextCloudPi_12-04-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      1.6T/2.8T
rootfs usage         2.9G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
[... 103 lines cut ...]
```

{"reqId":"XIx7mn8AAQEAAHBX4joAAABT","level":3,"time":"2019-03-16T04:29:19+00:00","remoteAddr":"80.82.77.33","user":"--","app":"PHP","method":"GET","url":"\/sitemap.xml","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocup3agyjd4f\/css\/theming\/ca9f-ae5b-theming.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"--","version":"15.0.5.3"}
{"reqId":"XIx7nH8AAQEAAHBX4jsAAABO","level":3,"time":"2019-03-16T04:29:19+00:00","remoteAddr":"80.82.77.33","user":"--","app":"PHP","method":"GET","url":"\/.well-known\/security.txt","message":"unlink(\/media\/myCloudDrive\/ncdata\/appdata_ocup3agyjd4f\/css\/updatenotification\/35c3-a617-admin.css): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#227","userAgent":"--","version":"15.0.5.3"}
{"reqId":"XJyEAH8AAQEAAHlKPH4AAAAU","level":3,"time":"2019-03-28T08:21:20+00:00","remoteAddr":"109.49.139.60","user":"carla","app":"no app in context","method":"PUT","url":"\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 1776489 got 8192","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":39,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["0000000000000000-0000000001776489",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["0000000000000000-0000000001776489",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpPut"],[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:PUT",[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":217,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.5.1","version":"15.0.5.3"}
{"reqId":"XJyEAH8AAQEAAHlKPH4AAAAU","level":4,"time":"2019-03-28T08:21:20+00:00","remoteAddr":"109.49.139.60","user":"carla","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 1776489 got 8192","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":39,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["0000000000000000-0000000001776489",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["0000000000000000-0000000001776489",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpPut"],[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:PUT",[{"absoluteUrl":"https:\/\/ricarcloud.duckdns.org\/remote.php\/dav\/uploads\/carla\/10befbe006a9d3a66e14c735ba336f45\/0000000000000000-0000000001776489","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var [... cut ...] "line":95,"function":"loadCommands","class":"OC\\Console\\Application","type":"->","args":[{"__class__":"Symfony\\Component\\Console\\Input\\ArgvInput"},{"__class__":"Symfony\\Component\\Console\\Output\\ConsoleOutput"}]},{"file":"\/var\/www\/nextcloud\/occ","line":11,"args":["\/var\/www\/nextcloud\/console.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/files_trashbin\/appinfo\/app.php","Line":31,"CustomMessage":"--"},"userAgent":"--","version":"15.0.5.3"}

```
</details>

Ricardosgeral, 2019-04-05:
I get this trying to update manually

pi@nextcloudpi:~ $ sudo -u www-data php /var/www/nextcloud/occ upgrade                                                    Nextcloud or one of the apps require upgrade - only a limited number of commands                                           are available
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Updating database schema
Updated database
UnexpectedValueException: The files of the app "files" were not correctly replac                                          ed before running the update
Update failed
Maintenance mode is kept active
Reset log level

nachoparker, 2019-04-05:
This seems to be a Nextcloud issue. Please use the forums for NC help.

help.nextcloud.com

It is recommended to use `nc-update-nextcloud` to upgrade NC in nextcloudpi

closed 2019-04-05 by nachoparker