nextcloud/nextcloudpi #872: letsencrypt fails

theCalcaholic, 2019-04-05:
Letsencrypt fails renewing the certificate with the following error, however it succeeds when stopping apache and performing a standalone renewal:

       Failed authorization procedure. wolke.knoeppler.org (http-01): urn:ietf:params:acme:error:connection :: The server could not connect to the client to verify the domain :: Fetching https://wolke.knoeppler.org/.well-known/acme-challenge/q0POSqD0wlDXlfbL0h6N-6-08kFgybYHvLYOLwy0xEE: Timeout during connect (likely firewall problem)

        IMPORTANT NOTES:

        - The following errors were reported by the server:

           Domain: wolke.knoeppler.org
           Type:   connection
           Detail: Fetching
           https://my-nextcloud.domain/.well-known/acme-challenge/q0POSqD0wlDXlfbL0h6N-6-08kFgybYHvLYOLwy0xEE:
           Timeout during connect (likely firewall problem)

           To fix these errors, please make sure that your domain name was
           entered correctly and the DNS A/AAAA record(s) for that domain
           contain(s) the right IP address. Additionally, please check that
           your computer has a publicly routable IP address and that no
           firewalls are preventing the server from communicating with the
           client. If you're using the webroot plugin, you should also verify
           that you are serving files from the webroot path you provided.
        - Your account credentials have been saved in your Certbot
           configuration directory at /etc/letsencrypt. You should make a
           secure backup of this folder now. This configuration directory will
           also contain certificates and private keys obtained by Certbot so
           making regular backups of this folder is ideal.

Letterus, 2019-04-10:
I had the same issue and could resolve it by turning off the „force https“ setting. Due to the HSTS settings (or another configuration?) the remote host is not able to access the domain.

So the workflow is:
1. disable force https
2. renew certificate
3. enable force https

Of course it would be better if this would be handled by the renewing script itself.

nachoparker, 2019-05-26:
Couldn't reproduce on my end. Please reopen if we find a way to reproduce or a solution.

closed 2019-05-26 by nachoparker