nextcloud/nextcloudpi #873: Certbot has problem setting up the virtual environment.

luisbraganca, 2019-04-07:
First of all, thanks a lot for this project!

I actually managed to set it up a few weeks ago (using Nextcloud 15.0.5), but since this last version came out (with Nextcloud 15.0.6) I always get stuck on a step.

Tried countless times running:
```bash
$ curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash
```
But it crashes on a step, showing the following output:
```
Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip
output.

Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
for possible solutions.
You may also find some support resources at https://certbot.eff.org/support/ .
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin username.
Set an admin password.
```

Error reproduces on both `Raspbian Full (with recommended software) (2018-11-13)` and on `Raspbian Lite (2018-11-13).`

I have both ports `80` and `443` forwarding to my RPi.

Every time I ran your script, the system was already updated by running:
```bash
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
```

Output of `sudo ncp-report`:
```bash
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

cat: /usr/local/etc/ncp-version: No such file or directory
NextCloudPi version
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda
datadir               (doesn't exist)
rootfs usage         1.9G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               53min

</details>

<details>
<summary>Nextcloud configuration</summary>

Nextcloud is not installed - only a limited number of commands are available

  There are no commands defined in the "config" namespace.

</details>

<details>
<summary>HTTPd logs</summary>

[Sun Apr 07 14:13:37.730223 2019] [mpm_event:notice] [pid 8848:tid 1996280048] AH00489: Apache/2.4.25 (Raspbian) configured -- resuming normal operations
[Sun Apr 07 14:13:37.730828 2019] [core:notice] [pid 8848:tid 1996280048] AH00094: Command line: '/usr/sbin/apache2'
[Sun Apr 07 14:24:00.492833 2019] [mpm_event:notice] [pid 8848:tid 1996280048] AH00491: caught SIGTERM, shutting down
[Sun Apr 07 14:24:00.701571 2019] [mpm_event:notice] [pid 26036:tid 1995759856] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2r configured -- resuming normal operations
[Sun Apr 07 14:24:00.701709 2019] [core:notice] [pid 26036:tid 1995759856] AH00094: Command line: '/usr/sbin/apache2'

</details>

<details>
<summary>Database logs</summary>

2019-04-07 14:22:27 1996173104 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-04-07 14:22:27 1996173104 [Note] InnoDB: Using Linux native AIO
2019-04-07 14:22:27 1996173104 [Note] InnoDB: Using generic crc32 instructions
2019-04-07 14:22:27 1996173104 [Note] InnoDB: Initializing buffer pool, size = 256.0M
2019-04-07 14:22:27 1996173104 [Note] InnoDB: Completed initialization of buffer pool
2019-04-07 14:22:27 1996173104 [Note] InnoDB: Highest supported file format is Barracuda.
2019-04-07 14:22:27 1996173104 [Warning] InnoDB: Resizing redo log from 2*3072 to 2*2048 pages, LSN=1616737
2019-04-07 14:22:28 1996173104 [Warning] InnoDB: Starting to delete and rewrite log files.
2019-04-07 14:22:28 1996173104 [Note] InnoDB: Setting log file ./ib_logfile101 size to 32 MB
2019-04-07 14:22:52 1996173104 [Note] InnoDB: Setting log file ./ib_logfile1 size to 32 MB
2019-04-07 14:23:02 1996173104 [Note] InnoDB: Renaming log file ./ib_logfile101 to ./ib_logfile0
2019-04-07 14:23:02 1996173104 [Warning] InnoDB: New log files created, LSN=1616908
2019-04-07 14:23:02 1996173104 [Note] InnoDB: 128 rollback segment(s) are active.
2019-04-07 14:23:02 1996173104 [Note] InnoDB: Waiting for purge to start
2019-04-07 14:23:02 1996173104 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616737
2019-04-07 14:23:02 1233122112 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-04-07 14:23:02 1996173104 [Note] Plugin 'FEEDBACK' is disabled.
2019-04-07 14:23:02 1996173104 [Note] Server socket created on IP: '127.0.0.1'.
2019-04-07 14:23:02 1996173104 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

</details>

<details>
<summary>Nextcloud logs</summary>

tail: cannot open '/nextcloud.log' for reading: No such file or directory

</details>
```

nachoparker, 2019-04-07:
looks like a Letsencryt bug. Probably better to post an issue to them

closed 2019-04-07 by nachoparker

LimitedModz, 2019-04-17:
getting this error too

luisbraganca, 2019-04-22:
> 
> 
> getting this error too

Not sure if this approach fixes your problem, but I managed to use the pre-configured NextCloudPi system image instead:
https://ownyourbits.com/downloads/

plche, 2019-04-26:
> First of all, thanks a lot for this project!
> 
> I actually managed to set it up a few weeks ago (using Nextcloud 15.0.5), but since this last version came out (with Nextcloud 15.0.6) I always get stuck on a step.
> 
> Tried countless times running:
> 
> ```shell
> $ curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash
> ```
> 
> But it crashes on a step, showing the following output:
> 
> ```
> Certbot has problem setting up the virtual environment.
> 
> We were not be able to guess the right solution from your pip
> output.
> 
> Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
> for possible solutions.
> You may also find some support resources at https://certbot.eff.org/support/ .
> Running nc-init
> Setting up a clean Nextcloud instance... wait until message 'NC init done'
> Setting up database...
> Setting up Nextcloud...
> Set an admin username.
> Set an admin password.
> ```
> Error reproduces on both `Raspbian Full (with recommended software) (2018-11-13)` and on `Raspbian Lite (2018-11-13).` The device is a `Raspberry Pi 3 Model B+`.
> 
> I have both ports `80` and `443` forwarding to my RPi.
> 
> Every time I ran your script, the system was already updated by running:
> 
> ```shell
> sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
> ```
> 
> Output of `sudo ncp-report`:
> 
> ```shell
> <--! Paste this in GitHub report -->
> 
> <details>
> <summary>NextCloudPi diagnostics</summary>
> 
> 
> cat: /usr/local/etc/ncp-version: No such file or directory
> NextCloudPi version
> distribution         Raspbian GNU/Linux 9 \n \l
> automount            no
> USB devices          sda
> datadir               (doesn't exist)
> rootfs usage         1.9G/30G
> swapfile             /var/swap
> dbdir                /var/lib/mysql
> Nextcloud check      ok
> Nextcloud version    15.0.6.1
> HTTPD service        up
> PHP service          up
> MariaDB service      up
> Redis service        up
> Postfix service      up
> internet check       ok
> port check 80        open
> port check 443       open
> IP                   ***REMOVED SENSITIVE VALUE***
> gateway              ***REMOVED SENSITIVE VALUE***
> interface            eth0
> certificates         ***REMOVED SENSITIVE VALUE***
> NAT loopback         no
> uptime               53min
> 
> </details>
> 
> 
> <details>
> <summary>Nextcloud configuration</summary>
> 
> 
> Nextcloud is not installed - only a limited number of commands are available
> 
> 
>   There are no commands defined in the "config" namespace.
> 
> 
> 
> </details>
> 
> 
> <details>
> <summary>HTTPd logs</summary>
> 
> 
> [Sun Apr 07 14:13:37.730223 2019] [mpm_event:notice] [pid 8848:tid 1996280048] AH00489: Apache/2.4.25 (Raspbian) configured -- resuming normal operations
> [Sun Apr 07 14:13:37.730828 2019] [core:notice] [pid 8848:tid 1996280048] AH00094: Command line: '/usr/sbin/apache2'
> [Sun Apr 07 14:24:00.492833 2019] [mpm_event:notice] [pid 8848:tid 1996280048] AH00491: caught SIGTERM, shutting down
> [Sun Apr 07 14:24:00.701571 2019] [mpm_event:notice] [pid 26036:tid 1995759856] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2r configured -- resuming normal operations
> [Sun Apr 07 14:24:00.701709 2019] [core:notice] [pid 26036:tid 1995759856] AH00094: Command line: '/usr/sbin/apache2'
> 
> </details>
> 
> 
> <details>
> <summary>Database logs</summary>
> 
> 
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Compressed tables use zlib 1.2.8
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Using Linux native AIO
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Using generic crc32 instructions
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Initializing buffer pool, size = 256.0M
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Completed initialization of buffer pool
> 2019-04-07 14:22:27 1996173104 [Note] InnoDB: Highest supported file format is Barracuda.
> 2019-04-07 14:22:27 1996173104 [Warning] InnoDB: Resizing redo log from 2*3072 to 2*2048 pages, LSN=1616737
> 2019-04-07 14:22:28 1996173104 [Warning] InnoDB: Starting to delete and rewrite log files.
> 2019-04-07 14:22:28 1996173104 [Note] InnoDB: Setting log file ./ib_logfile101 size to 32 MB
> 2019-04-07 14:22:52 1996173104 [Note] InnoDB: Setting log file ./ib_logfile1 size to 32 MB
> 2019-04-07 14:23:02 1996173104 [Note] InnoDB: Renaming log file ./ib_logfile101 to ./ib_logfile0
> 2019-04-07 14:23:02 1996173104 [Warning] InnoDB: New log files created, LSN=1616908
> 2019-04-07 14:23:02 1996173104 [Note] InnoDB: 128 rollback segment(s) are active.
> 2019-04-07 14:23:02 1996173104 [Note] InnoDB: Waiting for purge to start
> 2019-04-07 14:23:02 1996173104 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 1616737
> 2019-04-07 14:23:02 1233122112 [Note] InnoDB: Dumping buffer pool(s) not yet started
> 2019-04-07 14:23:02 1996173104 [Note] Plugin 'FEEDBACK' is disabled.
> 2019-04-07 14:23:02 1996173104 [Note] Server socket created on IP: '127.0.0.1'.
> 2019-04-07 14:23:02 1996173104 [Note] /usr/sbin/mysqld: ready for connections.
> Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
> 
> </details>
> 
> 
> <details>
> <summary>Nextcloud logs</summary>
> 
> 
> tail: cannot open '/nextcloud.log' for reading: No such file or directory
> 
> </details>
> ```

For this error, the solution is to remove the /etc/pip.conf as explained here: [https://community.letsencrypt.org/t/certbot-auto-certificates-fails-while-installing-phyton-packages-with-these-packages-do-not-match-the-hashes/90363](url)
Basically that error it is because Raspbian has configured `pip` to use third party mirrors which `certbot-auto` doesn’t trust, but configuring `pip` to not use [piwheels.org](url) will fix it...