nextcloud/nextcloudpi #875: [Rock64] NextCloudPi_Rock64_03-10-19 - USB 3.0 SSD not seen, but 2.0 is

SirRyanTheGeek, 2019-04-08:
eMMC flashed with NextCloudPi_Rock64_03-10-19.img

On setup, when the installer asks do you want to move data to USB and you say "Yes", the USB device is never seen when you plug it into the USB 3.0 port, but does see it when you plug it into the USB 2.0 slot. 

lsblk -ln while plugged into USB 2.0:

root@nextcloudpi:~# lsblk -ln
sda            8:0    0 223.6G  0 disk
sda1           8:1    0 223.6G  0 part /media/myCloudDrive
mtdblock0     31:0    0     4M  0 disk
mmcblk0      179:0    0  57.6G  0 disk
mmcblk0p1    179:1    0    57G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
mmcblk0rpmb  179:96   0     4M  0 disk
zram0        252:0    0    50M  0 disk /var/log
zram1        252:1    0 498.6M  0 disk [SWAP]
zram2        252:2    0 498.6M  0 disk [SWAP]
zram3        252:3    0 498.6M  0 disk [SWAP]
zram4        252:4    0 498.6M  0 disk
root@nextcloudpi:~#

lsblk -ln while plugged into 3.0 port:

root@nextcloudpi:~# lsblk -ln
mtdblock0     31:0    0     4M  0 disk
mmcblk0      179:0    0  57.6G  0 disk
mmcblk0p1    179:1    0    57G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
mmcblk0rpmb  179:96   0     4M  0 disk
zram0        252:0    0    50M  0 disk /var/log
zram1        252:1    0 498.6M  0 disk [SWAP]
zram2        252:2    0 498.6M  0 disk [SWAP]
zram3        252:3    0 498.6M  0 disk [SWAP]
zram4        252:4    0 498.6M  0 disk
root@nextcloudpi:~#

Tried fdisk -l and still no /dev/sd* device shown anywhere. 

Wondering if it's something with this specific version of Linux used for the image? I have tried numerous different USB drives & keys - nothing is seen. 

Also, when I boot with Ubuntu Bionic, the SSD is seen perfectly fine as /dev/sda.

Thanks!

root@nextcloudpi:~# sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.9.3
NextCloudPi image    NextCloudPi_03-11-19
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      317M/224G
rootfs usage         3.8G/57G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
[... 134 lines cut ...]
```

{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:39+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::info: Repair info: SCSS cache cleared","userAgent":"Mozilla\/5.0 (Windows N                                                                              T 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"                                                                              }
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::info: Repair info: JS cache cleared","userAgent":"Mozilla\/5.0 (Windows NT                                                                               10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::step: Repair step: Clear every generated avatar on major updates","userAgen                                                                              t":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\                                                                              /66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::step: Repair step: Add preview background cleanup job","userAgent":"Mozilla                                                                              \/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","ver                                                                              sion":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::step: Repair step: Queue a one-time job to cleanup old backups of the updat                                                                              er","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/2010                                                                              0101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKo                                                                              sRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Rep                                                                              air::step: Repair step: Repair pending cron jobs","userAgent":"Mozilla\/5.0 (Win                                                                              dows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.                                                                              0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00"                                                                              ,"remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\                                                                              /core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg [... cut ...] OC","type":"::","args":[]}],"File":"\/var\/www\                                                                              /nextcloud\/lib\/private\/Files\/Storage\/Local.php","Line":387,"CustomMessage":                                                                              "--"},"userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20                                                                              100101 Firefox\/66.0","version":"15.0.6.1"}

```
</details>

ovpc, 2019-04-08:
Looks like this happened and was resolved [here](https://www.linuxquestions.org/questions/slackware-14/usb-3-0-port-not-working-925355/)
The solution was
> Try adding pci=nomsi to the boot parameters

nachoparker, 2019-04-08:
I suggest you take this to the Armbian forums

closed 2019-04-08 by nachoparker