nextcloud/nextcloudpi #876: [Rock64] NextCloudPi_Rock64_03-10-19 - First time user logs in - Internal Server Error - no directories created

SirRyanTheGeek, 2019-04-08:
Once NextCloud is setup, while acting as the ncp user, I create a user in the GUI then logout. When I log back in with the new username and password, there is an "Internal Server Error". When I go back and login again as the new user, I can login - but no folders like "Documents", etc., are created. It's a totally blank directory.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.9.3
NextCloudPi image    NextCloudPi_03-11-19
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      317M/224G
rootfs usage         3.8G/57G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4:26

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "70.90.210.67",
            "1": "192.168.86.31",
            "4": "mynextcloud.iotai.online"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.6.1",
        "overwrite.cli.url": "https:\/\/mynextcloud.iotai.online\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
[... 89 lines cut ...]
```

{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:39+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::step: Repair step: Clear every generated avatar on major updates","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::info: Repair info: No need to repair pending cron jobs.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Repair::step: Repair step: Extract the vcard uid and store it in the db","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:40+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:53+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.5.3"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:53+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.6.1"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:53+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQz3FppIxb8vj%2FxTNydOtCJXy7BvsUhDtpaNcU%3D","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.6.1"}
{"reqId":"XKqEi38AAQEAAClamFoAAAAP","level":1,"time":"2019-04-07T23:15:53+00:00","remoteAddr":"192.168.86.1","user":"--","app":"updater","method":"GET","url":"\/core\/ajax\/update.php?requesttoken=uS5f5MoG7Dg7E1GZ0FeZGoyqa7HbOlev%2F%2FdbRKosRYQ%3D%3A4VkLroQ [... cut ...] gin"}]},{"file":"\/var\/www\/nextcloud\/lib\/base.php","line":987,"function":"match","class":"OC\\Route\\Router","type":"->","args":["\/login"]},{"file":"\/var\/www\/nextcloud\/index.php","line":42,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"\/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php","Line":387,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.6.1"}

```
</details>

SirRyanTheGeek, 2019-04-08:
Not sure if related, but did see this in apache's error.log file:

[Mon Apr 08 05:06:02.937220 2019] [proxy_fcgi:error] [pid 1400:tid 548047491504] [client 192.168.86.38:55861] AH01071: Got error 'PHP message: PHP Notice:  Undefined index: app in /var/www/ncp-web/index.php on line 219\n'

SirRyanTheGeek, 2019-04-08:
In case someone wants to poke around in my server, let me know. I'm leaving it spun up for testing. Any user I create gets the same situation - server error on first login, then they attempt a login again and can get in - but no directories such as documents, instruction PDF and other things are setup. It's just blank.

TheUntouchable, 2019-05-05:
Same here! Just installed the latest image on my RPI3+ today and updated everything before the first user I created but its always the same..

Co0p3r21, 2019-05-07:
I had the same issue before and I managed to fix it. But beware that my solution isn´t necessary your solution. 

I have my data on a USB device. This device is mounted from the ncp interface. This creates the normal mount and a symlink ("myCloudDrive" or something like this)  to the mount. I than configured that my data should be on the usb device and used the symlink (this is default behavior, i think). But apache is disallowed to use symlinks as far as I know, so this results in the described error. 

The Solution would be to just don´t use the symlink at all. 
If you already use the symlink and use the ncp-interface function to move your data, you will get another error. Because this basicly moves the database to the same place it was before. But a backup folder will be created in the process, so no worries.

nachoparker, 2019-05-26:
Sounds like a Nextcloud server issue. Thanks for reporting!

closed 2019-05-26 by nachoparker