nextcloud/nextcloudpi #877: NextCloud Version 15.0.6 doesnt exist.

lotze1234, 2019-04-08:
Hi all,

I try to install mynextcloudpi with : https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh .....

Everything worksfine till the script will download the nextcloud release. 

It tryed to download the version 15.0.6 but it doesnt exist. 

How i can fix that?

Regards

ovpc, 2019-04-08:
Probably a time out of the NC server, just wait a cpl min or hr and try again.
Please use the [forum](https://help.nextcloud.com/tags/ncp) to get help.

closed 2019-04-08 by ovpc

lotze1234, 2019-04-08:
Hi thank you for your answer, but please visit the nextcloud homepage - they took the release 15.0.6 off. Latest working version is now again 15.0.5.

ovpc, 2019-04-08:
Weird! But nothing to do with NCP tho
Please report it on the Nextcloud forum or hub.
If NCP is installed, you can run nc-nextcloud and at least install 15.0.5 
and just wait for NC devs to get it fixed.