nextcloud/nextcloudpi #884: LetsEncrypt won't renew expired certificate - says it doesn't need updated

Zackptg5, 2019-04-14:
I've been unable to renew letsencrypt certificate, keeps saying that it's current and doesn't need updated. Below is the ncp-report and the letsencrypt log

`root@nextcloudpi:~# sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.11.2
NextCloudPi image    NextCloudPi_12-01-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      252G/932G
rootfs usage         3.7G/15G
swapfile             /var/swap
dbdir                /media/myCloudDrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630c6f1
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               6days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.3",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "2": "nextcloud.zackptg5.com",
            "4": "nextcloud.zackptg5.com",
            "11": "24.210.23.177"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.6.1",
        "overwrite.cli.url": "https:\/\/nextcloud.zackptg5.com\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 151 lines cut ...]
    "caaIdentities": [
      "letsencrypt.org"
    ],
    "termsOfService": "https://letsencrypt.org/documents/LE-SA-v1.2-November-15-2017.pdf",
    "website": "https://letsencrypt.org"
  },
  "newAccount": "https://acme-v02.api.letsencrypt.org/acme/new-acct",
  "newNonce": "https://acme-v02.api.letsencrypt.org/acme/new-nonce",
  "newOrder": "https://acme-v02.api.letsencrypt.org/acme/new-order",
  "revokeCert": "https://acme-v02.api.letsencrypt.org/acme/revoke-cert",
  "uRvqmJwwSCc": "https://community.letsencrypt.org/t/adding-random-entries-to-the-directory/33417"
}
2019-04-14 19:39:39,535:DEBUG:certbot.cert_manager:Renewal conf file /etc/letsencrypt/renewal/nextcloud.zackptg5.com.conf is broken. Skipping.
2019-04-14 19:39:39,537:DEBUG:certbot.cert_manager:Traceback was:
Traceback (most recent call last):
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/certbot/cert_manager.py", line 384, in _search_lineages
    candidate_lineage = storage.RenewableCert(renewal_file, cli_config)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/certbot/storage.py", line 443, in __init__
    "file reference".format(self.configfile))
CertStorageError: renewal config file {} is missing a required file reference

2019-04-14 19:39:39,552:INFO:certbot.renewal:Cert not yet due for renewal
2019-04-14 19:39:39,552:INFO:certbot.main:Keeping the existing certificate`

Since the log says it's broken, can a feature be added to remove letsencrypt in the event something like this happens? I found the you can use 'cerbot delete' but that doesn't work in ncp

Zackptg5, 2019-04-16:
So I'm not sure what happened, but something got borked at some point and ncp was using a different set of certs. Fixing the conf file in /etc/letsencrypt/renewal fixed the issue

jasondefuria, 2019-04-20:
I have this exact issue. Looks like the same as another reddit user. Looked and found two .conf files in /etc/letsencrypt/renewal but don't know how to fix.

jasondefuria, 2019-04-20:
Actually, was able to replicate on another ncp instance. It seems on update to 1.11.2, it creates a copy of the certificates during migration for some reason. I had mydomain.duckdns.org and mydomain.duckdns.org-0001. The conf file (mydomain.duckdns.org-0001.conf) housed the details to duplicate keys that were never migrated. The original conf (mydomain.duckdns.org.conf) was a blank file. I copied what was found in the 0001 .conf file to the original .conf, and modified the certs (all had the 0001 path) Voila, it auto updates. Have seen same thing on 2 instances. Not sure if it's because the two installs that I have use basically the same configuration settings, or if it is wider spread and just a few of our installs have certs due now.

fsan06, 2019-04-24:
Hi, I'm facing the same issue my certificate is expired, when I try to see what is in /etc/letsencrypt/renewal it is a blank file, what I can do to renew my certificate or fix the issue? thanks.

Zackptg5, 2019-04-24:
Since other people have this issue too, I'm going to reopen this (since it wasn't due to me being stupid)

jasondefuria, 2019-04-24:
@fsan06 SSH in, and run this:
```
cd /etc/letsencrypt/renewal
ls -a
```

if you have a -0001.conf

    ```
mv yourdomain.com.conf yourdomain.com.conf.bak
    cp yourdomain.com-0001.conf yourdomain.com.conf
    sed -i "s/-0001//g" yourdomain.com.conf
```

I can't take credit for the above code, as I'm not as proficient as the guy who wrote it, but I did write many more of the steps on [this reddit thread.](https://www.reddit.com/r/NextCloud/comments/bfdsal/security_certificate_expired_3_days_ago/eld6ye9?utm_source=share&utm_medium=web2x)

fsan06, 2019-04-25:
Thank you for your help, I can't fix the issue, now I facing another problem I can't access to my nextcloudpi from my browser "Unable to connect" it's the message displayed, I can connect through SSH everything seems right. Any thoughts? thanks in advance

codac, 2019-04-26:
I've had the same Issue. The HTTPD service is down and can't be (re) started. In my case it was due to the file 

> /etc/apache2/sites-enabled/nextcloud.conf

 which points to a folder 

> /var/www/nextcloud/data/

.

```
CustomLog /var/www/nextcloud/data/access.log combined
ErrorLog /var/www/nextcloud/data/error.log
```

In my case the folder /data/ in /var/www/nextcloud/ did not exist anymore. 
So I did a:
`mkdir data`

and started the service with:

`service apache2 restart`

After that I executed the letsencrypt option in ncp-config.

tomsrobots, 2019-04-27:
Is there anyway to start from scratch with letsencrypt. I don't really know what's going on here and I think I got my system in a really messed up state. I removed the letsencrypt directories for my duckdns account (I had the 001 version mentioned here), but when I went to re-run letsencrypt I've fallen down a rabbit hole of certbot and Docker and all kinds of things I don't really understand.

Will there be a Nextcloud Pi patch that fixes this? Do I need to tear everything up and start from scratch?

fsan06, 2019-04-27:
Hi, thank you @codac for the reply, I fixed the issue following another solution that @nachoparker suggest. I think cert is releted to apache2.
there is the link: https://help.nextcloud.com/t/solved-cannot-access-nextcloudpi-connection-refused-httpd-service-down/40637/13
@tomsrobots maybe that link can help you. Let me know if you have questions.

tomsrobots, 2019-04-27:
I went through the above steps and got stuck on rerunning ncp-update after removing letsencrypt. Here's the accompanying error:

`

Requirement already satisfied: setuptools>=1.0 in /opt/eff.org/certbot/venv/lib/python2.7/site-packages (from josepy==1.1.0->-r /tmp/tmp.Bqx9Mc9lY9/letsencrypt-auto-requirements.txt (line 87))         THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may ha ve tampered with them.                                                                                                                                                                                       pycparser==2.19 from https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl#sha256=344870c533812cb119c5475b5aed033ec546118711338ff9b2b78e67098d64e7 (from -r /tmp/tmp.Bqx9Mc9 lY9/letsencrypt-auto-requirements.txt (line 105)):                                                                                                                                                               Expected LONG HEX CODE                                                                                                                              Got        DIFFERENT LONG HEX CODE                                                                                                                 =====================================================                                                                                                                                                                                                                                                                                                                                                             Certbot has problem setting up the virtual environment.                                                                                                                                                                                                                                                                                                                                                           We were not be able to guess the right solution from your pip                                                                                                                                            output.                                                                                                                                                                                                                                                                                                                                                                                                           Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment                                                                                                               for possible solutions.                                                                                                                                                                                  You may also find some support resources at https://certbot.eff.org/support/ .
--`

nachoparker, 2019-04-27:
Try installing letsencrypt from `apt` -> `sudo apt-get install letsencrypt`

That is what we are going to do soon, since the vulnerability that made us change has been fixed

fsan06, 2019-04-27:
Hi, I don't know if i"m still having issues letsencrypt related,finally I can get access to my nextcloud from my computer (via browser), but I cannot get access from any smartphone via browser or app, I got an error like unable to get the remote host, please any help will be relly appreciated, there is my ncp-report:

details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.11.2
NextCloudPi image    NextCloudPi_11-22-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      37G/118G
rootfs usage         37G/118G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.7.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               9min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "2601:645:8003:6865::5785",
            "1": "10.0.0.147",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "2": "depressiva.duckdns.org",
            "4": "depressiva.duckdns.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.7.0",
        "overwrite.cli.url": "https:\/\/depressiva.duckdns.org\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
[... 84 lines cut ...]

```

{"reqId":"XMSaAn8AAQEAAF4ZlDMAABAb","level":3,"time":"2019-04-27T18:05:54+00:00","remoteAddr":"199.230.10.197","user":"--","app":"index","method":"GET","url":"\/index.php\/apps\/music\/ampache\/server\/xml.server.php?action=_get_cover&filter=684&auth=9b0d2a7d86f3ce5657f7563fb6a3591c","message":{"Exception":"OCP\\AppFramework\\Db\\DoesNotExistException","Message":"Did expect one result but found none when executing: query \"SELECT * FROM `*PREFIX*music_albums` WHERE `id` = ? AND `user_id` = ?\"; parameters Array\n(\n    [0] => 684\n    [1] => ncp\n)\n; limit \"\"; offset \"\"","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/public\/AppFramework\/Db\/Mapper.php","line":372,"function":"findOneQuery","class":"OCP\\AppFramework\\Db\\Mapper","type":"->","args":["SELECT * FROM `*PREFIX*music_albums` WHERE `id` = ? AND `user_id` = ?",["684","ncp"],null,null]},{"file":"\/var\/www\/nextcloud\/apps\/music\/db\/basemapper.php","line":44,"function":"findEntity","class":"OCP\\AppFramework\\Db\\Mapper","type":"->","args":["SELECT * FROM `*PREFIX*music_albums` WHERE `id` = ? AND `user_id` = ?",["684","ncp"]]},{"file":"\/var\/www\/nextcloud\/apps\/music\/businesslayer\/albumbusinesslayer.php","line":40,"function":"find","class":"OCA\\Music\\Db\\BaseMapper","type":"->","args":["684","ncp"]},{"file":"\/var\/www\/nextcloud\/apps\/music\/utility\/coverhelper.php","line":192,"function":"find","class":"OCA\\Music\\BusinessLayer\\AlbumBusinessLayer","type":"->","args":["684","ncp"]},{"file":"\/var\/www\/nextcloud\/apps\/music\/utility\/coverhelper.php","line":78,"function":"readCover","class":"OCA\\Music\\Utility\\CoverHelper","type":"->","args":["684","ncp",{"__class__":"OC\\Files\\Node\\Folder"}]},{"file":"\/var\/www\/nextcloud\/apps\/music\/utility\/coverhelper.php","line":55,"function":"getCoverAndHash","class":"OCA\\Music\\Utility\\CoverHelper","type":"->","args":["684","ncp",{"__class__":"OC\\Files\\Node\\Folder"}]},{"file":"\/var\/www\/nextcloud\/apps\/music\/controller\/ampachecontroller.php","line":401,"function":"getCover","class":"OCA\\Music\\Utility\\CoverHelper","type":"->","args":["684","ncp",{"__class__":"OC\\Files\\Node\\Folder"}]},{"file":"\/var\/www\/nextcloud\/apps\/music\/controller\/ampachecontroller.php","line":141,"function":"get_cover","class":"OCA\\Music\\Controller\\AmpacheController","type":"->","args":["684"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Dispatcher.php","line":166,"function":"ampache","class":"OCA\\Music\\Controller\\AmpacheController","type":"->","args":["_get_cover",null,null,"9b0d2a7d86f3ce5657f7563fb6a3591c","684",null,null,null]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Dispatcher.php","line":99,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OCA\\Music\\Controller\\AmpacheController"},"ampache"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/App.php","line":118,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OCA\\Music\\Controller\\AmpacheController"},"ampache"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/A [... cut ...]
{"reqId":"XMTGPX8AAQEAAAN@@mwAAFMH","level":2,"time":"2019-04-27T21:14:37+00:00","remoteAddr":"10.0.0.228","user":"--","app":"core","method":"GET","url":"\/","message":"Renewing session token failed","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.7.0"}
{"reqId":"XMTGRX8AAQEAAAN@@pAAAE4H","level":2,"time":"2019-04-27T21:14:46+00:00","remoteAddr":"10.0.0.228","user":"--","app":"core","method":"POST","url":"\/index.php\/login?redirect_url=\/index.php\/apps\/files\/","message":"Login failed: 'ncp' (Remote IP: '10.0.0.228')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:66.0) Gecko\/20100101 Firefox\/66.0","version":"15.0.7.0"}

```

tomsrobots, 2019-04-28:
> 
> 
> Try installing letsencrypt from `apt` -> `sudo apt-get install letsencrypt`
> 
> That is what we are going to do soon, since the vulnerability that made us change has been fixed.
> 
> The github version is definitely a buggy nightmare

Thank you for reaching out. I've installed letsencrypt with apt, but I don't know what to do next. The error I received was after letsencrypt was installed with apt.

fsan06, 2019-04-28:
@tomsrobots Did you try this command after that? sudo apt install -f

fsan06, 2019-04-28:
I mean after your ncp-ugrade fail?

tomsrobots, 2019-04-28:
I am saying I installed letsencrypt through apt, but I don't know what to do now. In the passed I just did ncp-config and it took care of everything. Now manually letsencrypt is asking for certbot and the only way I can see to install certbot is with Docker, and I know I'm out of my element.

nachoparker, 2019-04-28:
Ok, I just pushed a fix for this, please help me test it by running `sudo ncp-update devel` before I release it to everyone

closed 2019-04-28 by nachoparker