nextcloud/nextcloudpi #886: Rename configuration variables into self-documenting ones

Hofer-Julian, 2019-04-16:
Especially when it comes to number-based configuration variables, I often had to look into the source code to be 100% sure what they meant. That is why I suggest to rename some of them to make their purpose more clear to other users. 

- Backups
  - nc-backups-auto
    - Backup days -> Backup periodicity (in days)
    - Backup limit -> Number of backups to keep
  - nc-backup
    - Backup limit -> Number of backups to keep
  - nc-rsync-auto
    - Sync days -> Sync periodicity (in days)
  - nc-snapshot-sync
    - Sync days -> Sync periodicity (in days)
- Config
  - nc-previews-auto
    - Run time -> Runtime (in minutes)
  - nc-scan-auto
    - Scan interval -> Scan periodicity (in minutes)
- Networking
  - freeDNS
    - Update Interval -> Update periodicity (in minutes)
  - no-ip
    - Time -> ?

I used the word periodicity a lot of times, maybe there is a better one, but it would be great if it would be the same for all entries which occur periodically :)

nachoparker, 2019-04-18:
good point. Please send a PR with your proposed changes, it's only a bunch of JSON files so it's an easy edit

nachoparker, 2019-05-26:
PR merged, thanks!

PR #889 (2019-05-01): Rename configuration variables into self-documenting ones
Fixes #886
closed 2019-05-26 by nachoparker