nextcloud/nextcloudpi #892: installation doesn't work

dh-connect, 2019-04-27:
i try to install ncp on raspi3, but my installation is not running through, it ends with
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin username.
Set an admin password.

any idea what the problem is?

`<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-version: No such file or directory
NextCloudPi version
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir               (doesn't exist)
rootfs usage         8.4G/58G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:32

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

                                                            
  There are no commands defined in the "config" namespace.  
                                                            

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sat Apr 27 12:45:45.532288 2019] [ssl:warn] [pid 18988] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Apr 27 12:45:45.532565 2019] [ssl:error] [pid 18988] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C1E97DC279CF3B40 / notbefore: Apr  8 20:52:31 2018 GMT / notafter: Apr  5 20:52:31 2028 GMT]
[Sat Apr 27 12:45:45.532587 2019] [ssl:error] [pid 18988] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat Apr 27 12:45:45.533356 2019] [ssl:warn] [pid 18988] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Apr 27 12:45:45.533621 2019] [ssl:error] [pid 18988] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C1E97DC279CF3B40 / notbefore: Apr  8 20:52:31 2018 GMT / notafter: Apr  5 20:52:31 2028 GMT]
[Sat Apr 27 12:45:45.533678 2019] [ssl:error] [pid 18988] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat Apr 27 12:45:45.533824 2019] [http2:warn] [pid 18988] AH10034: The mpm module (prefork.c) is not supported by mod_http2. The mpm determines how things are processed in your server. HTTP/2 has more demands in this regard and the currently selected mpm will just not do. This is an advisory warning. Your server will continue to work, but the HTTP/2 protocol will be inactive.
[Sat Apr 27 12:45:45.851659 2019] [mpm_prefork:notice] [pid 18988] AH00163: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2r configured -- resuming normal operations
[Sat Apr 27 12:45:45.851731 2019] [core:notice] [pid 18988] AH00094: Command line: '/usr/sbin/apache2'
[Sat Apr 27 12:45:52.494570 2019] [mpm_prefork:notice] [pid 18988] AH00171: Graceful restart requested, doing restart
AH00112: Warning: DocumentRoot [/var/www/ncp-web/] does not exist
[Sat Apr 27 12:45:52.638159 2019] [ssl:warn] [pid 18988] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Apr 27 12:45:52.638430 2019] [ssl:error] [pid 18988] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C1E97DC279CF3B40 / notbefore: Apr  8 20:52:31 2018 GMT / notafter: Apr  5 20:52:31 2028 GMT]
[Sat Apr 27 12:45:52.638451 2019] [ssl:error] [pid 18988] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat Apr 27 12:45:52.639334 2019] [ssl:warn] [pid 18988] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat Apr 27 12:45:52.639616 2019] [ssl:error] [pid 18988] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C1E97DC279CF3B40 / notbefore: Apr  8 20:52:31 2018 GMT / notafter: Apr  5 20:52:31 2028 GMT]
[Sat Apr 27 12:45:52.639637 2019] [ssl:error] [pid 18988] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat Apr 27 12:45:52.639783 2019] [http2:warn] [pid 18988] AH10034: The mpm module (prefork.c) is not supported by mod_http2. The mpm determines how things are processed in your server. HTTP/2 has more demands in this regard and the currently selected mpm will just not do. This is an advisory warning. Your server will continue to work, but the HTTP/2 protocol will be inactive.
[Sat Apr 27 12:45:52.913323 2019] [mpm_prefork:notice] [pid 18988] AH00163: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2r configured -- resuming normal operations
[Sat Apr 27 12:45:52.913391 2019] [core:notice] [pid 18988] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2019-04-27 13:04:12 1971110720 [Note] /usr/sbin/mysqld: Shutdown complete

2019-04-27 13:04:15 1995512624 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2019-04-27 13:04:15 1995512624 [Note] InnoDB: The InnoDB memory heap is disabled
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2019-04-27 13:04:15 1995512624 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Using Linux native AIO
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Using generic crc32 instructions
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Initializing buffer pool, size = 256.0M
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Completed initialization of buffer pool
2019-04-27 13:04:15 1995512624 [Note] InnoDB: Highest supported file format is Barracuda.
2019-04-27 13:04:16 1995512624 [Note] InnoDB: 128 rollback segment(s) are active.
2019-04-27 13:04:16 1995512624 [Note] InnoDB: Waiting for purge to start
2019-04-27 13:04:16 1995512624 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.41-84.1 started; log sequence number 71152977
2019-04-27 13:04:17 1233122112 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-04-27 13:04:17 1995512624 [Note] Plugin 'FEEDBACK' is disabled.
2019-04-27 13:04:17 1995512624 [Note] Server socket created on IP: '127.0.0.1'.
2019-04-27 13:04:17 1995512624 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.37-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

`

ovpc, 2019-04-27:
Looks to me like you need installation help, but it does not look like a bug. 

So please post on the [forum](https://help.nextcloud.com/tags/ncp) first, if it turns out to be a bug, we can re-open it here.Thanks for providing steps you took, image version, what documentation or installation methode you used and link to it on the forum.

dh-connect, 2019-04-27:
I used the installer provided on the homepage and was hopping that it works:
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh \| bash
--
raspberry pi 3, 4.14.98-v7+ 
It doesn't seem like that re-running the installer will fix the problem.
 now it breaks here:
Site ncp-activation already enabled
useradd: user 'ncp' already exists

dh-connect, 2019-04-27:
here's the install log

`>>> WARNING: existing mysqld configuration will be changed <<<
Getting build code...
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:3 https://packages.sury.org/php stretch InRelease
Ign:4 https://attic.owncloud.com/org/download/repositories/stable/Debian_9.0  InRelease
Hit:5 https://download.owncloud.org/download/repositories/stable/Debian_9.0  Release
Fetched 25.4 kB in 3s (8,175 B/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
ca-certificates is already the newest version (20161130+nmu1+deb9u1).
sudo is already the newest version (1.8.19p1-2.1).
wget is already the newest version (1.18-5+deb9u3).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
/tmp/nextcloudpi.wMbcL7 /home
/home

Installing NextCloudPi
Installing lamp
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://packages.sury.org/php stretch InRelease
Ign:4 https://attic.owncloud.com/org/download/repositories/stable/Debian_9.0  InRelease
Hit:5 https://download.owncloud.org/download/repositories/stable/Debian_9.0  Release
Fetched 25.4 kB in 3s (7,006 B/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
apt-transport-https is already the newest version (1.4.9).
ca-certificates is already the newest version (20161130+nmu1+deb9u1).
gnupg2 is already the newest version (2.1.18-8~deb9u4).
wget is already the newest version (1.18-5+deb9u3).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
OK
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:3 https://packages.sury.org/php stretch InRelease
Ign:4 https://attic.owncloud.com/org/download/repositories/stable/Debian_9.0  InRelease
Hit:5 https://download.owncloud.org/download/repositories/stable/Debian_9.0  Release
Fetched 25.4 kB in 3s (7,120 B/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
apt-utils is already the newest version (1.4.9).
cron is already the newest version (3.0pl1-128+deb9u1).
curl is already the newest version (7.52.1-5+deb9u9).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
apache2 is already the newest version (2.4.25-3+deb9u7).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
php7.2 is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-bz2 is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-cli is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-common is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-curl is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-fpm is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-gd is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-intl is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
php7.2-json is already the newest version (7.2.17-1+0~20190412071344.20+stretch~1.gbp23a36d).
[... 139 lines cut ...]
Building dependency tree...
Reading state information...
dialog is already the newest version (1.3-20160828-2).
git is already the newest version (1:2.11.0-3+deb9u4).
jq is already the newest version (1.5+dfsg-1.3).
whiptail is already the newest version (0.52.19-1).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
libapache2-mod-authnz-external is already the newest version (3.3.2-0.1+b1).
pwauth is already the newest version (2.3.11-0.2).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Module authnz_external already enabled
Module authn_core already enabled
Considering dependency authn_core for auth_basic:
Module authn_core already enabled
Module auth_basic already enabled
Site nextcloud disabled.
To activate the new configuration, you need to run:
  systemctl reload apache2
Site ncp-activation already enabled
useradd: user 'ncp' already exists

`

ovpc, 2019-04-30:
curl installer can have issues, as there is not enough manpower to keep it up to date, and things may have changed upstream. Best use one of the images or docker to get going with ncp.

I will give curl installer a spin, later on today, to see if I can reproduce this.

closed 2019-04-30 by ovpc