nextcloud/nextcloudpi #893: Curl installer script does not install everything

oerkenrotten, 2019-04-28:
I am trying to install nextcloudpi on my raspbian desktop via the curl installer. After fixing my locales, it seems to complete without errors. However, logging in at https://X.X.X.X gives "The requested URL / was not found on this server." and https://X.X.X.X:4443 gives "refused to connect". I cannot use nc-init or many other nc commands. And the directories seem only partially populated with content.

sudo systemctl status apache2 gives me:

`apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-04-28 12:06:02 CEST; 16min ago
  Process: 2950 ExecStop=/usr/sbin/apachectl stop (code=exited, status=0/SUCCESS)
  Process: 2771 ExecReload=/usr/sbin/apachectl graceful (code=exited, status=0/SUCCESS)
  Process: 2956 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 2961 (apache2)
   CGroup: /system.slice/apache2.service
           ├─2961 /usr/sbin/apache2 -k start
           ├─2967 /usr/sbin/apache2 -k start
           └─2968 /usr/sbin/apache2 -k start

Apr 28 12:06:02 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Apr 28 12:06:02 raspberrypi apachectl[2956]: AH00112: Warning: DocumentRoot [/var/www/ncp-web/] does not exist
Apr 28 12:06:02 raspberrypi systemd[1]: Started The Apache HTTP Server.`

sudo ncp-report gives me this:

`pi@raspberrypi:~ $ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-version: No such file or directory
NextCloudPi version
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir               (doesn't exist)
rootfs usage         5.4G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      up
internet check       ok
port check 80        open
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:06

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

                                                            
  There are no commands defined in the "config" namespace.  
                                                            

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun Apr 28 12:05:56.778466 2019] [core:notice] [pid 756:tid 1996275712] AH00094: Command line: '/usr/sbin/apache2'
[Sun Apr 28 12:06:02.393226 2019] [mpm_event:notice] [pid 756:tid 1996275712] AH00491: caught SIGTERM, shutting down
[Sun Apr 28 12:06:02.548828 2019] [ssl:warn] [pid 2960:tid 1995722752] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 1.0.2r  26 Feb 2019, version currently loaded is OpenSSL 1.0.2q  20 Nov 2018) - may result in undefined or erroneous behavior
[Sun Apr 28 12:06:02.550692 2019] [ssl:warn] [pid 2960:tid 1995722752] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sun Apr 28 12:06:02.551098 2019] [ssl:error] [pid 2960:tid 1995722752] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C4FC9BA4CC77AC57 / notbefore: Dec  7 11:23:19 2018 GMT / notafter: Dec  4 11:23:19 2028 GMT]
[Sun Apr 28 12:06:02.551140 2019] [ssl:error] [pid 2960:tid 1995722752] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sun Apr 28 12:06:02.604991 2019] [ssl:warn] [pid 2961:tid 1995722752] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 1.0.2r  26 Feb 2019, version currently loaded is OpenSSL 1.0.2q  20 Nov 2018) - may result in undefined or erroneous behavior
[Sun Apr 28 12:06:02.607160 2019] [ssl:warn] [pid 2961:tid 1995722752] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sun Apr 28 12:06:02.607709 2019] [ssl:error] [pid 2961:tid 1995722752] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C4FC9BA4CC77AC57 / notbefore: Dec  7 11:23:19 2018 GMT / notafter: Dec  4 11:23:19 2028 GMT]
[Sun Apr 28 12:06:02.607810 2019] [ssl:error] [pid 2961:tid 1995722752] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sun Apr 28 12:06:02.613772 2019] [mpm_event:notice] [pid 2961:tid 1995722752] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2q configured -- resuming normal operations
[Sun Apr 28 12:06:02.613919 2019] [core:notice] [pid 2961:tid 1995722752] AH00094: Command line: '/usr/sbin/apache2'
[Sun Apr 28 12:28:03.758788 2019] [mpm_event:notice] [pid 2961:tid 1995722752] AH00493: SIGUSR1 received.  Doing graceful restart
AH00112: Warning: DocumentRoot [/var/www/ncp-web/] does not exist
[Sun Apr 28 12:28:03.812927 2019] [ssl:warn] [pid 2961:tid 1995722752] AH01882: Init: this version of mod_ssl was compiled against a newer library (OpenSSL 1.0.2r  26 Feb 2019, version currently loaded is OpenSSL 1.0.2q  20 Nov 2018) - may result in undefined or erroneous behavior
[Sun Apr 28 12:28:03.816898 2019] [ssl:warn] [pid 2961:tid 1995722752] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sun Apr 28 12:28:03.817174 2019] [ssl:error] [pid 2961:tid 1995722752] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: C4FC9BA4CC77AC57 / notbefore: Dec  7 11:23:19 2018 GMT / notafter: Dec  4 11:23:19 2028 GMT]
[Sun Apr 28 12:28:03.817197 2019] [ssl:error] [pid 2961:tid 1995722752] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sun Apr 28 12:28:03.819430 2019] [mpm_event:notice] [pid 2961:tid 1995722752] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2q configured -- resuming normal operations
[Sun Apr 28 12:28:03.819506 2019] [core:notice] [pid 2961:tid 1995722752] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2019-04-28 12:03:18 1517425472 [Note] /usr/sbin/mysqld: Shutdown complete

2019-04-28 12:03:25 1995767808 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2019-04-28 12:03:25 1995767808 [Note] InnoDB: The InnoDB memory heap is disabled
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2019-04-28 12:03:25 1995767808 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Using Linux native AIO
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Using generic crc32 instructions
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Initializing buffer pool, size = 370.0M
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Completed initialization of buffer pool
2019-04-28 12:03:25 1995767808 [Note] InnoDB: Highest supported file format is Barracuda.
2019-04-28 12:03:26 1995767808 [Note] InnoDB: 128 rollback segment(s) are active.
2019-04-28 12:03:26 1995767808 [Note] InnoDB: Waiting for purge to start
2019-04-28 12:03:26 1995767808 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.42-84.2 started; log sequence number 1616928
2019-04-28 12:03:26 1995767808 [Note] Plugin 'FEEDBACK' is disabled.
2019-04-28 12:03:26 1132458816 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-04-28 12:03:27 1995767808 [Note] Server socket created on IP: '127.0.0.1'.
2019-04-28 12:03:27 1995767808 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.38-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>
`
I would be greatful for some help.

Regards
Tom

ovpc, 2019-04-30:
curl installer can have issues, as there is not enough manpower to keep it up to date, and things may have  changed upstream. Best use one of the images or docker to get going with ncp.

I will give curl installer a spin, later on today, to see if I can reproduce this.

closed 2019-04-30 by ovpc

oerkenrotten, 2019-04-30:
I'm checking myself now with a clean raspbian desktop and the curl
installer I'll get back with a result.

oerkenrotten, 2019-04-30:
Thankyou for your reply and for checking it out. I can confirm that it cannot reproduce the result, so I will keep my clean install instead. Thankyou for your time, and sorry for the inconvenience.