nextcloud/nextcloudpi #894: Force HTTPS

Fosphatic, 2019-04-30:
Latest NextcloudPI Version on Odroid HC1: 

Even I enable force HTTPS in the WebUI I'm still able to access the device through HTTP, even it's a blank page. However, when this switch is used I would expect that any try to access would be redirected to https.

Regards

Fossy

ovpc, 2019-04-30:
Please use the [forum](https://help.nextcloud.com/tags/ncp) to get help, orr ask questions.

Fosphatic, 2019-04-30:
> Please use the [forum](https://help.nextcloud.com/tags/ncp) to get help or ask questions.

Well, I would like, but unfortunately, the provided link seems to be not valid...

ovpc, 2019-04-30:
Yeah, forum seem to be some issues since yesterday, reported at https://github.com/nextcloud/nextcloud.com/issues/1098
Nothing we can do about it, but be patient ;-) 
You could try https://t.me/NextCloudPi in meantime.

closed 2019-04-30 by ovpc