nextcloud/nextcloudpi #897: Fresh ArmHF-Docker install: /usr/bin/letsencrypt missing

langfingaz, 2019-05-04:
On a fresh Nextcloudpi-Docker-ArmHF installation I set up today the folder `/usr/bin/letsencrypt` is missing. Therefore the _letsencrypt_ script gives the following error:

```
Running letsencrypt
/usr/local/bin/ncp/NETWORKING/letsencrypt.sh: line 57: /usr/bin/letsencrypt: No such file or directory
Done. Press any key...
```

Could this be because of the recent letsencrypt update/change?

Greetings,
Daniel

**System information**
Running nc-info
Gathering information...
NextCloudPi version  v1.10.0
NextCloudPi image    NextCloudPi_docker_03-14-19
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      3.1G/30G
rootfs usage         3.1G/30G
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   172.16.238.10
gateway              172.16.238.1
interface            eth0
certificates         none
NAT loopback         no
uptime               19min

You should run Lets Encrypt for trusted encrypted access
Done. Press any key...

nachoparker, 2019-05-04:
yes, you probably removed the container. Try updating to the latest version from ncp-web

langfingaz, 2019-05-05:
I already updated ncp via command line once yesterday, so I thought I was on the newest version. But today I got an other update and it's all working again. Thanks for the help!

langfingaz, 2019-05-05:
Only strange thing is that I did a complete new installation and Nextcloud setup and the /etc/letsencrypt folder was missing. My first thought was just that maybe there's a small bug in some Dockerfile or so.
Anyhow the automatic update fixed it ;)

closed 2019-05-06 by nachoparker