nextcloud/nextcloudpi #900: complex admin password prevents ncp-update and other functions

e2thenegpii, 2019-05-11:
I had a complex password in a docker raspberry pi install and many functions in the NCP application didn't seem to work.

For example to update the NCP app I would click the update button and the console box would dropdown but it would never show any output.  The apache error logs showed various errors in php which I couldn't make since of being related.

When I ran "docker exec -it nextcloudpi /bin/bash" and ran ncp-update I got the following:

```
# ncp-update
Downloading updates
Performing updates
jq: error: Invalid escape at line 1, column 4 (while parsing '"\'"') at <top-level>, line 1:
.params[1].value = "Pf-Dh:lf'\''\'\'''\''y^H6s]2KUA!"                             
jq: error: Invalid escape at line 1, column 6 (while parsing '"\'\'"') at <top-level>, line 1:
.params[1].value = "Pf-Dh:lf'\''\'\'''\''y^H6s]2KUA!"                                
jq: error: Invalid escape at line 1, column 4 (while parsing '"\'"') at <top-level>, line 1:
.params[1].value = "Pf-Dh:lf'\''\'\'''\''y^H6s]2KUA!"                                      
jq: 3 compile errors
```

The crazy text string you see jq trying to parse WAS my admin password.  I changed the password to something more standard and everything seemd to start working as expected.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.12.0
NextCloudPi image    NextCloudPi_docker_03-14-19
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sda sdb sdc 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      33G/3.6T
rootfs usage         33G/3.6T
swapfile             none
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    15.0.5.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               14days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
[... 41 lines cut ...]
```

[Fri May 10 03:35:53.010424 2019] [core:notice] [pid 80:tid 4146739696] AH00094: Command line: '/usr/sbin/apache2'
[Fri May 10 03:36:04.292243 2019] [proxy_fcgi:error] [pid 87:tid 3834655792] [client 192.168.21.175:41382] AH01071: Got error 'PHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\n'
[Fri May 10 03:37:01.915244 2019] [proxy_fcgi:error] [pid 87:tid 3834655792] [client 192.168.21.175:41444] AH01071: Got error 'PHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP Warning:  sizeof(): Parameter must be an array or an object that implements Countable in /var/www/ncp-web/elements.php on line 190\nPHP message: PHP War [... cut ...] r foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/elements.php on line 21\nPHP message: PHP Notice:  Undefined index: app in /var/www/ncp-web/index.php on line 219\n'

```
</details>

nachoparker, 2019-05-11:
what was the old pasword?

e2thenegpii, 2019-05-18:
Pf-Dh:lf'\''\'\'''\''y^H6s]2KUA!

nachoparker, 2019-06-01:
Fixed in `v1.13.1`. Please reopen if the issue persists.

For now it doesn't let you use `'` and `"` characters anymore

Thanks for the heads up!

closed 2019-06-01 by nachoparker