nextcloud/nextcloudpi #901: Odroid XU4 USB-Drive problem

amilopowers, 2019-05-13:
I run an Odroid XU4 with the latest NCP.
[image]

As soon as I connect a second USB-Drive even just a stick it doesn't recognize the USB-Drive with the data on. The main data-drive is powered with an external power supply though.

ovpc, 2019-05-16:
One should generally use LABEL's when formatting drives.
So automount will mount your drives as
**/media/YOURLABEL/** and create a symbolic link to it named **/media/USBdrive#**
You can edit your config file to point to /media/YOURLABEL/ncdata

It seems, adding a second or third drive when automount is enable, can move **/media/USBdrive** to **/media/USBdrive1**

I'd check content of /media/ with
> ls -lh /media/

nachoparker, 2019-05-26:
Thanks Oliver. Yes this is explained in docs.nextcloudpi.com

closed 2019-05-26 by nachoparker