nextcloud/nextcloudpi #902: NextcloudPI stuck at v1.11.2

rspierenburg, 2019-05-14:
Hi,
I'm running Nextcloud PI on an Odroid HC-1 and up until a week ago or so it was working flawlessly.  It had told me it had updated to V1.11.2 and everything seemed fine, but every day after that it kept notifying me that it updated to V1.11.2.

I then tried to use the NCP Config WebGui to update to V1.12.3 and it started the update but there was an error.

I now get an "Nextcloud is in maintenance mode" when I try to log in via the Android app and receive this error when I try to connect over the internet : "Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log."

I then tried to log in via SSH and update that way using both "sudo ncp-update" and "ncp-config" but neither worked.

I've attached the Report file below.

Please let me know if there anything I can try.

Thank you,
Robert

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
PHP Warning:  fileperms(): stat failed for /media/myCloudDrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"l8n3jS8AmFqSSSrpKOym","level":3,"time":"2019-05-14T13:35:15+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/myCloudDrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.7.0"}
{"reqId":"l8n3jS8AmFqSSSrpKOym","level":2,"time":"2019-05-14T13:35:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/myCloudDrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"15.0.7.0"}
PHP Warning:  fileperms(): stat failed for /media/myCloudDrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"l8n3jS8AmFqSSSrpKOym","level":3,"time":"2019-05-14T13:35:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/myCloudDrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.7.0"}
{"reqId":"l8n3jS8AmFqSSSrpKOym","level":3,"time":"2019-05-14T13:35:16+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#141","userAgent":"--","version":"15.0.7.0"}
NextCloudPi version  v1.11.2
NextCloudPi image    NextCloudPi_11-22-17
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          none
datadir              /media/myCloudDrive/ncdata (doesn't exist)
rootfs usage         2.4G/7.1G
swapfile             /dev/zram5
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630ca1a
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               13:06

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
PHP Warning:  fileperms(): stat failed for /media/myCloudDrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"lb2xj7gr84xtcml5tHhb","level":3,"time":"2019-05-14T13:35:19+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/myCloudDrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.7.0"}
{"reqId":"lb2xj7gr84xtcml5tHhb","level":2,"time":"2019-05-14T13:35:19+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory \/media\/myCloudDrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"15.0.7.0"}
Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/15/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:166
Stack trace:
#0 /var/www/nextcloud/console.php(95): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}PHP Warning:  fileperms(): stat failed for /media/myCloudDrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 140
{"reqId":"lb2xj7gr84xtcml5tHhb","level":3,"time":"2019-05-14T13:35:20+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for \/media\/myCloudDrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#140","userAgent":"--","version":"15.0.7.0"}
{"reqId":"lb2xj7gr84xtcml5tHhb","level":3,"time":"2019-05-14T13:35:20+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#141","userAgent":"--","version":"15.0.7.0"}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Tue May 14 06:25:35.279902 2019] [ssl:warn] [pid 1530:tid 3069676816] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 06:25:36.000903 2019] [mpm_event:notice] [pid 1530:tid 3069676816] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured -- resuming normal operations
[Tue May 14 06:25:36.000953 2019] [core:notice] [pid 1530:tid 3069676816] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/myCloudDrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

nachoparker, 2019-05-15:
you don't have any USB device connected, or is not recognized by your Odroid, that's seems to be your issue

rspierenburg, 2019-05-15:
I have a 4tb SATA drive connected, so either the drive is corrupt or the Odroid.  Its odd because up until I updated to V1.11.2 everything ran perfectly smoothly.  I will have to do some investigation when I get home tonight I think.

rspierenburg, 2019-05-15:
So I rebooted via SSH and Nextcloud loaded ok (able to see the drive again) but its still stuck at V1.11.2  below is the new NCP-Report

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.11.2
NextCloudPi image    NextCloudPi_11-22-17
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      160G/3.7T
rootfs usage         2.4G/7.1G
swapfile             /dev/zram5
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.7.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630ca1a
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               20min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.102",
            "5": "nextcloudpi.local",
            "2": "spierenburg.ca",
            "3": "www.spierenburg.ca",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "4": "www.spierenburg.ca",
            "11": "45.2.197.44"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/www.spierenburg.ca\/",
        "dbtype": "mysql",
        "version": "15.0.7.0",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 83 lines cut ...]
{"reqId":"XNFfYX8AAQEAAC6KLGYAAFAA","level":3,"time":"2019-05-07T10:35:13+00:00","remoteAddr":"45.2.197.44","user":"--","app":"index","method":"GET","url":"\/","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":511,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["11f246796a320b451d9e2d6695b94a29",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":165,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":1533,"function":"query","class":"OC\\ServerContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":697,"function":"getMemCacheFactory","class":"OC\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":165,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{ [... cut ...] ction":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":166,"function":"handleException","args":[{"__class__":"RedisException"}]}],"File":"\/var\/www\/nextcloud\/remote.php","Line":70,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"15.0.7.0"}

```
</details>

root@nextcloudpi:~#
var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\Ap

rspierenburg, 2019-05-15:
I logged into Web Admin Panel and attempted to run the V1.12.3 update again and it seems to have failed again.  Below is the most recent NCP-Report

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.11.2
NextCloudPi image    NextCloudPi_11-22-17
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      160G/3.7T
rootfs usage         2.4G/7.1G
swapfile             /dev/zram5
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.7.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e0630ca1a
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               37min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.102",
            "5": "nextcloudpi.local",
            "2": "spierenburg.ca",
            "3": "www.spierenburg.ca",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "4": "www.spierenburg.ca",
            "11": "45.2.197.44"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/www.spierenburg.ca\/",
        "dbtype": "mysql",
        "version": "15.0.7.0",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 90 lines cut ...]
```

{"reqId":"XNFfYX8AAQEAAC6KLGUAAABR","level":3,"time":"2019-05-07T10:35:13+00:00","remoteAddr":"45.2.197.44","user":"--","app":"remote","method":"GET","url":"\/status.php","message":{"Exception":"OC\\HintException","Message":"Memcache \\OC\\Memcache\\Redis not available for local cache","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":511,"function":"__construct","class":"OC\\Memcache\\Factory","type":"->","args":["11f246796a320b451d9e2d6695b94a29",{"__class__":"OC\\Log"},"\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis","\\OC\\Memcache\\Redis"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":165,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":1533,"function":"query","class":"OC\\ServerContainer","type":"->","args":["MemCacheFactory"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/Server.php","line":697,"function":"getMemCacheFactory","class":"OC\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":117,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/ServerContainer.php","line":132,"function":"query","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/lib\/private\/AppFramework\/Utility\/SimpleContainer.php","line":165,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/pimple\/pimple\/src\/Pimple\/Container.php","line":114,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive  [... cut ...] parameter replaced ***","*** sensitive parameter replaced ***"]},{"file":"\/var\/www\/nextcloud\/remote.php","line":72,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":166,"function":"handleException","args":[{"__class__":"RedisException"}]}],"File":"\/var\/www\/nextcloud\/remote.php","Line":70,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.3 (build 1) (Nextcloud)","version":"15.0.7.0"}

```
</details>

rspierenburg, 2019-05-15:
And finally this is the output if I try to update via SSH:

Downloading updates
Performing updates
Running nc-autoupdate-nc
automatic Nextcloud updates enabled
Cannot load Zend OPcache - it was already loaded
Config value squareSizes for app previewgenerator set to 32
Cannot load Zend OPcache - it was already loaded
Config value widthSizes for app previewgenerator set to 128 256 512
Cannot load Zend OPcache - it was already loaded
Config value heightSizes for app previewgenerator set to 128 256
Cannot load Zend OPcache - it was already loaded
System config value jpeg_quality set to string 60
Running unattended-upgrades
Unattended upgrades active: yes (autoreboot true)
--2019-05-15 13:06:44--  https://packages.sury.org/php/apt.gpg
Resolving packages.sury.org (packages.sury.org)... 104.31.95.169, 104.31.94.169, 2606:4700:30::681f:5ea9, ...
Connecting to packages.sury.org (packages.sury.org)|104.31.95.169|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1769 (1.7K) [application/octet-stream]
Saving to: ‘/etc/apt/trusted.gpg.d/php.gpg’

/etc/apt/trusted.gpg.d/p 100%[==================================>]   1.73K  --.-KB/s    in 0s

2019-05-15 13:06:44 (11.6 MB/s) - ‘/etc/apt/trusted.gpg.d/php.gpg’ saved [1769/1769]

Running nc-autoupdate-ncp
automatic NextCloudPi updates enabled
Running nc-notify-updates
update web notifications enabled
Installing nc-backup
Hit:1 http://security.debian.org stretch/updates InRelease
Hit:4 https://packages.sury.org/php stretch InRelease
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:6 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:7 http://cdn-fastly.deb.debian.org/debian stretch Release
Hit:3 https://apt.armbian.com stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
pigz is already the newest version (2.3.4-1).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Installing nc-restore
Hit:1 http://security.debian.org stretch/updates InRelease
Hit:2 https://packages.sury.org/php stretch InRelease
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:6 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:7 http://cdn-fastly.deb.debian.org/debian stretch Release
Hit:4 https://apt.armbian.com stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 file : Depends: libmagic1 (= 1:5.30-1+deb9u2) but 1:5.34-2 is to be installed
E: Unable to correct problems, you have held broken packages.

Medvosteri, 2019-05-15:
Looks to be similar to this:

https://github.com/nextcloud/nextcloudpi/issues/896

Which seems to be fixed by doing:

sudo apt-get install libmagic1=1:5.30-1+deb9u2 libmagic-mgc=1:5.30-1+deb9u2

rspierenburg, 2019-05-15:
Thank you so much.  That's exactly what fixed it.

closed 2019-05-15 by rspierenburg