nextcloud/nextcloudpi #903: Nextcloudpi Image for Odroid HC-1 Update 1.12.3

Fosphatic, 2019-05-14:
Nextcloud Image for Odroid HC updated to version 1.12.3… Well the party is over for me as apache2 keeps crashing all the time without any reason somehow. At least I can’t find a thing in the log that could bring some help.

Nextcloud Report brings this:

NextCloudPi version v1.12.3
NextCloudPi image NextCloudPi_03-09-19
distribution Debian GNU/Linux 9 \n \l
automount yes
USB devices sda
datadir /media/USBdrive/nextcloud/data
data in SD no
data filesystem ext2/ext3
data disk usage 53G/688G
rootfs usage 4.0G/30G
swapfile /var/swap
dbdir /var/lib/mysql
Nextcloud check ok
Nextcloud version 16.0.0.9
HTTPD service down
PHP service up
MariaDB service up
Redis service up
Postfix service up
internet check ok
port check 80 closed
port check 443 closed
IP REMOVED SENSITIVE VALUE
gateway REMOVED SENSITIVE VALUE
interface enx001e0636626c
certificates REMOVED SENSITIVE VALUE
NAT loopback no
uptime 11:55

</details>

<details>
<summary>Nextcloud configuration</summary>

{
“system”: {
“passwordsalt”: “REMOVED SENSITIVE VALUE”,
“secret”: “REMOVED SENSITIVE VALUE”,
“trusted_domains”: {
“0”: “localhost”,
“5”: “nextcloudpi.local”,
“7”: “nextcloudpi”,
“8”: “nextcloudpi.lan”,
“21”: “",
“22”: "”
},
“datadirectory”: “REMOVED SENSITIVE VALUE”,
“dbtype”: “mysql”,
“version”: “16.0.0.9”,
“overwrite.cli.url”: “REMOVED SENSITIVE VALUE manually…”,
“dbname”: “REMOVED SENSITIVE VALUE”,
“dbhost”: “REMOVED SENSITIVE VALUE”,
“dbport”: “”,
“dbtableprefix”: “oc_”,
“mysql.utf8mb4”: true,
“dbuser”: “REMOVED SENSITIVE VALUE”,
“dbpassword”: “REMOVED SENSITIVE VALUE”,
“installed”: true,
“instanceid”: “REMOVED SENSITIVE VALUE”,
“memcache.local”: “\OC\Memcache\Redis”,
“memcache.locking”: “\OC\Memcache\Redis”,
“redis”: {
“host”: “REMOVED SENSITIVE VALUE”,
“port”: 0,
“timeout”: 0,
“password”: “REMOVED SENSITIVE VALUE”
},
“tempdirectory”: “/media/USBdrive/nextcloud/data/tmp”,
“mail_smtpmode”: “smtp”,
“mail_smtpauthtype”: “LOGIN”,
“mail_from_address”: “REMOVED SENSITIVE VALUE”,
“mail_domain”: “REMOVED SENSITIVE VALUE”,
“overwriteprotocol”: “https”,
“htaccess.RewriteBase”: “/”,
“mail_sendmailmode”: “smtp”,
“mail_smtpsecure”: “ssl”,
“mail_smtpauth”: 1,
“mail_smtphost”: “REMOVED SENSITIVE VALUE”,
“mail_smtpport”: “465”,
“mail_smtpname”: “REMOVED SENSITIVE VALUE”,
“mail_smtppassword”: “REMOVED SENSITIVE VALUE”,
“maintenance”: false,
“logfile”: “/media/USBdrive/nextcloud/data/nextcloud.log”,
“theme”: “”,
“loglevel”: 2,
“jpeg_quality”: “60”,
“app_install_overwrite”: [
“tasks”,
“nextcloudpi”
]
}
}

</details>

<details>
<summary>HTTPd logs</summary>

[Tue May 14 09:00:20.791116 2019] [ssl:warn] [pid 1126:tid 3069562000] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 09:00:21.062778 2019] [ssl:warn] [pid 1312:tid 3069562000] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Tue May 14 09:00:22.006021 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured – resuming normal operations
[Tue May 14 09:00:22.006278 2019] [core:notice] [pid 1312:tid 3069562000] AH00094: Command line: ‘/usr/sbin/apache2’
[Tue May 14 09:02:09.769906 2019] [proxy_fcgi:error] [pid 1738:tid 2782680112] [client 192.168.218.226:55716] AH01071: Got error ‘PHP message: PHP Notice: Undefined index: app in /var/www/ncp-web/index.php on line 238\nPHP message: PHP Notice: Undefined index: app in /var/www/ncp-web/index.php on line 244\n’
[Tue May 14 09:03:28.074027 2019] [proxy_fcgi:error] [pid 1738:tid 2799481904] [client 192.168.218.226:55716] AH01067: Failed to read FastCGI header
[Tue May 14 09:03:28.074176 2019] [proxy_fcgi:error] [pid 1738:tid 2799481904] (104)Connection reset by peer: [client 192.168.218.226:55716] AH01075: Error dispatching request to :4443:
[Tue May 14 09:03:28.074642 2019] [proxy_fcgi:error] [pid 1738:tid 2950616112] [client 192.168.218.226:55716] AH01067: Failed to read FastCGI header
[Tue May 14 09:03:28.074738 2019] [proxy_fcgi:error] [pid 1738:tid 2950616112] (104)Connection reset by peer: [client 192.168.218.226:55716] AH01075: Error dispatching request to :4443:
[Tue May 14 09:09:16.276441 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00493: SIGUSR1 received. Doing graceful restart
[Tue May 14 09:09:17.000841 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured – resuming normal operations
[Tue May 14 09:09:17.000881 2019] [core:notice] [pid 1312:tid 3069562000] AH00094: Command line: ‘/usr/sbin/apache2’
[Tue May 14 09:09:36.920789 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00493: SIGUSR1 received. Doing graceful restart
[Tue May 14 09:09:37.000214 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2r configured – resuming normal operations
[Tue May 14 09:09:37.000253 2019] [core:notice] [pid 1312:tid 3069562000] AH00094: Command line: ‘/usr/sbin/apache2’
[Tue May 14 12:25:59.725566 2019] [core:warn] [pid 1312:tid 3069562000] AH00045: child process 1738 still did not exit, sending a SIGTERM
[Tue May 14 12:26:01.727750 2019] [core:warn] [pid 1312:tid 3069562000] AH00045: child process 1738 still did not exit, sending a SIGTERM
[Tue May 14 12:26:03.729899 2019] [core:warn] [pid 1312:tid 3069562000] AH00045: child process 1738 still did not exit, sending a SIGTERM
[Tue May 14 12:26:05.731891 2019] [core:error] [pid 1312:tid 3069562000] AH00046: child process 1738 still did not exit, sending a SIGKILL
[Tue May 14 12:26:06.733420 2019] [mpm_event:notice] [pid 1312:tid 3069562000] AH00491: caught SIGTERM, shutting down

</details>

<details>
<summary>Database logs</summary>

2019-05-14 8:59:32 3044238144 [Note] /usr/sbin/mysqld: Shutdown complete

2019-05-14 9:00:40 3069877040 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2019-05-14 9:00:40 3069877040 [Note] InnoDB: The InnoDB memory heap is disabled
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2019-05-14 9:00:40 3069877040 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Compressed tables use zlib 1.2.8
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Using Linux native AIO
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Using generic crc32 instructions
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Initializing buffer pool, size = 798.0M
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Completed initialization of buffer pool
2019-05-14 9:00:40 3069877040 [Note] InnoDB: Highest supported file format is Barracuda.
2019-05-14 9:00:41 3069877040 [Note] InnoDB: 128 rollback segment(s) are active.
2019-05-14 9:00:41 3069877040 [Note] InnoDB: Waiting for purge to start
2019-05-14 9:00:41 3069877040 [Note] InnoDB: Percona XtraDB (http://www.percona.com) 5.6.42-84.2 started; log sequence number 1651119639
2019-05-14 9:00:41 1715467072 [Note] InnoDB: Dumping buffer pool(s) not yet started
2019-05-14 9:00:41 3069877040 [Note] Plugin ‘FEEDBACK’ is disabled.
2019-05-14 9:00:41 3069877040 [Note] Server socket created on IP: ‘127.0.0.1’.
2019-05-14 9:00:41 3069877040 [Note] /usr/sbin/mysqld: ready for connections.
Version: ‘10.1.38-MariaDB-0+deb9u1’ socket: ‘/var/run/mysqld/mysqld.sock’ port: 3306 Debian 9.8

</details>

<details>
<summary>Nextcloud logs</summary>

Anyone here got the same issue and a workaround probably? It’s unusable for me this way and I’m already considering to do the oldschool manual way with nginx and no issues like i did in the past.

Regards

Fossy

p.s. sorry but I’m actually disappointed slighlty because the project was looking promising at all.

markobremer, 2019-05-15:
@Fosphatic My NCP instance in HC-1 also crashed a few days ago, and since then is not working at all. Maybe we're having the same issue?

`NextCloudPi version  v1.12.3
NextCloudPi image    NextCloudPi_03-09-19
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      199G/932G
rootfs usage         3.9G/15G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.87
gateway              192.168.1.1
interface            enx001e06328ef9
certificates         markobremer.ddns.net
NAT loopback         no
uptime               30days`

markobremer, 2019-05-15:
Nevermind... I rebooted and it worked :'D

Fosphatic, 2019-05-15:
@markobremer Hi Marko, well it could be same issue just check if it is stable. In my case i did before reboots too, but realized that just apache 2 service is goind down always after a while and i dont see the reason for that. If i start it manually it works again for several hours then the same happens again. I had no such issues with older nextcloudpi versions and on my old dev board with older debian image and nginx as well, was most stable thing ever. However I hope that nachoparker or someone from the team can localize, reproduce and fix this problem as it is very frustrating actually...

Fosphatic, 2019-05-20:
@nachoparker 

Diving deeper into the problem with apache2 brought some enlightment into the darkness,

Would you please adopt the solution for odroid hc-2 / 1 image?

Seems to be that you forgot that php mpm_prefork module isn't compatible with http2 module at all. The warnings in the apache.log were telling exactly that. For me it seems to be that the solution is to switch to php-fpm. For that i needed to disable this mpm_prefork mod as php7.2.

Running on root console:

a2dismod php7.0
a2dismod mpm_prefork

If I was doing that apache2 denied to startup again. Well the solution was pretty simple:

running on root console:

apt-get install libapache2-mod-fcgid
systemctl restart apache2

Well if anything goes wrong check if following php mods are enabled and restart apache2 again:

a2enmod mpm_event
a2enmod actions

Regards

Fossy

p.s. would be really kind of you if you could adopt this into next updates I would dislike to do that on each nextcloudpi version...

nachoparker, 2019-05-26:
Unless there is any strong indication that this is not a HW issue, closing this.

Thanks for reporting

closed 2019-05-26 by nachoparker

Fosphatic, 2019-05-26:
This device is not even half year old... Hardware issue really? I'm sure if I setup everything manually the device will run fine as it did before latest uptade. But seems ti be that you don't want to support anything in your image and everything is hardware issue...

Fosphatic, 2019-11-11:
@nachoparker since I opened the ticket and you were not even willing to look into the issue. I have formated the sd card and installed armbian and nextcloud as did before using your image. Well since then the machine runs and runs and runs and runs. Actuall with version 17.0.1 and still no issue, your this must be a hardware issue claim is wrong as i do not change any hardware and it runs fine.