nextcloud/nextcloudpi #904: Docker - nextcloudpi panel automatically auto-reloads to "Something went wrong. Try refreshing the page."

svenb1234, 2019-05-16:
When I open nextcloudpi web panel it starts to load the system information. But after five seconds it reloads the page and the right-hand side panel shows "Something went wrong. Try refreshing the page."

This started after the last update to nextcloudpi.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.12.3
NextCloudPi image    NextCloudPi_docker_03-14-19
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sdc
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.3T/3.2T
rootfs usage         2.3T/3.2T
swapfile             /share/CACHEDEV1_DATA/.swap/qnap_swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               55min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.178.36",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.6.1",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 95 lines cut ...]
```

{"reqId":"XNCO18CosiQAAABVAlUAAABJ","level":4,"time":"2019-05-06T19:45:28+00:00","remoteAddr":"192.168.178.23","user":"sven","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 1853766 got 647168","Code":0,"Trace":[{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":39,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["0000000000000000-0000000001853766",null]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["0000000000000000-0000000001853766",null]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpPut"],[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:PUT",[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/65d9190cf6e0a4987c4e2307962dfa68\/0000000000000000-0000000001853766","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/data\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/data\/nextcloud\/remote.php","line":163,"args":["\/data\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/data\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":222,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.6.0","version":"15.0.6.1"}
{"reqId":"XNCd08CosiQAAABUDvcAAAAE","level":3,"time":"2019-05-06T20:49:26+00:00","remoteAddr":"192.168.178.23","user":"sven","app":"no app in context","method":"PUT","url":"\/remote.php\/dav\/uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 10240000 got 8314880","Code":0,"Trace":[{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Upload\/UploadFolder.php","line":39,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["0000000010240000-0000000020479999",null]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Upload\\UploadFolder","type":"->","args":["0000000010240000-0000000020479999",null]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpPut"],[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:PUT",[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/data\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/nextcloudpi\/remote.php\/dav\/uploads\/sven\/87c945bb857c88dba6e0e299e6b177ce\/0000000010240000-0000000020479999","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/data\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/data\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/data\/nextcloud\/remote.php","line":163,"args":["\/data\/nextcloud\/apps\/dav\/appinfo\/v2\ [... cut ...] ,"args":[]},{"file":"\/data\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/data\/nextcloud\/remote.php","line":163,"args":["\/data\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/data\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":222,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.6.1","version":"15.0.6.1"}

```
</details>

nachoparker, 2019-05-17:
Thanks for reporting.

It would be useful to see the network and console logs in the browser dev tools.

svenb1234, 2019-05-20:
Network: See attachment. ncp-output.php goes on forever with "content download".

Console:
ncp.js:727 Refused to load the font 'https://nextcloudpi:4443/core/fonts/OpenSans-Light.woff' because it violates the following Content Security Policy directive: "default-src 'none'". Note that 'font-src' was not explicitly set, so 'default-src' is used as a fallback.

(anonymous) @ ncp.js:727
N @ minified.js:1
(anonymous) @ minified.js:1
(anonymous) @ minified.js:1
p @ minified.js:1
v @ minified.js:1
O @ minified.js:1
(anonymous) @ minified.js:1
ncp.js:727 Refused to load the font 'https://nextcloudpi:4443/core/fonts/OpenSans-Semibold.woff' because it violates the following Content Security Policy directive: "default-src 'none'". Note that 'font-src' was not explicitly set, so 'default-src' is used as a fallback.

(anonymous) @ ncp.js:727
N @ minified.js:1
(anonymous) @ minified.js:1
(anonymous) @ minified.js:1
p @ minified.js:1
v @ minified.js:1
O @ minified.js:1
(anonymous) @ minified.js:1
minified.js:1 XHR finished loading: POST "https://nextcloudpi:4443/ncp-launcher.php".
request @ minified.js:1
print_dashboard @ ncp.js:120
(anonymous) @ ncp.js:736
N @ minified.js:1
(anonymous) @ minified.js:1
(anonymous) @ minified.js:1
p @ minified.js:1
v @ minified.js:1
O @ minified.js:1
(anonymous) @ minified.js:1
minified.js:1 XHR finished loading: POST "https://nextcloudpi:4443/ncp-launcher.php".
request @ minified.js:1
print_backups @ ncp.js:277
success @ ncp.js:131
N @ minified.js:1
e @ minified.js:1
(anonymous) @ minified.js:1
p @ minified.js:1
(anonymous) @ minified.js:1
setTimeout (async)
i.fire @ minified.js:1
f.onreadystatechange @ minified.js:1
XMLHttpRequest.send (async)
request @ minified.js:1
print_dashboard @ ncp.js:120
(anonymous) @ ncp.js:736
N @ minified.js:1
(anonymous) @ minified.js:1
(anonymous) @ minified.js:1
p @ minified.js:1
v @ minified.js:1
O @ minified.js:1
(anonymous) @ minified.js:1

[nextcloudpi.zip](https://github.com/nextcloud/nextcloudpi/files/3199248/nextcloudpi.zip)

nachoparker, 2019-05-26:
I don't see anything and can't reproduce unfortunately. Please reopen if anybody have more info on this.

closed 2019-05-26 by nachoparker

Tommy-01, 2021-02-02:
I got the same Issue. I use a nextcloudpi Container on an Qnap NAS.
At first install it worked without Problem. But at some time the behavior, as described above starts.
Even an new Container shows the same at some point.
To update the Nextcloudpi i only use the update functionality from the WebPanel and not the Update functionality that is integrated within Nextcloud.

fifi634, 2024-03-30:
Hi,
I have the same problem on my nextcloudpi installed on Raspberry Pi 4. I haven't issue of this one, but i think this probleme is caused by auto upload fonction in administration nextcloudpi panel. I think is that because I have try it recently and I have this problem. In a previous installation I don't use this auto upload function and I have never see this problem, I see it only when I use this function.
I hope this response help you on for a issue. 
Thank for this program, have a nice day.