nextcloud/nextcloudpi #910: Udate nextcloudpi doesn't work v1.12.0 to v1.12.7

kmpfwrst, 2019-05-26:
Trying to update nextcloudpi from v1.12.0 to v1.12.7 doesn't work.

`sudo ncp-update`
`Downloading updates
Performing updates
Running nc-autoupdate-nc
automatic Nextcloud updates enabled
Config value squareSizes for app previewgenerator set to 32
Config value widthSizes for app previewgenerator set to 128 256 512
Config value heightSizes for app previewgenerator set to 128 256
System config value jpeg_quality set to string 60
Running unattended-upgrades
Unattended upgrades active: yes (autoreboot true)
--2019-05-26 10:05:15--  https://packages.sury.org/php/apt.gpg
Resolving packages.sury.org (packages.sury.org)... 104.31.95.169, 104.31.94.169,                 2606:4700:30::681f:5ea9, ...
Connecting to packages.sury.org (packages.sury.org)|104.31.95.169|:443... connec                ted.
HTTP request sent, awaiting response... 200 OK
Length: 1769 (1.7K) [application/octet-stream]
Saving to: ‘/etc/apt/trusted.gpg.d/php.gpg’

/etc/apt/trusted.gp 100%[===================>]   1.73K  --.-KB/s    in 0s

2019-05-26 10:05:16 (4.96 MB/s) - ‘/etc/apt/trusted.gpg.d/php.gpg’ saved [1769/1                769]

Running nc-scan-auto
/usr/local/bin/ncp/CONFIG/nc-scan-auto.sh: line 24: npc/files/Documents: divisio                n by 0 (error token is "files/Documents")
`

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.12.0
NextCloudPi image    NextCloudPi_01-14-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      92G/932G
rootfs usage         2.2G/30G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    15.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

[... 116 lines cut ...]
```

{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Move .step file of updater to backup location","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Fix potential broken mount points","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No mounts updated","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair invalid paths in file cache","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add log rotate job","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Clear frontend caches","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:28+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Image cache cleared","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:30+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: SCSS cache cleared","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: JS cache cleared","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Clear every generated avatar on major updates","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add preview background cleanup job","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to cleanup old backups of the updater","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair pending cron jobs","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: No need to repair pending cron jobs.","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Extract the vcard uid and store it in the db","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:31+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:55+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"15.0.5.3"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:55+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"15.0.6.1"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:55+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"15.0.6.1"}
{"reqId":"fZxdRjkOsuHgqwRFXrGR","level":1,"time":"2019-05-26T08:14:55+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"15.0.6.1"}

```
</details>

ovpc, 2019-05-26:
Update to 1.12.7 working fine on my instances. Looks like there is something else wrong with yours, Please go to [forum](https://help.nextcloud.com/tags/ncp) for help. 
Closing this, feel free to comment here and reopen if turns out to be a bug.

closed 2019-05-26 by ovpc

nachoparker, 2019-05-26:
I think you put something invalid in "SCANINTERVAL" in ncp-web  -> nc-scan-auto. If that doesn't help please use the forums :)

kmpfwrst, 2019-05-27:
you are totally right - that was the issue. Thank you for the help and ncp!