nextcloud/nextcloudpi #911: /etc/cron.d/ncp-notify-updates spam mails

Letterus, 2019-05-27:
/etc/cron.d/ncp-notify-updates is reset after each update of ncp.

I changed /etc/cron.d/ncp-notify-updates to output stdout to /dev/null so I do not get a mail that nothing was updated every hour (this is quiet annoying). Maybe you could make this the default? Or do you have another way to solve this?

Thank you!

ovpc, 2019-05-27:
Same here, I insert 

  MAILTO=""

To avoid hourly mail, after every update to ncp and /etc/cron.d/ncp-notify-updates

Assumed I was the only one having that, and that it is because I use ssmtp and bsd-mailx for sending mail and postfix is removed in that process, and postfix generally not configured to send mail for root.

nachoparker, 2019-05-28:
Hi,

Please run `sudo ncp-update devel` to test `v1.12.9` and see if that fixes it.

ovpc, 2019-05-28:
Thanks, v1.12.9 does the trick for me ;-)

closed 2019-06-01 by nachoparker