nextcloud/nextcloudpi #912: NCP on Rock64 w 4GB Slow after update to 1.12.7 and wont update to 1.12.9

xelcho, 2019-05-28:
**System information**

https://pastebin.com/SEjLvKT6

ovpc, 2019-05-28:
Looks like your are ports closed, please go to forum for help. Unless you are reporting a bug.
v1.12.9 is still in devel, but works and updates fine here. So it should not update unless one appends devel to ncp-update.
Also you are using an old version 14.0.3 of NC and older image of NCP NextCloudPi_09-23-18, better use the most recent one 03-09-19.

xelcho, 2019-05-28:
I know it looks like that but I am using custom ports. The ports are not an issue.

closed 2019-05-28 by ovpc

xelcho, 2019-05-28:
Please explain why you closed this issue? 
Bearing in mind the following:
The ports are not an issue.
I am not seeking support.

ovpc, 2019-05-28:
Anyone is free to re-open this if needed. And if it turns out be a bug.
I closed it because there seems to be no indication that you are reporting a bug.
Can you reproduce the issue? Can someone else reproduce it too?
You being concerned about your users running into cert issues, is maybe understandable, but not a bug!  Also no other reports, about v1.12.7 slowing things down, have been made, and I have 4 instances up and running, without any issues!

xelcho, 2019-05-28:
Fair enough. :)
After testing the db and finding it all good, I'll go with your suggestion of moving to a current image and restoring db and data.
Thanks

xelcho, 2019-05-30:
To close this out and provide some semblance of an RCA:
If you are running linux as a desktop/vm (im on ubuntu 18.04 kde, but gnome users are having the same experience) and employ the nextcloud client watch-out!
Client package (From Nextcloud or distro) (Linux only): 2.5.2git version sux. Too many issues to list, suffice it to say that it crippled the nextcloud server with it's ill formed requests etc. 
See issue here:
https://github.com/nextcloud/desktop/issues/1264

This is why I made the bad assumption that the issue was associated with the server side.
If you are in this situation do the following to regain stability:
remove client and the ppa repo until it is updated.
download and use the appimage (2.5.0):
https://download.nextcloud.com/desktop/releases/Linux/Nextcloud-2.5.0-x86_64.AppImage

My bad, thanks again Oliver!