nextcloud/nextcloudpi #914: Auto-backup does not block due to space check failed

diningphil, 2019-05-30:
**Description**  
It would be useful to have automatic deletion of old backups (except for the most recent one) when there is not enough space on disk for more backups and the backup limit exceeds the number of current backups on disk. If this is not possible, the user should be warned that backup disk capacity is becoming an issue.

**Example**  
I set the backup limit to a very big number e.g. 100, so that at a certain point the disk will be completely filled with, let's assume, 10 daily backups.
Now, if I look into the `nc-backup.sh` code, only the oldest backup is removed after the space check successfully completes. However, if the space check fails, the oldest backups are not deleted and the automatic backup process will stop forever unless I manually delete older backups.

I think we should have a routine that eliminates the minimum number of older backups (but never removing the most recent one) so that the new backup will fit on disk. If this is still not possible, the user should be warned that backup disk capacity is becoming an issue.

What do you think?

nachoparker, 2019-06-01:
If you are using `nc-backup-auto` you should be receiving a notification in Nextcloud whenever a backup fails.

I am not very convinced about automatically removing old backups ignoring the `LIMIT` setting. It sounds to me like you should adjust a lower limit when you receive the notification.

Unless we come up with a better idea :D

nachoparker, 2019-06-01:
I'll close but we can keep talking

closed 2019-06-01 by nachoparker