nextcloud/nextcloudpi #917: Connection blocked by ufw

musklor, 2019-06-03:
Hi, when UFW is enabled, I loose sometimes my connexion and I can't connect to my rpi (ssh, website etc...).
It is refused.
When UFW is disabled, no more issues.
UFW.log is empty.

nachoparker, 2019-06-04:
anything in the fail2ban logs?

musklor, 2019-06-04:
Unit fail2ban.service has begun starting up. May 30 15:03:34 nextcloudpi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:ad SRMay 30 15:03:40 nextcloudpi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e: SRMay 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING Wrong value for 'dbpurgeage' in 'Definition'. Using default one: '86400' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING 'backend' not defined in 'ssh'. Using default one: 'auto' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING 'backend' not defined in 'nextcloud'. Using default one: 'auto' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: ERROR Found no accessible config files for 'filter.d/ufwban' under /etc/fail2ban May 30 15:03:44 nextcloudpi fail2ban-client[24290]: ERROR No section: 'Definition' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: ERROR Unable to read the filter May 30 15:03:44 nextcloudpi fail2ban-client[24290]: ERROR Errors in jail 'ufwban'. Skipping... May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING 'filter' not defined in 'sshd'. Using default one: '' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING No filter set for jail sshd May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING 'backend' not defined in 'sshd'. Using default one: 'auto' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: WARNING 'filter' not defined in 'sshd'. Using default one: '' May 30 15:03:44 nextcloudpi fail2ban-client[24290]: ERROR Failed during configuration: Bad value substitution: option 'action' in section 'sshd'May 30 15:03:44 nextcloudpi systemd[1]: fail2ban.service: Control process exited, code=exited status=255 May 30 15:03:44 nextcloudpi systemd[1]: Failed to start Fail2Ban Service. -- Subject: Unit fail2ban.service has failed -- Defined-By: systemd -- Support: https://www.debian.org/support -- -- Unit fail2ban.service has failed. -- -- The result is failed. May 30 15:03:44 nextcloudpi systemd[1]: fail2ban.service: Unit entered failed state. May 30 15:03:44 nextcloudpi systemd[1]: fail2ban.service: Failed with result 'exit-code'

musklor, 2019-06-04:
I will make a new test with the last version of nextcloudpi

musklor, 2019-06-05:
No more issues.

closed 2019-06-05 by musklor