nextcloud/nextcloudpi #919: kernel error

sirdrakeistda, 2019-06-08:
I tried to run the newest image NextCloudPi_RPi_03-09-19.img on the Raspberry Pi B and got some kernel errors. 
[image]
After the kernel errors there came up another error:
Could not expand filsystem, please try raspi-config or rc-gui.
Could not determine partitions.

I tried to type something, but it didn't reacted at all. Could you also provide older images, because they work for me.

nachoparker, 2019-06-08:
Hi,

The Raspberry Pi B is not supported. Thanks for reporting.

closed 2019-06-08 by nachoparker

sirdrakeistda, 2019-06-08:
Could you provide also older Images, because they still work on the Raspberry Pi B. Updating then to the newest version works too. 
It is hard to find which devices are supported. It would be better to have a list on the main github Page:
https://github.com/nextcloud/nextcloudpi

nachoparker, 2019-06-08:
> It is hard to find which devices are supported. It would be better to have a list on the main github Page:
https://github.com/nextcloud/nextcloudpi

Good suggestion, thanks. Will do

> Could you provide also older Images

No, sorry, updates don't work and everything outdated is insecure so I won't post it for downloading. I suggest that you get a more powerful device (not necessarily a pi)

sirdrakeistda, 2019-06-08:
Updates work for an old image. I have taken the image NextCloudPi_RPi_06-27-18.img (NextcloudPi v0.57.1) and updated then to the newest version (Nextcloudpi v1.13.1) on the Raspberry Pi B. The only thing to do is to install jq, but thats all. So I wouldn't say it's insecure. There is only a Problem for me if I take directly the newest Image.