nextcloud/nextcloudpi #931: Destination Directory in nc-backup

ratzdi, 2019-06-26:
The error message "path doesn't exist" confuses me when I start nc-backup to a non existing path. 

As I understand even when the path is valid but does not exist the backup still executes. For me this is very confusing because I assume that when the path does not exists the procedure should fail and the error message "path doesn't exist" is fine. But, as I said, the procedure still executes. 

I think, it would be less confusing when the system checks the path and warns user about non existing path without executing the backup procedure. Additionally, the system can offer over a checkbox to auto-create the non existing path.

Zakspeed, 2019-07-11:
and to verify routes by ssh, type?
`root@192.168.0.200:/home/Backup`

PascalHeynmoeller, 2020-03-06:
Yes truely irritating the false report of non existing path which does exist. Could be a Regex parser error?
Stumbled up on this after checking, if after the backup restore all settings are fine and I will have future backups.