nextcloud/nextcloudpi #932: ncp-backup-auto disables server access

Hugogerva, 2019-06-27:
So, I have ncp-backup-auto toggled but when the automatic backup is launched, the server seems to have an error, and the backup doesn't finish. 

Nextcloud becomes inaccessible, so I have to ssh locally to my server to reboot it. Once it's rebooted it becomes stuck in maintenance mode and ssh becomes unavailable.
Then I have to reboot it manually (plug and unplug, I bet it's bad practice), ssh to it and disable maintenance mode.

Two things are strange in the report below: that "Cannot load Zend OPCache" message and the fact that even though I have my data in my drive, the field "data in sd" returns "no".

PS: I can't really tell for sure if normal manual backups do work, as well...

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.13.3
NextCloudPi image    NextCloudPi_03-09-19
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      8.2G/932G
rootfs usage         2.7G/30G
swapfile             /var/swap
dbdir                /media/myCloudDrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.8.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               17:47

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.1.100",
            "4": "nuagerva.ddns.net",
            "11": "2a00:5881:4008:5801:8e6b:b1bf:bb3c:d7af"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "15.0.8.1",
        "overwrite.cli.url": "https:\/\/nuagerva.ddns.net\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/myCloudDrive\/ncdata\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/media\/myCloudDrive\/ncdata\/nextcloud.log",
        "loglevel": "2",
        "log_type": "file",
        "jpeg_quality": "60",
        "theme": ""
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Jun 27 06:25:03.396208 2019] [ssl:warn] [pid 849:tid 1996329200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Thu Jun 27 06:25:04.000886 2019] [mpm_event:notice] [pid 849:tid 1996329200] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2r configured -- resuming normal operations
[Thu Jun 27 06:25:04.000995 2019] [core:notice] [pid 849:tid 1996329200] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

nachoparker, 2019-06-27:
Weird to explain behavior is often a hardware issue (power supply, SD card). You can check `/var/log/ncp.log` for more clues.

Hugogerva, 2019-06-28:
At some point on the log I get this:
[ nc-backup ]
Cannot load Zend OPcache - it was already loaded
check free space...
Cannot load Zend OPcache - it was already loaded
Maintenance mode enabled
backup database...
backup files...
[ nc-limits ]
Using 728300544B for PHP
Using 4 PHP threads
Using 388426956 memory for the database

There's a weird syntax that seems to indicate the backup didn't go all the way through...

I did change the power supply of the server not so long ago, since the old one stopped working... The new one has pretty much the same characteristics as the old one, except for the amperage: 2A instead of 3A, but I checked on the internet and people seemed to say that it was still enough to power a raspberry...
Should I get another one ?

Co0p3r21, 2019-07-02:
I think I may have the same problem. I can´t even access my Harddrive (I have attatched). Can you confirm this? If not I´m sorry to interrupt you with my problem.

funkmastergst, 2019-07-02:
Regarding the error "Cannot load Zend OPcache - it was already loaded" I resolved this with
`sudo rm   /etc/php/7.2/fpm/conf.d/20-opcache.ini` and
`sudo rm   /etc/php/7.2/cli/conf.d/20-opcache.ini`
because in each of the above directories also exists a file named "10-opcache.ini". Probably the same module is loaded twice. The error reoccurs after every PHP update...

nachoparker, 2019-07-03:
>  The error reoccurs after every PHP update...

Just a warning. Not much we can do about it. It's a stupid debian bug but it's safe to ignore

nachoparker, 2019-07-03:
@Hugogerva 

https://ownyourbits.com/2019/02/02/whats-wrong-with-the-raspberry-pi/

closed 2019-07-03 by nachoparker

Hugogerva, 2019-07-10:
Hi, so a few days ago I purchased the official power supply, plugged it in and I activated ncp-backup-auto once again.
Turns out I still get the same error: the server becomes inaccessible by https and ssh, I have to unplug and plug it in again and then disable maintenance mode to make it accessible.
I wonder if it has anything to do with that "data in sd: no" parameter...

By the way, it had no issue with performing a backup when updating to the new ncp, or even nextcloud version (even though these backups aren't stored at the same path than my other backups.