nextcloud/nextcloudpi #933: Support Raspberry Pi 4

Larsene, 2019-06-28:
The new Raspberry 4 does not even boot on the lastest available Nextcloudpi image (NextCloudPi_RPi_03-09-19)

The Raspberry Pi 4 is a great improvement, specially this usage (USB3, real gigabit interface, more ram).

Thanks !

VinceMasuka, 2019-06-28:
This is probably due to that the Raspberry Pi 4 is not supposed to work under stretch (that is the base of the current nextcloudpi image). They did the development under buster and just saw no sense in making this work with stretch, as buster is just ready to release.

So the Raspberry Pi 4 requires Raspbian buster.

Actually I'm trying to get this run under buster, too, but there are several issues while installing it (although some guy over here [https://canox.net/2016/06/die-eigene-cloud-mit-dem-raspberry-pi-und-nextcloud/](url) says he's able to run it), leaving me with no luck...

nachoparker, 2019-06-28:
Yes, NCP runs over Stretch so we will have to wait until we move NCP to Buster.

This entails other changes, such as moving to PHP7.3 so we'll do everything at once when NC16.0.2 comes out (expected July).

For the time being it can be run using the docker container.

misanthrophy, 2019-07-03:
I have a Pi4 4GB and with some workarounds I was able to install docker on Buster. The latest docker image for nextcloudpi works just fine. it installs nextcloud v15

You can also update the nextcloudpi panel and nextcloud (v16) to the latest version and it still works, it just has some issues with indicies, but the rest works. But the main problem right now is the docker image which does not work out of the box with the standard install command/script.

nachoparker, 2019-07-03:
I have a preliminary image but haven't tested it on a pi yet. Will upload it later

nachoparker, 2019-07-05:
Here folks

https://ownyourbits.com/downloads/testing/NextCloudPi_RPi_07-03-19/

Only thing I detected is that there is no PHP7.3 `php-smbclient` in Raspbian (at least yet).

nachoparker, 2019-07-07:
well, good news, seems like the Raspbian devs are looking at the php-smbclient "issue"

https://www.raspberrypi.org/forums/viewtopic.php?p=1493654&sid=38ddf1175fcaad897f294ad95c7f2d9a#p1493654

closed 2019-07-07 by nachoparker