nextcloud/nextcloudpi #938: dhcpcd does not properly start on boot or reboot - must be started manually

Halloran, 2019-07-06:
Each time the machine is booted, dhcpcd fails to start properly, and therefore, the it does not get a proper ip v4 address. In order to resolve, I need to manually log in @ nextcloudpi.local, and then `sudo dhcpcd` to start dhcpcd.

This works fine, but obviously it prevents unattended reboots, which may happen if power is interrupted, etc.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.14.0
NextCloudPi image    NextCloudPi_09-28-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/nextCloudDrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      769G/3.6T
rootfs usage         5.0G/15G
swapfile             /media/USBdrive/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    15.0.8.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               10min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.20",
            "3": "hallorancloud.ddns.net",
            "4": "hallorancloud.ddns.net",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/192.168.1.20\/",
        "dbtype": "mysql",
        "version": "15.0.8.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 82 lines cut ...]
```

{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:14:33+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::info: Repair info: Removed potentially over exposing link shares","userAgent":"--","version":"15.0.6.1"}
{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:14:33+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::startCheckCodeIntegrity: Starting code integrity check...","userAgent":"--","version":"15.0.6.1"}
{"reqId":"2ZQ56aCtQVceEQlOGLlr","level":0,"time":"2019-06-17T11:15:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":"Update required, skipping cron","userAgent":"--","version":"15.0.6.1"}
{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:15:17+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::finishedCheckCodeIntegrity: Finished code integrity check","userAgent":"--","version":"15.0.6.1"}
{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:15:17+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::updateEnd: Update successful","userAgent":"--","version":"15.0.8.1"}
{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:15:17+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"15.0.8.1"}
{"reqId":"wmQxANFaCpofCkqbxG2d","level":1,"time":"2019-06-17T11:15:17+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"15.0.8.1"}
{"reqId":"XQeCn38AAQEAAAXyZUkAAAAH","level":4,"time":"2019-06-17T12:07:59+00:00","remoteAddr":"76.15.52.150","user":"john.halloran","app":"webdav","method":"PUT","url":"\/remote.php\/dav\/files\/john.halloran\/Projects\/SnappSchool\/Snapp%20School\/Spanish%20Translations\/K.CC.4-5%09Counting%20Objects.gdoc","message":{"Exception":"OCA\\DAV\\Connector\\Sabre\\Exception\\InvalidPath","Message":"File name contains at least one invalid character","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Tree.php","line":76,"function":"getChild","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["K.CC.4-5\tCounting Objects.gdoc"]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/LockPlugin.php","line":59,"function":"getNodeForPath","class":"Sabre\\DAV\\Tree","type":"->","args":["files\/john.halloran\/Projects\/SnappSchool\/Snapp School\/Spanish Translations\/K.CC.4-5\tCounting Objects.gdoc"]},{"function":"getLock","class":"OCA\\DAV\\Connector\\Sabre\\LockPlugin","type":"->","args":[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/john.halloran\/Projects\/SnappSchool\/Snapp%20School\/Spanish%20Translations\/K.CC.4-5%09Counting%20Objects.gdoc","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"OCA\\DAV\\Connector\\Sabre\\LockPlugin"},"getLock"],[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/john.halloran\/Projects\/SnappSchool\/Snapp%20School\/Spanish%20Translations\/K.CC.4-5%09Counting%20Objects.gdoc","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":466,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["beforeMethod",[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/john.halloran\/Projects\/SnappSchool\/Snapp%20School\/Spanish%20Translations\/K.CC.4-5%09Counting%20Objects.gdoc","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/john.halloran\/Projects\/SnappSchool\/Snapp%20School\/Spanish%20Translations\/K.CC.4-5%09Counting%20Objects.gdoc","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":301,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","Line":227,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.5.1final (build 20181204) (Nextcloud)","version":"15.0.8.1"}
{"reqId":"XQj6gn8AAQEAABzpBCsAAAAI","level":3,"time":"2019-06-18T14:56:47+00:00","remoteAddr":"76.15.52.150","user":"john.halloran","app":"no app in context","method":"PUT","url":"\/remote.php\/dav\/files\/john.halloran\/Projects\/Solidity\/inbox\/node_modules\/secp256k1\/src\/secp256k1-src\/src\/ecmult_impl.h","message":{"Exception":"Sabre\\DAV\\Exception\\BadRequest","Message":"expected filesize 42374 got 16384","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","line":156,"function":"put","class":"OCA\\DAV\\Connector\\Sabre\\File","type":"->","args":[null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1096,"function":"createFile","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["ecmult_impl.h",null]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":525,"function":"createFile","class":"Sabre\\DAV\\Server","type":"->","args":["files\/john.halloran\/Projects\/Solidity\/inbox\/node_modules\/secp256k1\/src\/secp256k1-src\/src\/ecmult_impl.h",null,null]},{"function":"httpPut","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/ [... cut ...] \/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":222,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.5.1final (build 20181204) (Nextcloud)","version":"15.0.8.1"}

```
</details>

nachoparker, 2019-07-06:
so when you reboot, what do you see? any errors in the systemd unit?

we are making any modifications to the DHCP service

TheGreatestJannet, 2019-07-07:
I can confirm I also experienced this on a Pi 3B using the prebuilt image. I then installed Raspbian Lite and ran the Debian script and haven’t experienced any issues since.

Halloran, 2019-07-07:
In my case, when I boot and watch the sequence, there is a line reporting that dhcpcd is starting, and that line simply hangs for ~90seconds, and then the boot sequence continues. Once boot completes, I can login either from a connected terminal, or over ssh using `pi@nextcloudpi.local`. At that point running `sudo dhcpcd` works just fine. So my guess is that there is some problem with the sequencing of boot up.

To be clear, the problem is that the machine is not getting static ip address without manual intervention. Therefore, the forwarding that I've set up for my ddns configuration does not work (until dhcpcd is started manually).

This problem started happening relatively recently (< 6 mos) after an upgrade that was installed via the NextCloudPi admin console. Before that, everything was working fine. I apologize for not remembering the exact version upgrade, but my recollection is that it was an upgrade that included moving to dhcpcd from another ip client.

The machine is RPi 3 Model B Rev 1.2, networking is via ethernet cable, and WiFi is not configured or being used (based on the forums, I'd guess that hard-wiring of the network is unusual, so may play into the problem).

Thanks for any advice/suggestions.

nachoparker, 2019-07-10:
@Halloran did you use `nc-static-IP` at any point or touch the configuration? or are you just using Raspbian defaults?

Halloran, 2019-07-11:
The problem started happening before I was aware of that setting (nc-static-IP). But during the process of trying to troubleshoot, I did toggle the setting. In addition, FWIW, I have tried pretty much every combination of setting the ip manually on the NCP machine and having it set to a static address by the router (which is the DHCP server). None of the combinations seemed to have any impact on the problem.

Currently, things are set up so that there is no configuration on the client, but the router is set to assign a static address to the RPi based on it's MAC. The Router is an ASUS model, and the firmware is up to date.

TimMolenaar, 2019-07-13:
Not sure if I should open another issue for this, but I encountered a very similar problem. Upon booting NextCloudPi with my keyboard plugged in, dhcpcd starts normally and is able to fetch an IP-address from the router. However, if the keyboard is unplugged and the PI is rebooted, dhcpcd returns a timeout and no IP-address is assigned. My guess is that plugging in/out the keyboard changes the default interface name for my ethernet cable (not sure if it has something to do with it, but I read something about predictable interface names being disabled in rasbian stretch: https://madlab5.blogspot.com/2017/08/raspbian-stretch-renames-network.html).

I enabled a static ip-address using ncp-config (tried both the web-UI as terminal).

My dhcpcd.conf:
```
...
# Local loopback
auto lo
iface lo inet loopback

# NextCloudPi autogenerated
# don't modify! better use ncp-config
interface eth0
static ip_address=192.168.2.130/24
static routers=192.168.2.254
static domain_name_servers=192.168.2.254

# Local loopback
auto lo
iface lo inet loopback
```
Note: journalctl mentions the following:
```
...
dhcpcd[377]: unknown option: auto
dhcpcd[377]:unknown option: iface
...
```

NextCloudPi version: 1.14.3
OS: Raspbian Stretch

Halloran, 2019-07-13:
FWIW, my nextcloudpi has not had a keyboard attached at any time while this problem has been in play, so that could explain something. Unfortunately, I don't have a keyboard available to test.

@TimMolenaar I'm unclear about your use of static-ip via nap-config. Did that cause dhcpcd to succeed at boot? Or not?

TimMolenaar, 2019-07-13:
My point was that, whether or not I used ncp-static-ip, dhcpcd did not start correctly on boot because it timed out

TimMolenaar, 2019-07-13:
I can also confirm that problems started after upgrading to the latest nextcloudpi, like TheGreatestJannet mentioned

nachoparker, 2019-07-16:
Cannot reproduce here and we don't deviate from Raspbian behavior so unless someone else can help by diagnosing and proposing a fix not much I can do

closed 2019-07-16 by nachoparker

jcharbonneau15, 2019-07-18:
Hello,

I see the issue has been closed, but I might still have a solution you might want to try. I ran into this same problem running headless and found a post of the raspberry pi forums that fixed the issue for me:

https://lb.raspberrypi.org/forums/viewtopic.php?t=230779

Basically, install then enable/start haveged

`sudo apt-get install haveged`
`systemctl enable haveged.service`

Its worth reading about what this is and if you feel comfortable installing it on your system:

https://wiki.archlinux.org/index.php/Haveged

nachoparker, 2019-07-18:
very interesting @jcharbonneau15 , thanks

If I see a bunch of people confirming that this fixes their issues I will definitely add it

TimMolenaar, 2019-07-18:
@jcharbonneau15 thank you for the possible solution. I've just installed and enabled ```havegd``` on my device, and it survived a headless reboot (I can connect again via SSH). Don't wanna jump to conclusions, but it seems like your solution is the correct one for me :+1:

nachoparker, 2019-07-19:
I am adding those lines to the next image. Does this solution fix the issue for you @Halloran?

SoongJr, 2019-07-20:
Thanks for the patch @nachoparker, but you forgot the -y parameter to prevent the query for user input. Bad for the update via web interface ;) 
I've created a Pull Request (glad github makes that easy without cloning the whole thing :)): https://github.com/nextcloud/nextcloudpi/pull/952

nachoparker, 2019-07-20:
ah right, fixed thanks :)

Halloran, 2019-07-21:
Sorry to be out of touch on this one. I am on vacation and won't be able to access the pi in question for another couple of weeks. But I will try it then as it seems to make sense, and update the thread.

nachoparker, 2019-07-23:
I added it to the latest image. Still would love some more feedback. Enjoy your vacation time

Halloran, 2019-08-10:
Yup. Fixed it.

Thx, @jcharbonneau15 

@nachoparker apologies for taking so long; the machine is usually remote to me - hence the issue. But this fixed me. Thanks for all the work on the project. Really useful for me.