nextcloud/nextcloudpi #947: Add support for Debian Buster

pmeckoni, 2019-07-12:
With the [recent release of Debian Buster](https://www.debian.org/News/2019/20190706) please consider adding Debian 10 Buster to the list of supported systems.

feutl, 2019-07-12:
I also would like to know how to safely upgrade my debian stretch running NCP to debian buster.

Thanks

ovpc, 2019-07-12:
@pmeckoni Buster is being tested, there already is an image for RPI at https://ownyourbits.com/downloads/testing/

@feutl This is not the place to get assistance, please use the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm).

If you want to help testing, join us at https://t.me/NextCloudPi

ovpc, 2019-07-12:
I tested upgrade using v 1.15.0 devel and ncp-dist-upgrade on a rpi3B 
It seems to break my install, get 403 Forbidden 
Output at https://pastebin.com/UNBtZbtJ

feutl, 2019-07-15:
@ovpc I know how to upgrade a debian system, but I already learned something new - the command ncp-dist-upgrade ;)
I am waiting until someone officially says that an upgrade is safe, but thanks

nachoparker, 2019-07-15:
@feutl I tested it on three devices. More testers are welcome. OPVC's pi died with HW problems so his upgrade didn't go through. 

If it fails for whatever reason (like some package needs dependencies), `ncp-dist-upgrade` can be ran again after the conflict is fixed manually

feutl, 2019-07-16:
@nachoparker well thanks, I gonna test tomorrow evening :)

nachoparker, 2019-07-16:
Instructions:

```
ncp-update devel
ncp-dist-upgrade
```

nachoparker, 2019-07-17:
I would love a couple testers giving me the thumbs up so I can merge to master soon and make it available to everyone.

ovpc, 2019-07-17:
On rpi3b ncp-distr-upgrade ended with 👍 Maintenance mode disabled
Upgrade to buster successful 
 
Just imagick missing warning in NC admin
[image]

NCPweb shows Raspbian 10
[image]

Thank you!
Let me know if want big uploads tested here?

ovpc, 2019-07-17:
When trying to enable ufw:
> ERROR: Couldn't determine iptables version

nachoparker, 2019-07-17:
iptables should be fine after a reboot. 

I disabled imagick because of some security concerns 

https://github.com/nextcloud/server/issues/13099

Probably nothing serious but just to be safe, let's wait for feedback from the NC security team.

nachoparker, 2019-07-17:
many thanks!

ovpc, 2019-07-17:
Yep, ufw fine after reboot.

nachoparker, 2019-07-18:
merging Buster

closed 2019-07-18 by nachoparker

pmeckoni, 2019-07-19:
So now we should have Debian Buster listed in the "Supported Systems" along with Debian Stretch on the webpage https://ownyourbits.com/nextcloudpi/ ?

feutl, 2019-07-19:
Sorry for taking me so long
I did an upgrade on my x86 machine running debian installed via ncp script. It ran through without errors but errors I have

* imagick (which is stated above)
and

```
sudo ncp-config
cat: /var/run/.ncp-latest-version: No such file or directory
Running nc-update
Downloading updates
Performing updates
Update to buster available. Type 'sudo ncp-dist-upgrade' to upgrade
Unknown user
```

feutl, 2019-07-19:
I reran the ncp-dist-upgrade and did get the following, have no idea how to fix this.

```
$ sudo ncp-dist-upgrade 
Already on the lastest recommended distribution. Abort.
$ sudo ncp-config 
Running nc-update
Downloading updates
Performing updates
Update to buster available. Type 'sudo ncp-dist-upgrade' to upgrade
Unknown user
Done. Press any key...
```

I also figured out that `smbclient` was removed and not installed again. This made my mounted smb shares unusable inside nextcloud. `apt install smbclient` fixed it.

nachoparker, 2019-07-19:
@pmeckoni good point, thanks, I just did some updates to that page

@feutl you can ignore the Imagick warning. I removed it on purpose for security reasons but it's only used by the theming app. You can re-add it if you want it. Also fixed the false recommendation to dist upgrade again.

Thanks for the help

devincowan, 2019-07-20:
ncp-dist-upgrade
on a Rock64
received the following:
> Processing triggers for dbus (1.12.16-1) ...
> Errors were encountered while processing:
>  locales
> E: Sub-process /usr/bin/dpkg returned an error code (1)
> Something went wrong. Fix it and try again

Did
sudo dpkg --configure -a
then ncp-dist-upgrade worked fine.

Thanks!

cervorange, 2019-08-17:
Hello, 

I pass to ncp-dist-upgrade on rpi3b, i reboot two times and follow the command i need to do. The serveur works, i have acces with the android app and pc app. I have just after a problem with the internet interface (on firefox and chrome) but it seems to work 2 hours after (not tested NCP access).  

Thank you!!!