nextcloud/nextcloudpi #948: Can't upload files larger than 2Gb on Odroid HC2

F1orian, 2019-07-15:
**Issue:**
I had Nextcloud set up on OMV with MariaDB but wasn't able to upload files larger than 2GB. I was unable to fix the issue so I installed the pre-configured nextcloudpi image instead. However I'm still not able to upload files larger than 2Gb. I've tested WebDav and Firefox.

The last 2 entries in the log are after I tried implementing this fix https://github.com/nextcloud/server/issues/1707#issuecomment-291835214, so they should probaböy be disregarded. Before that the system was a fresh install.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.14.3
NextCloudPi image    NextCloudPi_03-09-19
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      4.4G/5.5T
rootfs usage         3.8G/58G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    16.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e063684d3
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               28min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "2a02:908:1587:19e0:235c:95b8:51a9:45c0",
            "1": "192.168.178.116",
            "22": "192.168.178.116"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "16.0.2.1",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
[... 93 lines cut ...]
```

{"reqId":"XSzZ5X8AAQEAABPbmF4AAACI","level":2,"time":"2019-07-15T19:54:38+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzZ5X8AAQEAABL8rxgAAABJ","level":2,"time":"2019-07-15T19:55:04+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzZ-38AAQEAABPbmHQAAACP","level":2,"time":"2019-07-15T19:55:04+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzZ5X8AAQEAABPbmGAAAACK","level":2,"time":"2019-07-15T19:55:29+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzaGH8AAQEAABL8rxwAAABN","level":2,"time":"2019-07-15T19:55:29+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzaGH8AAQEAABL8rx0AAABO","level":2,"time":"2019-07-15T19:55:54+00:00","remoteAddr":"192.168.178.104","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'rpi' (Remote IP: '192.168.178.104')","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzefH8AAQEAABPbmSYAAACT","level":3,"time":"2019-07-15T20:14:04+00:00","remoteAddr":"192.168.178.23","user":"ncp","app":"PHP","method":"PUT","url":"\/remote.php\/webdav\/Lightroom.7z.part","message":"filesize(): stat failed for \/media\/USBdrive\/ncdata\/ncp\/files\/Lightroom.7z.part at \/var\/www\/nextcloud\/lib\/private\/LargeFileHelper.php#171","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzefH8AAQEAABPbmSYAAACT","level":3,"time":"2019-07-15T20:21:16+00:00","remoteAddr":"192.168.178.23","user":"ncp","app":"PHP","method":"PUT","url":"\/remote.php\/webdav\/Lightroom.7z.part","message":"filesize(): stat failed for \/media\/USBdrive\/ncdata\/ncp\/files\/Lightroom.7z.part at \/var\/www\/nextcloud\/lib\/private\/LargeFileHelper.php#171","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzefH8AAQEAABPbmSYAAACT","level":3,"time":"2019-07-15T20:21:16+00:00","remoteAddr":"192.168.178.23","user":"ncp","app":"PHP","method":"PUT","url":"\/remote.php\/webdav\/Lightroom.7z.part","message":"filesize(): stat failed for \/media\/USBdrive\/ncdata\/ncp\/files\/Lightroom.7z.part at \/var\/www\/nextcloud\/lib\/private\/LargeFileHelper.php#171","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) KHTML\/5.59.0 (like Gecko) Konqueror\/5 KIO\/5.59","version":"16.0.2.1"}
{"reqId":"XSzefH8AAQEAABPbmSYAAACT","level":3,"time":"2019-07-15T20:21:16+00:00","remoteAddr":"192.168.178.23","user":"ncp","app":"no app in context","method":"PUT","url":"\/remote.php\/webdav\/Lightroom.7z.part","message":{"Exception":"Doctrine\\DBAL\\Exception\\NotNullConstraintViolationException","Message":"An exception occurred while executing 'INSERT INTO `oc_preview_generation` (`uid`, `file_id`) VALUES(?, ?)' with params [\"ncp\", null]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1048 Column 'file_id' cannot be null","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":184,"function":"convertException","class":"Doctrine\\DBAL\\Driver\\AbstractMySQLDriver","type":"->","args":["An exception occurred while executing 'INSERT INTO `oc_preview_generation` (`uid`, `file_id`) VALUES(?, ?)' with params [\"ncp\", null]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1048 Column 'file_id' cannot be null",{"errorInfo":["23000",1048,"Column 'file_id' cannot be null"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/DBALException.php","line":158,"function":"wrapException","class":"Doctrine\\DBAL\\DBALException","type":"::","args":[{"__class__":"Doctrine\\DBAL\\Driver\\PDOMySql\\Driver"},{"errorInfo":["23000",1048,"Column 'file_id' cannot be null"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"},"An exception occurred while executing 'INSERT INTO `oc_preview_generation` (`uid`, `file_id`) VALUES(?, ?)' with params [\"ncp\", null]:\n\nSQLSTATE[23000]: Integrity constraint violation: 1048 Column 'file_id' cannot be null"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Connection.php","line":1088,"function":"driverExceptionDuringQuery","class":"Doctrine\\DBAL\\DBALException","type":"::","args":[{"__class__":"Doctrine\\DBAL\\Driver\\PDOMySql\\Driver"},{"errorInfo":["23000",1048,"Column 'file_id' cannot be null"],"__class__":"Doctrine\\DBAL\\Driver\\PDOException"},"INSERT INTO `oc_preview_generation` (`uid`, `file_id`) VALUES(?, ?)",{"1":"ncp","2":null}]},{"file":"\/var\/www\/nextcloud\/lib\/private\/DB\/Connection.php","line":216,"function":"executeUpdate","class":"Doctrine\\DBAL\\Connection","type":"->","args":["INSERT INTO `oc_preview_generation` (`uid`, `file_id`) VALUES(?, ?)",["ncp",null],[2,2]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctri [... cut ...] d\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/File.php","Line":222,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:67.0) Gecko\/20100101 Firefox\/67.0","version":"16.0.2.1"}

```
</details>

nachoparker, 2019-07-16:
Thanks for reporting.

I just uploaded a 5GiB file. NC13.0.3, NCP1.15.0

That fix you mention is no longer needed since NC13

nachoparker, 2019-07-17:
Actually I spoke too soon. When  it looked like it was copied it failed with

[image]

This should be reported to the Nextcloud team

nachoparker, 2019-07-17:
Reported https://github.com/nextcloud/server/issues/16431

F1orian, 2019-07-17:
Thanks a lot!

F1orian, 2019-07-17:
This is a different error message I get when uploading through the web client.

[image]

delfer, 2019-10-24:
Fixed by docker-library/php#901
Tested in Docker image `nextcloud:17-apache`  (`nextcloud@sha256:58836d449406b396c1c119c2661577fd7d8284ff41a7d181a0e2be232d2757fc`, `2019-10-24T05:04:49.946856094Z`) **Not Pi**