nextcloud/nextcloudpi #950: nextcloudpi does not update letsencrypt certificate

fermat2a, 2019-07-19:
I installed a letsencrypt certificate using the ncp web interface 3 month ago. Now I could not update it, because the file `/usr/bin/letsencrypt` was not existing in my container. I could repair this temporarly by getting into the container with `docker exec` and than installing letsencrypt by `apt install letsencrypt`. Thereafter I could update the certificate from the ncp web interface.

I am running nextcloudpi v1.10.0. The webgui tells me that v1.15.0 is existing but it fails to update and does not give me any output.

I created the conatiner calling `docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /data/nextcloud/ncdata:/data --name nextcloudpi --restart always ownyourbits/nextcloudpi-armhf`. Running `docker pull ownyourbits/nextcloudpi-armhf` tells me my image is already update to date, so stopping and restarting the container nextcloudpi had no effect.

Can someone tell me how to update the container?

ovpc, 2019-07-19:
Should use forum for help please. But just pull new image

fermat2a, 2019-07-19:
I tried to pull new image:

 ```bash
fermat@web:/data/nextcloud/ncdata $ docker pull ownyourbits/nextcloudpi-armhf
Using default tag: latest
latest: Pulling from ownyourbits/nextcloudpi-armhf
Digest: sha256:876943feeab9c716708be94bef426ed09ebb1b3b6ddd31afdc9941e53df65c25
Status: Image is up to date for ownyourbits/nextcloudpi-armhf:latest
fermat@web:/data/nextcloud/ncdata $
```

So, I seam to have the most recent version. Nevertheless I restartet the container with `docker stop nextcloudpi && docker start nextcloudpi` therafter. No change...

Nevertheless I could update ncp now by calling ncp-update in the console within the container. Now I can also run the ncp update from the web interface, so problem seems to be solved?!? I am wondering why e.g. letsencrypt is still there after restarting the container. For my understanding such changes for the images should not be persistent.

fermat2a, 2019-07-19:
Thanks for the hint to the forum of nextcloud, I did not know that it handles also the container images. Regarding https://help.nextcloud.com/t/how-to-update-nextcloudpi-docker-container/29282 I have to remove the container after pulling the new one and to create it again with same volumes. I think I still do not completly understand the binding between images and container...

ovpc, 2019-07-19:
Image is what you pull from hub, once saved you create a container from it.
[more .....](https://docs.docker.com/engine/reference/commandline/docker/)

fermat2a, 2019-07-19:
After doing so, everything is dead... I started the container again but no webserver is coming up. also after opening a shall in the container the nc-commands are no more in the path.

closed 2019-07-19 by fermat2a

fermat2a, 2019-07-19:
Just for follow up: I created the forum topic https://help.nextcloud.com/t/update-of-nextcloudpi-container-failed/56892 as this seems not to be a bug but a need for support.